MarlinFirmware/Marlin #7225: Skipping due to excessive acceleration during retracts

c-born, 2017-07-03:
I've just updated from 1.1.0-RC2, which has been working really well for me  over the last couple of years, except that it would occasionally forget to deploy the z-probe and crash badly, to 1.1.3.
I found right from the start of the print (at the end of the skirt) it would start grinding and banging and skipping steps, and I spent a few days checking parameters, and testing lower acceleration, tuning jerk etc, to no avail.
I then stepped through the code and located the problem areas, which coincided to where Slic3r was doing a retract, which I have set at 3mm. I removed the retract, and all seems good, back on the original settings from 1.1.0-RC2, except of course no retract. Lots of searching, found some references to retraction when moving causing problems, but nothing current. But even when I set acceleration down to 50, I still get a huge attempted acceleration in Y whenever the retracts are tried.
I've just now updated to the latest 1.1.x-bugfix, but the problem is still there.
Seems strange nobody else has commented on this, have I missed something?
I'm not using the new LIN_ADVANCE or anything special.

c-born, 2017-07-04:
Following up on this, I enabled firmware retract in Marlin and on Slic3r, currently printing the same part, and the skips and stutters in Y are gone. The first half of the print is horribly over-extruded and blobby as the default firmware retract speed was too high and my extruder just buzzed and didn't move, but I've halved it during the print and it is behaving much nicer now.
When the retract takes place XY movement pauses rather than continuing smoothly though, as it did using Slic3r's software generated retract and the older (1.1.0) Marlin firmware

zlowred, 2017-07-05:
I can confirm the same issue with 1.1.3

waiwong614, 2017-12-03:
Does anyone know if this is fixed? I've been tearing my hair out with 1.1.6 until I read this.

Roxy-3D, 2017-12-03:
Configuration.h files would be helpful.    I haven't seen any of these issues.   But I typically don't turn on firmware retraction.

c-born, 2017-12-03:
I'm still running 1.1.4, with firmware retract I don't see the problem.
I don't know if it is still there with firmware retract turned off.

thinkyhead, 2018-02-10:
Presuming solved.

closed 2018-02-10 by thinkyhead