MarlinFirmware/Marlin #7229: RepRapDiscount FULL GRAPHIC Smart Controller

oz6ym, 2017-07-03:
# **Have a Compiler-error:**
Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\OZ6YM\AppData\Local\Temp\arduino_cache_519927\core\core_arduino_avr_mega_cpu_atmega2560_b327234c182a9507b43cdc52bd64fc97.a
libraries\U8glib\clib\u8g_font.c.o (symbol from plugin): In function `u8g_font_calc_vref_font':

(.text+0x0): multiple definition of `u8g_GetFontBBXOffX'

libraries\U8g2\clib\u8g2_font.c.o (symbol from plugin):(.text+0x0): first defined here

c:/program files/windowsapps/arduinollc.arduinoide_1.8.6.0_x64__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

libraries\U8glib\clib\u8g_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':

(.text+0x0): multiple definition of `pg_ClearPolygonXY'

libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin):(.text+0x0): first defined here

libraries\U8glib\clib\u8g_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':

(.text+0x0): multiple definition of `pg_AddPolygonXY'

libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin):(.text+0x0): first defined here

libraries\U8glib\clib\u8g_polygon.c.o (symbol from plugin): In function `pg_ClearPolygonXY':

(.text+0x0): multiple definition of `pg_DrawPolygon'

libraries\U8g2\clib\u8g2_polygon.c.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
-------------------------------------------------------------------------
I don't know what to do....
If i jump over this with //
the program compile normaly, so something must have changed...?
HELP...

fiveangle, 2017-07-03:
Compiles fine for me.  You may want to provide your configs in a zip format.

Bob-the-Kuhn, 2017-07-03:
What are you using to compile Marlin?  It doesn't look like the error messages I see on my Arduino Windows setup.

Maybe it's a library issue.  Try removing all libraries with U8g in the name and then add them back in starting with U8g2. 

If that doesn't work then we'll want to have your config files so we can try to reproduce the problem.

fiveangle, 2017-07-03:
> then add them back in starting with U8g2

I thought Marlin doesn't support U8g2 ?

oz6ym, 2017-07-03:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1120396/Configuration.zip)
I have made a HOMEMADE 3D-printer. 
Using ARDUINO IDE, Ramps 1.4 and Mega 2560.
An earlyer version is compiling fine with no error, but in that version the autolevel ajustment dit not work, so now I will try the latest software to se, if the old error is fixed.

That is my goel, but I am not a programmer....

Bob-the-Kuhn, 2017-07-04:
Your configuration.h file compiled for me with no problems.

I used bugfix-1.1.x from today.

What version of Marlin are you using?

----

Your comments all say you're running a RAMPS 1.4 board but your configuration.h file specifies RAMPS 1.3. The only thing this affects is the logical pin used for SERVO0.  As best I can tell you're not using a servo so this won't affect you.

oz6ym, 2017-07-04:
Bob...
I am using Arduino IDE version 1.6.5.0
Trying to Compile Marlin 1.1.x, downloaded 2017.07.01 as the latest version.
IDE needed the U8GLIB and U8G2, so i installed them via the IDE's "Include Library", did the compiling and get this error described.
And YES... I have made an error about the RAMPS, and this is now corrected and as you say, this is not making the error.
Thanks for trying to help me..

Bob-the-Kuhn, 2017-07-04:
Please ZIP up the following & drop on your reply:
- Your entire Marlin directory.
- The complete log from a failing compile.  Make sure you have "Show verbose output during: compilation" checked in Preferences under File.

oz6ym, 2017-07-04:
Dear Bob... 
Here I have what you asked for... I think... If anything is missing... sorry I am not a programmer..
I realy hope you can find the ERROR.
My computer is a HP Z820 Workstation, installed with the latest version og Windows 10, ver: 1703.
This Compilation is done with Arduino IDE ver: 1.8.3 - a clean new installation.
Marlin.zip is the packed ZIP of MARLIN as i would like to use, and the ERROR.zip is the compiler-error.
Hope you can help.

[preferences.zip](https://github.com/MarlinFirmware/Marlin/files/1121366/preferences.zip)
[ERROR.zip](https://github.com/MarlinFirmware/Marlin/files/1121369/ERROR.zip)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1121356/Marlin.zip)

Bob-the-Kuhn, 2017-07-04:
Good news - I was able to reproduce the problem using your Marlin.zip.

Looks like you've run into a bug in 1.1.3 that's already been fixed.  The solution is to download a new copy of Marlin.

The preferred Marlin is [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x).  It has bug fixes in it that the latest 1.1.3 doesn't.

The next best is the latest [1.1.3](https://github.com/MarlinFirmware/Marlin/archive/1.1.3.zip).

oz6ym, 2017-07-04:
Dear Bob...
You are right... This version Compiles without ERRORs, so now I will try to upload to my printer.
For NOW, I thank you so much for your help.
RadioHAM OZ6YM, Palle

oz6ym, 2017-07-04:
Now...
The new software i performing exatly as the old version from july 2016, and I still have a problem, so I will start a new tread..
Thanks to all for input to get the new version compiling...
de OZ6YM, Palle

closed 2017-07-04 by Bob-the-Kuhn