MarlinFirmware/Marlin #7230: Bed Leveling Procedure

devekrehbiel, 2017-07-03:
I do not have Pronterface or an old obsolete version of Cura. Every document I have seen so far rely on one of those programs to set up bed leveling. An additional problem is I do not understand how it works. Is there a mathematical algorithm behind this? Does Marlin keep track of the numbers and adjust the steppers accordingly? Sadly, the way these printers work, both Z motors are tied together, so there is no balancing ability which is a shame.

The way I have been doing it is below.  I have several printers so it would be nice to know how to do it right.

I choose Bi-Linear Leveling but not sure why.

1) Adjust the sensor's potentiometer so I am SURE it will not choke and try to put a hole in the bed. Meaning its about 2mm off the table when going through "Auto Bed Leveling". 

2) Go into Control/Motion/Z-Offset and adjust to -2mm. Trial and error until its perfect.

From time to time, I may have to twist the Z screws a few clicks by hand to get it exactly right.  If there is a document somewhere that does not require using software that we have no use for other than that, let me know. I just want to do it right. Thanks again!

Roxy-3D, 2017-07-03:
UBL's full feature set is now available from the LCD Panel.  (Thanks @Tannoo !)

fiveangle, 2017-07-03:
In short, you are correct: you need to manually set Z offset.  However, this doesn't enable bed levelling.  You will need to set G29 in your slicer start GCODE to trigger bed leveling before each print, then lower Z offset until it is giving you a good first later.

To get more indepth (which I recommend), get pronterface http://reprap.org/wiki/Printrun

Use proterface "terminal" feature to connect to the printer via USB.  Alternately, you can use any terminal program really to connect to the printer: it is just a TTY device.

Then issue G29 manually to the printer to begin bed leveling and observe the output.  M500 saves the results to EEPROM if enabled.

If you download the latest rcbugfix1.1.x it now allows enabling the LCD_BED_LEVELING feature to have a bit more control over the leveling functionality direclty from the LCD.

More info: http://marlinfw.org/docs/gcode/G029-abl.html

apballard, 2017-07-06:
If I may add my 2c worth.

I recently moved from bilinear to UBL. I'm glad I did and I'm not looking back. I had endless hassles with bilinear and all my frustrations are gone with UBL.  It is feature rich, easy to use and reliable.  I need to take the opportunity to commend all those that worked on it.

fiveangle, 2017-07-08:
Agreed UBL is nice and that BILINEAR is frustrating (and isn't near as robust, as I've managed to gouge my bed for the first time since buying in 2014 after just a few days of trying it).  Unfortunately UBL is only fully useful for boards with 256k PROGMEM storage at this time.

Roxy-3D, 2017-07-08:
> If I may add my 2c worth.

Or course!

> I recently moved from bilinear to UBL. I'm glad I did and I'm not looking back. I had endless hassles with bilinear and all my frustrations are gone with UBL. It is feature rich, easy to use and reliable. I need to take the opportunity to commend all those that worked on it.

That was its intent.  That and to handle the horrible calibration problems some low quality Delta printers experience (like mine).   But it is a sophisticated system and that makes it more complicated than most of the existing GCodes and MCodes.  So usually I hear complaints in the opposite direction.

>  Agreed UBL is nice and that BILINEAR is frustrating (and isn't near as robust, as I've managed to gouge my bed for the first time since buying in 2014 after just a few days of trying it). Unfortunately UBL is only fully useful for boards with 256k PROGMEM storage at this time.

@fiveangle   How big is the current UBL when compiled for your machine.   I think we are at the point we we can start turning off big parts of it in a granular basis.   What type of LCD Display do you have?  20x4 is easier on program memory.   But the goal would be to make it work on either type of LCD Panel.

fiveangle, 2017-07-08:
https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration_adv.h

```
 */
//#define AUTO_BED_LEVELING_3POINT // 6366 bytes (w/o LCD)
//#define AUTO_BED_LEVELING_LINEAR // 18386 bytes (w/o LCD)
#define AUTO_BED_LEVELING_BILINEAR // 9804 bytes (w/o LCD)
//#define AUTO_BED_LEVELING_UBL //48592 bytes (w/o LCD)
//#define MESH_BED_LEVELING // 7220 bytes (w/o LCD)
```

Roxy-3D, 2017-07-08:
OK...  I'll play with that setup and see what can be done.

fiveangle, 2017-07-08:
I have a graphics LCD (not ideal but wasn't aware of the PROGMEM requirements when I bought: my main criteria was pin usage since the Printrboard is somewhat limited). In retrospect a text board would have been better.

BTW- If you've not tried the LCD interface on a Prusa MK2, you owe it to yourself. Don't do it of course unless you want to make more work for everyone because it shows how inefficient the current marlin menu UX model is. 

I was so surprised that the top menu doesn't even use the full screen, you've got to dig deep through vaguely appropriate taxonomic sections to get anything done. I love the completeness of the features, but the UX  certainly could use some thought. 

But perhaps there's more complexity to it when features that I don't have are added (trying to give the benefit of the doubt here).

In reviewing the progmem use currently ( my config's above are marked up with a lot of them)  I don't think it's possible to have a graphics LCD and UBL. I was looking into a project that used an arduino that emulates an SPI board for smoothieware as a potential solution (push the graphics to the outboard pro mini that I have lying around) but the project doesn't have the capabilities today so scratched the idea. But I did like the notion of reducing load on the main board. These 8-bit boards just keep on chooch'n :)

fiveangle, 2017-07-08:
Since I have the floor for a second, if I wanted to revamp the graphics bit maps used on the Ug8 LCD, got a pointer to a discussion somewhere  that would give me a place to start ?

devekrehbiel, 2017-07-08:
It would be nice for the masses that are not real savvy with all this if someone made a Marlin Configurator. A simple program that allows you to check/uncheck features. Maybe it checks to see what board you have, then puts up the necessary items to remove then edits the correct files. Just a thought. You guys have a really nice system here.

Also, I noticed on Ebay when you do a search for Ramps 1.5 you get this:
http://www.ebay.com/itm/USA-MKS-Base-V1-5-3D-Printer-Controller-Board-Mega2560-Ramps-1-4-5-A4988-/252801138940

Is this a larger capacity (memory) board like Ramps 1.4 with the Arduino Mega or is it just like the old Anet 1.0 board? I would go to them just to have Marlins feature rich platform on all of my machines.

Roxy-3D, 2017-07-08:
I think this is EXACTLY a Mega2560 with a RAMPS v1.4 (and 5 step sticks).   Except, it is on a single board.   I would prefer to have the Mega2560 with a RAMPS board just so I can swap parts and not be so careful with the single board.   

But for a lot of people, this single board solution is going to make a lot of sense.  And I like how they color coded the various connectors.   And it is nice they have connectors instead of just pins.   This board looks nice.   I'm just in a different mindset.

fiveangle, 2017-07-08:
> …if someone made a Marlin Configurator

You mean like:

https://github.com/MarlinFirmware/Marlin/issues/6722#issuecomment-310838710

:)

Bob-the-Kuhn, 2017-07-08:
The mega2560 solution is something you can do right now.

If you can wait a couple of months the 32 bit version of Marlin will be out and have had enough run time to be confident in using it.  I don't know what CPUs and boards it'll support.

The Marlin fork [MK4duo](https://github.com/MKFirmware/MK4duo) is available now and supports Arduino Due compatible boards.  Since it's a fork there is some time lag between Marlin updates and updates to MK4duo.

closed 2018-02-09 by thinkyhead