MarlinFirmware/Marlin #7233: #define AUTO_BED_LEVELING_BILINEAR

oz6ym, 2017-07-04:
Bed leveling do NOT work in my compilation.
Starts with Homing - OK - go to first point - OK - go X-dir. to mittel of bed, leveling - OK -- trying to go Y-dir (no work, just motorbuzz) leveling - go left to mittel of bed, leveling - OK - go from left to right leveling - OK - (motorbuzz, no work) leveling - go X-dir. to mittel of bed, leveling - OK - go X-dir. to right point, leveling - OK - hereafter it goes to X-homepoint, but NOT Y-homepoint, ends with a Z-homeleveling.
In the display it says: X=15 - Y=330, Z = 10.
Mesuring this, X and Z is OK, but Y is out of trim.

I think I have a Configuration-problem, and this is my Configuration.h in attached ZIP-file.
My BED is 400 x 400 mm, but only using 375 x 375 of it, and the home-poss. is at the front-left corner.
Z-axis is going UP to ZERO, and expand down under printing.

I am not a programmer, so pse... HELP...
de OZ6YM, Palle

FHeilmann, 2017-07-04:
Lets look at the issues with your Y-Axis first, the way I understand is that your Y-Axis isn't moving at all?

Did you, by any chance, upgrade from marlin 1.0.x?

oz6ym, 2017-07-04:
Hello FHeilmann
Not UPGRADE - A compleate new installation of the latest ARDUINO IDE v 1.8.3, and then getting bugfix-1.1.x. to compiling an ERROR with the display, now fixet, then reconfigurated Configuration.h so it drives my homemade printer as the OLD version of MARLIN from juli 2016 - wich had the same problem.
That is why i think i have a configuration-problem.
Thanks for your reply...

oz6ym, 2017-07-04:
This is the SERIAL-Result:

echo:enqueueing "G28"
echo:enqueueing "G29"
echo:busy: processing
X:0.00 Y:0.00 Z:10.00 E:0.00 Count X:0 Y:0 Z:22000
echo:busy: processing
Bilinear Leveling Grid:
      0      1      2
 0 -0.023 -0.464 -1.077
 1 +0.365 -0.186 -0.940
 2 +0.251 -0.332 -1.055
echo:enqueueing "G1 Z10 F12000"
X:325.00 Y:324.00 Z:11.06 E:0.00 Count X:52000 Y:51840 Z:22000
echo:enqueueing "G1 X15 Y330"
echo:enqueueing "G1 Z0.5"
echo:enqueueing "G1 Z10"
-------------------------------------------------------------------
So.. The HEAD do not move in Y-direction under leveladjustment, even it say so.
If i use MOVE AXIS in Y-direction, it works both UP-counting and DOWN-counting, so do I have a software-movement problem when i try to leveladjust ??

I am confused...

FHeilmann, 2017-07-04:
do you have your configuration in a git repo? otherwise please zip up your whole marlin folder and attach it here so we can have a look.

oz6ym, 2017-07-04:
My limited english do not know "git repo", so here is the whole packed:

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1122346/Marlin.zip)

FHeilmann, 2017-07-04:
Does your Head move in Y-direction during G28?

Do you have a working firmware we can compare against?

(git repo = github.com repository, basically your own copy of marlin on github.com)

oz6ym, 2017-07-04:
OK...about "git repo"...

If i use MOVE AXIS in Y-direction, it works both UP-counting and DOWN-counting, so do I have a software-movement problem when i try to leveladjust ??

Sorry... no other working copy...
Now... my time is limited, so I have to put my problem away for some day, but will return..
Thanks...Palle

oz6ym, 2017-09-11:
Now I had the time again to look into my AUTO LEVEL PROBLEM, and i found that my ACCELERATION was my problem. After adjusting a little on thise parametre, I now can do a full 9 point level adjustment.
I thank you for your help and all responses.
VY 73 de OZ6YM, Palle

closed 2017-09-11 by oz6ym