MarlinFirmware/Marlin #7234: Dynamic Cooling Fan Control

wsobel, 2017-07-04:
I've had some issues with an over-ambitious cooling fan and the hot-end faulting with a thermal runaway. The solution was to create a dynamic adjustment to the cooling fan speed that decreases based on the raw PID value before it is scaled and then having the fan slowly creep up until the PID increases to nearly max output. 

I found this very effective and was wondering if anyone else has experienced this problem (maybe the better solution is a better duct, but I'm a programmer...)? I tried to adjust max output in the slicer, but I would still have issues because of the shape of the surface and airflow around the hotend. 

I have made the changes specific to my setup, but it can easily be parameterized and conditionally activated. 

I'm working off of the Skynet3D 2.4.x branch...

I was wondering if there is interested in this patch?

Best,
- Will Sobel

fiveangle, 2017-07-06:
Sounds like the wrong place to address the issue, since it affects the cooling fan output, which may be critical for the print at that point ?

Wouldn't an option to pre-load the PID loop with a new weight that hints at the need for more instantaneous power due to fan cooling while the PID loop resettles shortly after ?  Ideally the weight would be auto-adjust at runtime, or be made part of the autotune process then saved.

Or did I not understand your suggestion ?

wsobel, 2017-07-06:
The issue is when the hotend heater is not able to keep up, even when at maximum output. I'm swapping out for an E3D V6, but this was causing my prints to fail. I think my heater is having issues, so this was a helpful in preventing thermal runaway and alarming out, thus ruining the entire print. 

Pre-loading the PID loop does not help, tried that first by running auto-tune at full speed. If the fan overpowers the heater, there's not much you can do. I tried to adjust the max fan speed as well, but if the geometry caused the cooling to exceed the heater output, I was stuck. At higher levels, the airflow was better and the additional cooling was required. 

I'm hoping with the new hotend I can turn it off, or back it off so that it only kicks in emergencies. 

(Sent from mobile)

fiveangle, 2017-07-08:
If you overpower the hotend so it cannot recover if given enough time, failing is the correct response in my opinion

with an E3D w/40W cartridge and silicone sock, you should be good to 300C regardless of what fan you throw at it

wsobel, 2017-07-09:
The E3D solved the problem-easily able to maintain temp. I have turned off the fan adjustment. 

The dynamic fan adjustment got me past the point where I could print the mount for the E3D. I think I had a faulty heater, so worked as a crutch. 

The problem with failing is you never finish the print or, on the other extreme, have any cooling, so print was also failing. The dynamic adjustment was able to control the fan so things were not perfect, but better than nothing. 

E3D with Bowden is a great setup. 

- W

(Sent from mobile)

closed 2018-02-09 by thinkyhead