MarlinFirmware/Marlin #7236: Bulldog Extruder XL not working after first layer

Eloque, 2017-07-04:
This might not be Marlin related, but I can't rule it out. Bulldog XL geared extruder combined with a E3Dv6 0.8 hotend is not working correctly. The first layer is properly extruded, the other layers are getting less and less filament until only droplets come out. Obviously, this seemed to be a hardware issue, but that has become increasingly unlikely. All parts have been replaced and/or checked and the problem persists.

Since Marlin has been upgraded substantially since I started using it, I am trying to rule out a misconfiguration. The extruder is defined as such

```
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 495 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 3, 25 } 
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }
```

When in Sli3er the extrusion multiplier is set to 2 instead of 1, the problem still occurs, but quite less, leading me to believe that indeed the configuration is off.

Bob-the-Kuhn, 2017-07-04:
Sounds like your `DEFAULT_AXIS_STEPS_PER_UNIT` is off.

I saw where you got the 495 from.

Do you have microstepping enabled?  You may need to multiply the 495 by the microstepping setting.

Another other thing you might do is calibrate the extruder.
1. Remove the nozzle
2. enable cold extrudes
3. do a G0 E100 
4. calculate a new steps per unit based on the actual amount of filament moved
5. Use M92 Exxx to modify the steps per unit
6. do a G0 E100 and check that the new value results in the expected amount of filament

Eloque, 2017-07-05:
I will check. I do not know however where to check microstepping.

Bob-the-Kuhn, 2017-07-05:
Usually there are 2 or 3 jumpers near each stepper driver.

What controller are you using?

Eloque, 2017-07-05:
I am using a RAMPS13, been for years. Never ever checked those jumpers since assembly. But I will now.

Eloque, 2017-07-09:
RAMPS14 actually, I checked. All jumpers are present,  `yes  yes   yes   1/128 step` but since it's an pololu md09b driver, the max of 1/16th would be used, not 1/128th. Now, you say multiply by; do you mean 

`495 x 1/16 = 30,9375 steps`
or
`495 / 1/c16 = 7.920 steps`

Both seem to result in vastly different settings then what I've been using.

Eloque, 2017-07-10:
For context, video of it failing: https://youtu.be/-NTpEYpY1qU

Bob-the-Kuhn, 2017-07-10:
It would be` 495 x 16 = 7920` but that seems excessive.  My plain jane greg/wade geared extruder with microstepping of 8 has a steps per unit of 391.

Based on the "2x but still has some problems" I expect you'll end up in the 1500 - 2500 range.

You really need to calibrate the extruder.  Here's one of the many links that popped up when I googled "calibrate extruder": http://www.instructables.com/id/How-to-calibrate-the-Extruder-on-your-3d-Printer/

Eloque, 2017-07-10:
I calibrated. With hot hot-end, the math goes to 557. Higher, but significantly less than 1500 - 2500.  I am going to replace the hotend I think. The previous J-Head works fine for years with a MGS extruder, untill the plastic part broke off. At least I can figure out if it's hotend or extruder that is causing issues. 

Thank you for you ongoing help however, appreciate it.

Bob-the-Kuhn, 2017-07-10:
Or try increasing the steps per unit in increments of maybe 100 and see if you can find a sweet spot.

closed 2018-02-09 by thinkyhead