MarlinFirmware/Marlin #7239: AUTOLEVELLING WITH INDUCTIVE PROBE , FAULTY

mandeep255, 2017-07-05:
I am using Inductive probe for autolevelling with Marlin 1.01 using Grid Method.
Everything functions well, however , the movement of Z Axis is opposite to the incline, that is in case the tilt of bed requires the nozzle to mov down, it moves up.  Any suggestions pl. 

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Tannoo, 2017-07-05:
Try [bugfix-1.1.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x) for the latest of possible fixes. This may have already been addressed.

mandeep255, 2017-07-06:
Tried but the results are same. regards

Tannoo, 2017-07-09:
Ok. You stated "Autolevel" but did not state which one.

Which one are you having troubles with and have yoou tried any of the others?

mandeep255, 2017-07-09:
Using Inductive Sensor, Linear Method, regards

mandeep255, 2017-07-09:
18 mm NPN probe

Tannoo, 2017-07-09:
Have you tried Bilinear?

Tannoo, 2017-07-09:
Or netter yet, UBL?

mandeep255, 2017-07-09:
No , none of them

mandeep255, 2017-07-09:
Will these help?

Tannoo, 2017-07-09:
Try and see.

Tannoo, 2017-07-09:
On bugfix-1.1.x.

Tannoo, 2017-07-09:
I used to run bilinear with no issues. I now run UBL with very little issues.

mandeep255, 2017-07-09:
Ok will try and revert

Tannoo, 2017-07-09:
Umm. That would not be a revert. Just an option change. Bugfix-1.1.x is the lastest bugfixes applied to all versions of 1.1.x.

mandeep255, 2017-07-09:
Ok

closed 2018-02-09 by thinkyhead