MarlinFirmware/Marlin #7240: V1.1.4 with Anet and  Bed Leveling Sensor -> "Sketch too big;"

Lord-Quake, 2017-07-05:
This is an info to those involved with Marlin.
Up to v1.1.3 I was able to compile successfully using the needed functions in Marlin.
With v1.1.4 a successful compilation is no longer possible.

[image]

The only option I can see without sacrificing important internal functions is to deactivate "SDSUPPORT".
Unfortunately this could be a problem for some when doing long prints as a USB connection is not always reliable while printing.

Leo

benlye, 2017-07-05:
Over in #7190 @fiveangle posted some great info on options you can disable to save space.  You should be able to get 4KB back fairly easily. 

To be honest, I gave up and disabled SD support on my A8 months ago. OctoPrint for the win!

oakdesign, 2017-07-05:
@Lord-Quake I am able to compile 1.1.4 with ABL enabled. What I'm wondering about your maximum shows up wiht 130048 bytes, wheras mine with 131072 bytes. Even so my complied Sketch size is 123596 bytes

Richard

Lord-Quake, 2017-07-05:
@benlye I'm guessing you are using an incorrect board definition file which can cause boot loader corruption.
Here is the correct version for the Anet board: https://github.com/SkyNet3D/anet-board

benlye, 2017-07-05:
@oakdesign, you need to get the latest Anet board definition for the Arduino IDE from https://github.com/SkyNet3D/anet-board.

You see a max upload size of 131072 because your board definition is incorrectly allowing hte IDE access to all the flash space, and you are risking corrupting the bootloader.

benlye, 2017-07-05:
Snap ;-)

oakdesign, 2017-07-05:
@benlye  thanks for the hint used the board definition of a SkyNet2.3.2 comunity edition. I'll the updated version with reduced flash space.

oakdesign, 2017-07-05:
Just switches to the updated board definition from https://github.com/SkyNet3D/anet-board using the bordd layout for the Atmega 1284p bootloader, not Optiboot. My 1.1.4 compile runs through. 
Sketch size 123610 bytes max 126976 bytes

Lord-Quake, 2017-07-05:
I forgot and should have added in my first post that I am running the M600 option (ADVANCED_PAUSE_FEATURE).
Without it but with SDS_SUPPORT enabled I get 123632 bytes (95%). So one has to decide which needs are more important.
@benlye Thanks for the tips on saving space!

closed 2017-07-05 by Lord-Quake