MarlinFirmware/Marlin #7241: UBL not compensating for first layer in G26

andycjw, 2017-07-05:
### Bug Report
- Description: UBL doesn't seems to be working compensating Z-axis during G26 Mesh Validation
- Expected behaviour: FW should compensate for bed leveling and move Z-axis during print
- Actual behaviour: The Z-axis never moved during the print of G26
- Steps to reproduce:
I followed the steps in this documenation below, G28 then G29 P1 then G26
http://marlinfw.org/docs/features/unified_bed_leveling.html

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1123952/Marlin.zip)

Tannoo, 2017-07-05:
>G28 then G29 P1 then G26

`G28` good.
`G29 P1`: basic, but okay.
`G26` by itself won't do much. `G26 C P T3.0` would be better.

Roxy-3D, 2017-07-05:
Some more suggestions:

Did you do a G29 A to activate UBL?   You can run G26's with bed leveling activated or deactivated.   G29 W will tell you what UBL thinks about the world (and whether it is activated or not).

Also, the Fade Height number will factor in if enabled.  Probably you want the Fade Height at some reasonable number such as 10.00 mm.

andycjw, 2017-07-06:
I followed the documentation steps by steps, I did G29 A and checked with G29 W, also set the fade height as described in the documentation.

I also disabled the software min endstops in configuration.h.

Wondering what else did I missed, I tested with the G29 Q test pattern, I can see that the it moved up 10mm when it hit the mesh with the peak, but it didn't move with my own measured mesh.

Roxy-3D, 2017-07-06:
If you post the G29 W people can offer suggestions.   But It might be appropriate to do a G29 Q of some flavor to generate a test matrix.

andycjw, 2017-07-06:
G29 T and G29 W as follows
```
SENT: G29 T
READ: 
READ: Bed Topography Report:
READ: 
READ: (0,9)                                                                   (9,9)
READ: (15,230)                                                                (230,230)
READ:   0.120    0.215    0.220    0.223    0.137    0.192    0.102    0.107    0.155    0.150  
READ:  
READ:   0.125    0.207    0.223    0.215    0.093    0.115    0.100    0.107    0.085    0.152  
READ:  
READ:   0.112    0.207    0.215    0.120    0.120    0.190    0.175    0.102    0.085    0.152  
READ:  
READ:   0.107    0.205    0.135    0.200    0.107    0.187    0.172    0.100    0.088    0.150  
READ:  
READ:   0.120    0.210    0.217    0.120    0.107    0.090    0.135    0.182    0.167    0.157  
READ:  
READ:   0.118    0.200    0.135    0.118    0.170    0.095    0.095    0.187    0.172    0.150  
READ:  
READ:   0.115    0.200    0.132    0.107    0.200    0.190    0.172    0.190    0.165    0.150  
READ:  
READ:   0.118    0.147    0.215    0.225    0.202    0.197    0.172    0.182    0.165    0.152  
READ:  
READ:   0.152    0.200    0.210    0.205    0.205    0.195    0.175    0.187    0.167    0.150  
READ:  
READ: [ 0.175]   0.225    0.227    0.245    0.215    0.200    0.180    0.192    0.167    0.155  
READ: (15,15)                                                                    (230,15)
READ: (0,0)                                                                     (9,0)
READ: ok
SENT: G29 W
READ: Unified Bed Leveling System v1.01 active.
READ: Mesh 1 Loaded.
READ: UBL object count: 1
READ: planner.z_fade_height : 10.0000
READ: zprobe_zoffset: 0.0000000
READ: UBL_MESH_MIN_X  (0 + 15)=15
READ: UBL_MESH_MIN_Y  (0 + 15)=15
READ: UBL_MESH_MAX_X  (245 - (15))=230
READ: UBL_MESH_MAX_Y  (245 - (15))=230
READ: GRID_MAX_POINTS_X  10
READ: GRID_MAX_POINTS_Y  10
READ: MESH_X_DIST  23.89
READ: MESH_Y_DIST  23.89
READ: X-Axis Mesh Points at: 15.000  38.889  62.778  86.667  110.556  134.444  158.333  182.222  206.111  230.000  
READ: Y-Axis Mesh Points at: 15.000  38.889  62.778  86.667  110.556  134.444  158.333  182.222  206.111  230.000  
READ: Kill pin on :41  state:1
READ: 
READ: ubl_state_at_invocation :0
READ: 
READ: ubl_state_recursion_chk :0
READ: 
READ: Meshes go from 0x0248 to 0x0F7F
READ: sizeof(ubl) :  1
READ: 
READ: z_value[][] size: 400
READ: 
READ: EEPROM free for UBL: 0x0D37
READ: EEPROM can hold 8 meshes.
READ: 
READ: Unified Bed Leveling sanity checks passed.
READ: ok
```

Roxy-3D, 2017-07-06:
Your bed is very flat and almost perfectly leveled....    More comments in the morning....

andycjw, 2017-07-06:
there are still some spot you can see it's > 0.2 those are spots where the first layer always fails, I was hoping UBL can mitigate this

Roxy-3D, 2017-07-06:
If you want to verify that UBL is alive and working on your machine...  You can do:
- G28
- G29 P0     (clear the mesh)
- G29 Q 1  (load a test mesh pattern)
- G29 T    (to see test mesh pattern)
- G1 Z 1.0  (Get nozzle close to bed)
- G1 X5 Y100
- G1 X180 Y100    (you should see the nozzle rise and lower as it crosses the line in the test pattern)

But for the problem where you get the failed prints...  Assuming you have UBL turned on, try doing a G26.   The G26 pattern will show you how well your nozzle is positioned at all of the mesh points.  If the nozzle is to high to get good adhesion in an an area it is simple to adjust that.   Move the nozzle to that area and do a G29 P4 R and adjust all the points by .1 mm and run another G26 validation pattern.   Very quickly, you will have a perfectly tuned mesh.   (and of course... you should save it.)

andycjw, 2017-07-07:
When I test with G29 Q, and move the nozzle, I can see it compensate when it crosses the higher line in test pattern, the problem is that z-axis doesn't move with my measured mesh for my bed.

Printing G26 on my mesh always fail on some spot no matter how I tuned it, because it didn't compensate for it, is it because my machine is coreXZ? Is it possible this kind of machine it cannot do Z fade height during printing?

Roxy-3D, 2017-07-07:
No...  It is amazing, but Core-?? worked with UBL right out of the chute.   The fact it is a Core machine doesn't matter to UBL.

OK...  can you post your G29 T mesh map?   And what area of the mesh is failing?   I presume you are failing to get adhesion.   If that is the case....  Lower that area (2" x 2" ???) area by .15mm and try the G26 again.   Is it better after you do that?

andycjw, 2017-07-07:
My G29 T is above, which you commented, it's perfectly leveled? it's failing at those spot > 0.2 mm, lower middle spot, and i don't see my Z-axis move at all during the whole G26 print, that's why I asked, is it not compensating when the z-height measured is below certain height?

Roxy-3D, 2017-07-07:
Do you have this turned on?  This will stop the nozzle from going to negative Z's.
```cpp
// If enabled, axes won't move below MIN_POS in response to movement commands.
#define MIN_SOFTWARE_ENDSTOP
```
But your Mesh Map is entirely positive...  So I don't think this is going to enter into it.

andycjw, 2017-07-10:
this is uncommented, I saw another issue here saying it should be removed

Roxy-3D, 2017-07-10:
And did turning this off resolve things?

closed 2018-02-09 by thinkyhead