MarlinFirmware/Marlin #7243: [Non-Issue / Hypothetical Config] Driving two Z steppers on Separate Channels

BlackLightUK, 2017-07-05:
As we all know, most dual (or more) Z-axis arrangements are usually set up with either the Z steppers wired up in series or parallel (situation dependent) or with a single Z stepper driving two (or more) leadscrews or belts after being chained together. 

It occurs to me that if there is a spare, unused Stepper channel on the board (in my case the channel for the 2nd extruder - MKS Base 1.4 with a single extruder will have E1 spare and unused) then a more clean solution to the dual-Z problem would be to convert E1 into a Z1 and replicate control signals to both Z and Z1 simultaneously...... 

... I just am not entirely sure how I'd go about doing it. 

This isn't urgent by any stretch of the imagination, but if any of the motion-devs can lend me a few minutes of their time to have a think about it, it'd be sincerely appreciated. 

Steve / BlackLightUK

teemuatlut, 2017-07-05:
So something like [Z_DUAL_STEPPER_DRIVERS](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Configuration_adv.h#L267)?

BlackLightUK, 2017-07-05:
Erm..... Yeah.... exactly like that. *facepalm*

How the hell didn't I spot that before ?!? *ticket closed*

closed 2017-07-05 by BlackLightUK