MarlinFirmware/Marlin #7244: ### Compile Error on 1.1.4 with STATUS_MESSAGE_SCROLLING defined

DaveAhrendt, 2017-07-05:
Errors in ultralcd_impl_DOGM.h at lines 429, 439, 465.  Looks like someone left some debug statements.

Arduino: 1.8.3 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

.....deleted for brevity.....

"C:\Users\Dave\AppData\Local\Temp\arduino_build_564315\sketch\ultralcd.cpp" -o "C:\Users\Dave\AppData\Local\Temp\arduino_build_564315\sketch\ultralcd.cpp.o"
In file included from C:\Users\Dave\AppData\Local\Temp\arduino_build_564315\sketch\ultralcd.cpp:75:0:

ultralcd_impl_DOGM.h:465: error: too many decimal points in number

 >>>>>>> bugfix-1.1.x

                ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_564315\sketch\ultralcd_impl_DOGM.h: In function 'void lcd_implementation_status_message(bool)':

ultralcd_impl_DOGM.h:429: error: expected primary-expression before '<<' token

 <<<<<<< HEAD

 ^

ultralcd_impl_DOGM.h:429: error: expected primary-expression before '<<' token

 <<<<<<< HEAD

   ^

ultralcd_impl_DOGM.h:429: error: expected primary-expression before '<<' token

 <<<<<<< HEAD

     ^

ultralcd_impl_DOGM.h:429: error: expected primary-expression before '<' token

 <<<<<<< HEAD

       ^

ultralcd_impl_DOGM.h:429: error: 'HEAD' was not declared in this scope

 <<<<<<< HEAD

         ^

ultralcd_impl_DOGM.h:439: error: expected primary-expression before '==' token

 =======

 ^

ultralcd_impl_DOGM.h:439: error: expected primary-expression before '==' token

 =======

   ^

ultralcd_impl_DOGM.h:439: error: expected primary-expression before '==' token

 =======

     ^

ultralcd_impl_DOGM.h:439: error: expected primary-expression before '=' token

 =======

       ^

ultralcd_impl_DOGM.h:440: error: expected primary-expression before 'const'

     const uint8_t slen = lcd_strlen(lcd_status_message);

     ^

ultralcd_impl_DOGM.h:465: error: expected primary-expression before '>>' token

 >>>>>>> bugfix-1.1.x

 ^

ultralcd_impl_DOGM.h:465: error: expected primary-expression before '>>' token

 >>>>>>> bugfix-1.1.x

   ^

ultralcd_impl_DOGM.h:465: error: expected primary-expression before '>>' token

 >>>>>>> bugfix-1.1.x

     ^

ultralcd_impl_DOGM.h:465: error: expected primary-expression before '>' token

 >>>>>>> bugfix-1.1.x

       ^

ultralcd_impl_DOGM.h:465: error: 'bugfix' was not declared in this scope

 >>>>>>> bugfix-1.1.x

         ^

ultralcd_impl_DOGM.h:475: error: a function-definition is not allowed here before '{' token

 static void lcd_implementation_status_screen() {

                                                ^

ultralcd.cpp:4678: error: expected '}' at end of input

     bool ubl_lcd_clicked() { return LCD_CLICKED; }

                                                  ^

ultralcd.cpp:4678: error: expected '}' at end of input

Using library U8glib at version 1.19.1 in folder: C:\Users\Dave\Documents\Arduino\libraries\U8glib 
exit status 1
too many decimal points in number

NiklasOlofsson, 2017-07-05:
Lots of compile errors here.
The travis build also failed.
```
In file included from C:\Users\Niklas\AppData\Local\Temp\arduino_build_845375\sketch\ultralcd.cpp:75:0:

C:\Users\Niklas\AppData\Local\Temp\arduino_build_845375\sketch\ultralcd_impl_DOGM.h: In function 'void lcd_implementation_status_message(bool)':

ultralcd_impl_DOGM.h:472: error: a function-definition is not allowed here before '{' token

 static void lcd_implementation_status_screen() {

                                                ^

ultralcd.cpp:4665: error: expected '}' at end of input

     bool lcd_detected() { return true; }

                                        ^

ultralcd.cpp:4665: error: expected '}' at end of input

C:\Users\Niklas\AppData\Local\Temp\arduino_build_845375\sketch\ultralcd.cpp: At global scope:

C:\Users\Niklas\AppData\Local\Temp\arduino_build_845375\sketch\ultralcd.cpp:63:17: warning: 'void lcd_babystep_z()' declared 'static' but never defined [-Wunused-function]

     static void lcd_babystep_z();

                 ^

In file included from C:\Users\Niklas\AppData\Local\Temp\arduino_build_845375\sketch\ultralcd.cpp:75:0:

C:\Users\Niklas\AppData\Local\Temp\arduino_build_845375\sketch\ultralcd_impl_DOGM.h:312:13: warning: 'void lcd_implementation_init()' defined but not used [-Wunused-function]

 static void lcd_implementation_init() {

             ^

Using library U8glib at version 1.19.1 in folder: C:\Users\Niklas\Google Drive\Utveckling\Arduino\libraries\U8glib 
exit status 1
a function-definition is not allowed here before '{' token
```

Some "=======", "<<<<<<< HEAD" and ">>>>>>> bugfix-1.1.x" in the code that should not be there. :)

benlye, 2017-07-05:
Yep, there's a problems in 1.1.4:
https://github.com/MarlinFirmware/Marlin/blob/1.1.4/Marlin/ultralcd_impl_DOGM.h#L429

```
<<<<<<< HEAD
    lcd_print_utf(lcd_status_message + status_scroll_pos);
    const uint8_t slen = lcd_strlen(lcd_status_message);
    if (slen > LCD_WIDTH) {
      const bool new_blink = lcd_blink();
      if (last_blink != new_blink) {
        last_blink = new_blink;
        // Skip any non-printing bytes
        while (!PRINTABLE(lcd_status_message[status_scroll_pos])) status_scroll_pos++;
        if (++status_scroll_pos > slen - LCD_WIDTH) status_scroll_pos = 0;
=======
    const uint8_t slen = lcd_strlen(lcd_status_message);
    const char *stat = lcd_status_message + status_scroll_pos;
    if (slen <= LCD_WIDTH)
      lcd_print_utf(stat);                                      // The string isn't scrolling
    else {
      if (status_scroll_pos <= slen - LCD_WIDTH)
        lcd_print_utf(stat);                                    // The string fills the screen
      else {
        uint8_t chars = LCD_WIDTH;
        if (status_scroll_pos < slen) {                         // First string still visible
          lcd_print_utf(stat);                                  // The string leaves space
          chars -= slen - status_scroll_pos;                    // Amount of space left
        }
        u8g.print('.');                                         // Always at 1+ spaces left, draw a dot
        if (--chars) {
          if (status_scroll_pos < slen + 1)                     // Draw a second dot if there's space
            --chars, u8g.print('.');
          if (chars) lcd_print_utf(lcd_status_message, chars);  // Print a second copy of the message
        }
      }
      if (last_blink != blink) {
        last_blink = blink;
        // Skip any non-printing bytes
        if (status_scroll_pos < slen) while (!PRINTABLE(lcd_status_message[status_scroll_pos])) status_scroll_pos++;
        if (++status_scroll_pos >= slen + 2) status_scroll_pos = 0;
>>>>>>> bugfix-1.1.x
```

pmbroth, 2017-07-05:
Same here, get the same

Arduino: 1.8.3 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\ultralcd.cpp:75:0:

ultralcd_impl_DOGM.h:465: error: too many decimal points in number

 >>>>>>> bugfix-1.1.x

                ^

sketch\ultralcd_impl_DOGM.h: In function 'void lcd_implementation_status_message(bool)':

ultralcd_impl_DOGM.h:429: error: expected primary-expression before '<<' token

 <<<<<<< HEAD

 ^

ultralcd_impl_DOGM.h:429: error: expected primary-expression before '<<' token

 <<<<<<< HEAD

   ^

ultralcd_impl_DOGM.h:429: error: expected primary-expression before '<<' token

 <<<<<<< HEAD

     ^

ultralcd_impl_DOGM.h:429: error: expected primary-expression before '<' token

 <<<<<<< HEAD

       ^

ultralcd_impl_DOGM.h:429: error: 'HEAD' was not declared in this scope

 <<<<<<< HEAD

         ^

ultralcd_impl_DOGM.h:439: error: expected primary-expression before '==' token

 =======

 ^

ultralcd_impl_DOGM.h:439: error: expected primary-expression before '==' token

 =======

   ^

ultralcd_impl_DOGM.h:439: error: expected primary-expression before '==' token

 =======

     ^

ultralcd_impl_DOGM.h:439: error: expected primary-expression before '=' token

 =======

       ^

ultralcd_impl_DOGM.h:440: error: expected primary-expression before 'const'

     const uint8_t slen = lcd_strlen(lcd_status_message);

     ^

ultralcd_impl_DOGM.h:465: error: expected primary-expression before '>>' token

 >>>>>>> bugfix-1.1.x

 ^

ultralcd_impl_DOGM.h:465: error: expected primary-expression before '>>' token

 >>>>>>> bugfix-1.1.x

   ^

ultralcd_impl_DOGM.h:465: error: expected primary-expression before '>>' token

 >>>>>>> bugfix-1.1.x

     ^

ultralcd_impl_DOGM.h:465: error: expected primary-expression before '>' token

 >>>>>>> bugfix-1.1.x

       ^

ultralcd_impl_DOGM.h:465: error: 'bugfix' was not declared in this scope

 >>>>>>> bugfix-1.1.x

         ^

ultralcd_impl_DOGM.h:475: error: a function-definition is not allowed here before '{' token

 static void lcd_implementation_status_screen() {

                                                ^

ultralcd.cpp:4678: error: expected '}' at end of input

     bool ubl_lcd_clicked() { return LCD_CLICKED; }

                                                  ^

ultralcd.cpp:4678: error: expected '}' at end of input

exit status 1
too many decimal points in number

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Bob-the-Kuhn, 2017-07-05:
Yes, the rebase/merge conflict marker is still in the file.

Use the file from bugfix-1.1.x.  That's what is supposed to be in 1.1.4.

NiklasOlofsson, 2017-07-05:
I removed the markers but just got other compile errors.

Bob-the-Kuhn, 2017-07-05:
What LCD controller are you selecting?

NiklasOlofsson, 2017-07-05:
Full graphic smart something ;-)

pmbroth, 2017-07-05:
Just pulled bugfix 1.1.x  all is working fine with this..

Bob-the-Kuhn, 2017-07-05:
I'm not getting a compile error which means I haven't selected the correct LCD controller.

Try this file.  It's the file from 1.1.x
[ultralcd_impl_DOGM.zip](https://github.com/MarlinFirmware/Marlin/files/1125443/ultralcd_impl_DOGM.zip)

NiklasOlofsson, 2017-07-05:
That one works. Thanks

benlye, 2017-07-05:
@Bob-the-Kuhn you have to download the 1.1.4 release zip file to see the error.

Bob-the-Kuhn, 2017-07-05:
Unfortunately I don't know what branch to do the PR against so we'll have to wait for @thinkyhead to make the correction.

thinkyhead, 2017-07-05:
I see them in the 1.1.4 download, but oddly not at the tip of the 1.1.x branch and not in my working copy of 1.1.x. I will try to patch up 1.1.x with the intended code.

thinkyhead, 2017-07-05:
Ok, apparently one of the merged files was not saved before pushing. All set now!

closed 2017-07-05 by thinkyhead