MarlinFirmware/Marlin #7245: Anet 1.0 Board Problems

devekrehbiel, 2017-07-05:
I am trying to put the latest Bugfix version on one of my Anet 1.0 boards. I get the code below. Any ideas? Also, seems the Anet 1.0 board doesnt show up on a regular Arduino version, you have to use an old Skynet APIM version for it to show up on the Board list in Arduino. Is there a way around this that you know of? Thanks again! Hope you all had a great holiday.

"C:\3DPrinting\AAA-Programs for 3D\Marlin 1.1.3x\arduino-1.8.0 (with APIM)\hardware\tools\avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega1284p  -o "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068/Marlin.ino.elf" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\G26_Mesh_Validation_Tool.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\I2CPositionEncoder.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\M100_Free_Mem_Chk.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\Marlin.ino.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\MarlinSerial.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\Marlin_main.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\Sd2Card.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\SdBaseFile.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\SdFatUtil.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\SdFile.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\SdVolume.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\blinkm.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\cardreader.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\configuration_store.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\dac_mcp4728.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\digipot_mcp4018.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\digipot_mcp4451.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\endstops.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\gcode.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\hex_print_routines.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\least_squares_fit.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\mesh_bed_leveling.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\nozzle.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\pca9632.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\planner.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\planner_bezier.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\printcounter.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\qr_solve.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\serial.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\servo.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\stepper.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\stepper_dac.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\stepper_indirection.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\stopwatch.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\temperature.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\twibus.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\ubl.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\ubl_G29.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\ubl_motion.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\ultralcd.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\utility.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\vector_3.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\sketch\watchdog.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\U8glib.cpp.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\chessengine.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_bitmap.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_circle.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_clip.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_api.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_api_16gr.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_attiny85_hw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_common.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_fast_parallel.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_hw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_hw_usart_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_no_en_parallel.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_parallel.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_port_d_wr.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_ssd_i2c.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_st7920_custom.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_st7920_hw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_st7920_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_std_sw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_sw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_t6963.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_arduino_uc_i2c.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_atmega_hw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_atmega_parallel.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_atmega_st7920_hw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_atmega_st7920_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_atmega_sw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_atxmega_hw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_atxmega_st7920_hw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_i2c.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_io.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_linux_ssd_i2c.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_msp430_hw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_null.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_psoc5_ssd_hw_parallel.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_raspberrypi_hw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_raspberrypi_ssd_i2c.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_com_std_sw_spi.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_cursor.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_delay.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries\U8glib\clib\u8g_dev_a2_micro_printer.c.o" "C:\Users\Deve\AppData\Local\Temp\arduino_build_445068\libraries [... cut ...] no.elf section `.text' will not fit in region `text'

c:/3dprinting/aaa-programs for 3d/marlin 1.1.3x/arduino-1.8.0 (with apim)/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: region `text' overflowed by 4156 bytes

collect2.exe: error: ld returned 1 exit status

Using library U8glib at version 1.19.1 in folder: C:\3DPrinting\AAA-Programs for 3D\Marlin 1.1.3x\arduino-1.8.0 (with APIM)\portable\sketchbook\libraries\U8glib 
exit status 1
Error compiling for board Anet V1.0.

benlye, 2017-07-05:
1. Go here and get the new Anet board definition for Arduino:
https://github.com/SkyNet3D/anet-board

2. The options you have selected are making your image too large:
```
ld.exe: region `text' overflowed by 4156 bytes
```

You need to edit the config files and disable some features.

devekrehbiel, 2017-07-05:
I wouldnt know how to burn a bootloader. I did everything else. Is there an instruction somewhere on how to do this and what config files need to be edited to remove the correct features? Do I need to burn the bootloader for Marlin to work? Or do I just delete features?

devekrehbiel, 2017-07-05:
Now I know why people are moving to Ramps, I didnt have this problem last week when it was on a Ramps.

benlye, 2017-07-05:
You don't need to burn the bootloader, just don't choose the Optiboot version of the board and you can use the existing bootloader.

devekrehbiel, 2017-07-05:
Ok, so I can now see my Anet board on my stock Arduino board list so thats good. What are your recommendations for shrinking the image by 4200 bytes? I have a leveling sensor but other than that, its the stock printer as far as the software is concerned. You would think as long as Arduino has been around, there would be some enterprising individual that could figure out how to store more memory items on the SD card or something. (not that I know what I am talking about).

benlye, 2017-07-05:
Take a look here #7190 to see the sizes of various things you can disable.

devekrehbiel, 2017-07-05:
I saw that but when it comes to prioritizing things you turn off, you really should understand what they are there for. So far tentatively I disabled
PRINTCOUNTER (configuration.h)  and
ARC_SUPPORT (configuration_adv.h) which should do it, but I do not know what arc support does so maybe that wasnt a good idea. I am using Octoprint, so need to keep that going.

benlye, 2017-07-05:
If you use OctoPrint, you probably don't need SDSUPPORT. Disabling that gets you a lot of headroom.

Lord-Quake, 2017-07-05:
I've decided to disable SD support to solve my problems with the Anet board.

s256, 2017-12-12:
Same problem here. Coming from SkyNet 2.3.1. Disabled sdsupport, but verify crashes with the same "overhead". It tells me it's 35794bytes to big. So the error message stays the same, even after I disabled sdsupport in my configuration.h . What am I missing?

thinkyhead, 2017-12-17:
@s256 First test with 1.1.7 and if the build is still too big, post your configuration files as a ZIP and we'll see if we can get the sketch to fit your board.

s256, 2017-12-18:
@thinkyhead Thanks. I adapted my config files. Disabled every feature I could think of. Went through line by line. Still too big. Please find the zip attached.
[Marlin-1.1.7-ANET-A6-Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1567457/Marlin-1.1.7-ANET-A6-Configs.zip)

devekrehbiel, 2017-12-19:
In the spirit of thinking ahead, it seems to me that this problem is bound to get worse as these guys invent new features for us. Right now, that poor old board is at its very top limit. There is a $40 solution in this: https://www.ebay.com/itm/MKS-Base-V1-5-3D-Printer-Controller-Board-Replace-Ramps-1-4-Mega-2560-A4988/262623311932?epid=14002809244&hash=item3d2591b83c:g:BIsAAOSwYvFZFcwJ I am thinking about going to that with the two of mine that are not Ramps 1.4.

thinkyhead, 2017-12-25:
> Disabled every feature I could think of.

You're unlikely to get any auto bed leveling to fit without going to some extremes. The 1284P has 16K of SRAM, so it may be possible to move several PROGMEM strings to SRAM and recover program memory that way.

>  it seems to me that this problem is bound to get worse as these guys invent new features for us

@devekrehbiel Maybe. Sometimes we do optimizations and reduce code size. New features are (generally) disabled by default so they won't affect the basic footprint.

boelle, 2019-02-19:
@devekrehbiel do you still have problems?

if not maybe consider closing the issue

devekrehbiel, 2019-02-19:
I am okay if you close it. Thanks for asking. The memory is just insufficient on the 1.0 (Anet) board to give you a lot of good choices, however, you did provide us with enough choices to make it work. For those having problems, this software is so good and the support is so good that its worth purchasing a mobo that has sufficient resources for anything you throw at it.

boelle, 2019-02-19:
i cant close it, only you can

boelle, 2019-02-19:
and i only asked since it was in 2017 you last added to this

devekrehbiel, 2019-02-19:
Thanks for your support of us Anet owners!

closed 2019-02-19 by devekrehbiel

boelle, 2019-02-19:
dont buy Anet printers, they dont follow the marlin license :-/