MarlinFirmware/Marlin #7247: [FR] Support for switching extruder using PCA9685 I2C PWM Servo Control

flodejr, 2017-07-06:
Hi,

I am using the Anet 1.0 board, because of the limited pins that this board is exposing, I have to use an I2C pwm servo controller instead of using extra pins.  I have tried adjusting the SWITCHING EXTRUDER functionality in the Marlin firmware, but this function uses servo controls tied to physical pins and there are a number of checks for these dependencies in the code.

At the moment, I have created customized gcodes for switching the servo via I2C and embedded the Adafruit PCA9685 servo library as well and it is working properly to switch the servo.  However, I am stuck with the tool_change functionality, as I would need to reverse the Extruder direction when the servo switches.

I think adding this support will be beneficial to the future 3D printing scene as more and more stuff will be added and eventually the boards will run out of I/O pins.

I can contribute to adding this support but I need some pointers as to the main places to look for these adjustments.

boelle, 2019-06-22:
missing correct label

sjasonsmith, 2020-11-10:
@flodejr I know this is old, but did you ever pursue this at all?