MarlinFirmware/Marlin #7248: PROBING_HEATERS_OFF - Heat bed still active when bed level probing begins.

Lord-Quake, 2017-07-06:
I tried the option "#define PROBING_HEATERS_OFF" because I want to avoid having the heat bed on while the bed level probing is in progress.

I was expecting that probing would wait until the heat bed turns off (visually heat bed light goes out).

However probing begins whilst heat bed is still active. Eventually during probing the heat bed will turn off.

My start G-Code is as follows:
...
M190 S70	 ; Wait for bed temperature to reach target temp
M140 S0           ; Disable heated bed
G29                   ; Begin with bed leveling
...

Bob-the-Kuhn, 2017-07-06:
Try `M190 R69` just before the `M140 S0`

The R parameter says to wait for both heating & cooling.

Lord-Quake, 2017-07-06:
No glory. Same symptoms persist.

Bob-the-Kuhn, 2017-07-06:
I took a look at the code and the intent to turn off the bed (along with all extruder heaters) is there.

Looks like the heaters are disabled when the Z axis is descending for the probe and then immediately re-enabled as the Z axis is raised.

Do you see the red LED blinking at all?

Lord-Quake, 2017-07-06:
If I understand you correctly then the led coming from the heat bed should turn off when the nozzle is going down and the heat bed led should go back on when the nozzle is moving up again.
If that's so then I should set the g-code like this to do the test:
```
M190 S70 ; Wait for bed temperature to reach target temp
G29 ; Begin with bed leveling
```
Correct?

benlye, 2017-07-06:
Hmm, I've enabled `PROBING_HEATERS_OFF` and it works as expected - on G28 the bed LED goes off as the sensor makes its slow approach, and on G29 the bed LED goes off for the downward movement and comes back on as soon as the probe starts to move up again.

Lord-Quake, 2017-07-07:
@benlye Could you please post the complete g-code sequence so that I can try to duplicate your test.

benlye, 2017-07-07:
First I did:
```
M140 S70;
G28;
G29;
```
Then I did:
```
M190 S70;
G28;
G29;
```

You can clearly see the bed LED switch off during both tests - I made a short video which I'll link shortly.

Lord-Quake, 2017-07-07:
Thanks. Video would be great.
I'll give it a try later in the day.

benlye, 2017-07-07:
Here's the video (click to open and play).

[image]

This was after the `M190 S70` command had completed.  The video starts with homing, then I run G29.

Lord-Quake, 2017-07-07:
Great.

Could you also add the 2 configs you are using. I'm assuming you are running v1.1.4. I'd like to do a comparison with my setup.

benlye, 2017-07-07:
[Configuration.h](https://github.com/benlye/Marlin/blob/benlye-anet/Marlin/Configuration.h)
[Configuration_adv.h](https://github.com/benlye/Marlin/blob/benlye-anet/Marlin/Configuration_adv.h)

Printer is an Anet A8 running bugfix-1.1.x built yesterday.

Lord-Quake, 2017-07-07:
Hmmmm.... please save the files as a zip as I can't download the files as is :-/

benlye, 2017-07-07:
'Raw' view is your friend, but here you go.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1130391/Marlin.zip)

Lord-Quake, 2017-07-07:
OK, I've done the first test:

M140 S70; Heat bed to 70°C with LED coming on.
G28; LED from bed turns off (however a little late for my taste) and after homing is complete turns on again. 
G29; LED never turns off while bed is still heating up!

I can provide video if needed.

Lord-Quake, 2017-07-07:
Update

I have now made changes to the default config settings that are provided with Marlin 1.1.4. and compared relevant settings @benlye config settings.

#define TEMP_RESIDENCY_TIME 10   // (seconds) Default was 6
#define TEMP_BED_RESIDENCY_TIME 10   // (seconds)  Default was 6
#define PID_FUNCTIONAL_RANGE 10  // Default was 15
#define PIDTEMPBED // Default inactive
//#define BED_LIMIT_SWITCHING // Default active

I then ran PID tuning for the bed:
// Anet A8 12V heated bed with Kapton and external MOSFET run@70°C
  #define  DEFAULT_bedKp 424.78
  #define  DEFAULT_bedKi 69.09
  #define  DEFAULT_bedKd 652.89

Now running the commands G28 and G29 the LED will go OFF and ON as intended! :-)

The consequence I see is that we need to update the configs that are included in Marlin for the Anet printer.  There are also other settings that need to be changed to more suite the Anet. I think I'll open a new thread for that.

benlye, 2017-07-07:
It would be interesting to mail down which change made the difference.  I would guess that it's `PIDTEMPBED` vs. `BED_LIMIT_SWITCHING`.  It would be interesting if `PROBING_HEATERS_OFF` only works with `PIDTEMPBED` enabled - my assumption is that is not intended.

Lord-Quake, 2017-07-07:
If `PIDTEMPBED` is enabled then `BED_LIMIT_SWITCHING` must be disabled or else the compilation will fail.
I too believe using `PIDTEMPBED` did the trick.

benlye, 2017-07-07:
> If PIDTEMPBED is enabled then BED_LIMIT_SWITCHING must be disabled or else the compilation will fail.

Yes, that is clear.  

I was able to reproduce the behaviour you see, but even more erratically - sometimes the bed went off, other times it stayed on.  I believe that this is because my homing rate on Z is slower than default so probing takes longer on my printer.

I suspect this is the problem:
```
#if DISABLED(PIDTEMPBED)
  #define BED_CHECK_INTERVAL 5000 // ms between checks in bang-bang control
  #if ENABLED(BED_LIMIT_SWITCHING)
    #define BED_HYSTERESIS 2 // Only disable heating if T>target+BED_HYSTERESIS and enable heating if T>target-BED_HYSTERESIS
  #endif
#endif
```

And I confirmed that if you set set `BED_CHECK_INTERVAL` to 1000 the heater is disabled as expected when using BED_LIMIT_SWITCHING.

If you update the subject of this issue to include that, someone may come along and tag it as a bug then work out how to fix it correctly ;-)

Lord-Quake, 2017-07-07:
Best you give me an example of what to put in the subject :-)

benlye, 2017-07-07:
I'd make it "PROBING_HEATERS_OFF doesn't as expected work with BED_LIMIT_SWITCHING"

benlye, 2017-07-07:
LOL, you missed my edit ;-)

Lord-Quake, 2017-07-07:
I thought it sounded corky.... but hey, U da boss :-D

thinkyhead, 2018-02-09:
Please test to see whether #9543 allows `PROBING_HEATERS_OFF` to work better with `BED_LIMIT_SWITCHING`.

closed 2018-02-25 by thinkyhead