MarlinFirmware/Marlin #7250: ### Compile error since Release 1.1 if G38 enabled

sergioausrio, 2017-07-06:
Got a Compiler Error if i enable the G38 Support in Configuration_adv.h. Need this for my MPCNC
Error from Arduino IDE;

sketch\Marlin_main.cpp: In function 'void process_next_command()':

Marlin_main.cpp:10527: error: 'subcode' was not declared in this scope

           if (subcode == 2 || subcode == 3)

               ^

exit status 1
'subcode' was not declared in this scope

thinkyhead, 2017-07-06:
Will patch shortly. Here's the fix:
```cpp
if (parser.subcode == 2 || parser.subcode == 3)
  gcode_G38(parser.subcode == 2);
```

sergioausrio, 2017-07-06:
Thank you. The Compiler error is gone but G38.2Z-99 dont stops on Probe Contact (Z Min Endstop on Ramps). G28Z0 works fine with the same Probe

thinkyhead, 2017-07-06:
Hmm, there haven't been any logical changes to `G38` (as far as I know). I will take a look at 1.1.3 and compare to see what might have broken.

thinkyhead, 2017-07-06:
Seems not to have changed since 1.1.3 — so does it not work in 1.1.3 either?

sergioausrio, 2017-07-06:
No. G38 doesn't work since the 1.1 release. I tried every release. I have one of the last RC8 Bugfix Version with working G38

thinkyhead, 2017-07-06:
So, RC8 seems to be the last? I will compare to that instead…

thinkyhead, 2017-07-06:
Are you using `ENDSTOP_INTERRUPTS_FEATURE`?

sergioausrio, 2017-07-06:
No

thinkyhead, 2017-07-06:
What is your `Z_MIN_PROBE_PIN` setting? The old `G38` used the `Z_MIN_PIN` while the current `G38` requires the use of `Z_MIN_PROBE_PIN`.

sergioausrio, 2017-07-06:
Ok. If have only #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN enabled. Is this my Problem ?
What should i enter on Z_MIN_PROBE_PIN for a normally open switch (Probe with a CNC Bit)

Bob-the-Kuhn, 2017-07-06:
The idea is to use Z_MIN_PIN for homing and Z_MIN_PROBE_PIN for setting your work piece origin.

I believe that this set of definitions will get G38 working:
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_ENDSTOP_INVERTING true

Some of the CNC bits are not very conductive if you're probing in the X or Y directions.  You may need to lightly rub some fine grit sandpaper on the side of the bit.

thinkyhead, 2017-07-07:
I see that #5897 is also concerned with `G38`. It looks like it can be closed.

Can `G38` be made to work with `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`? There may be some obscure reason someone is homing Z to max and choosing to use the Z min pin as the probe pin.

sergioausrio, 2017-07-07:
I got it working with 

#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_ENDSTOP_INVERTING true

Thank you !

PS:
It should be addet to the Documentation that G38 needs the Z_MIN_PROBE_ENDSTOP instead of Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Bob-the-Kuhn, 2017-07-07:
My robot is down so I can't verify this.

Please do a sanity check on the X & Y movement.  Assuming you have an alligator clip on the bit and a touch plate similar to the [triquetra](http://triquetra-cnc.com/) system, issue a G38.2 X100 F200 and then touch the clip to the plate.  

If the head doesn't stop then there's a software problem. 

If the head stops immediately then the problem is poor electrical contact on the side of the bit.

sergioausrio, 2017-07-07:
I tried it with a Alligator Clip direct on the Touchplate for many Times. It works only in both  Z Directions. The crazy Thing is: Marlin throws allways the "Error: failed to reach target" Message even if the Probe stops in Z Direction on contact.
 
The optional PROBE_DOUBLE_TOUCH does not work either with G38

Bob-the-Kuhn, 2017-07-07:
That's not what I was expecting to hear.

Time to jerry rig something so I can test.

sergioausrio, 2017-07-08:
This morning i give the latest 1.1.4 Bugfix a try.... and IT WORKS !!! G38.2 is full functional! All directions working. 

I put this in Configuration.h:
#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_PIN Z_MIN_PIN

Bob-the-Kuhn, 2017-07-08:
GREAT!!!

sergioausrio, 2017-07-10:
Is it normal that a G28Z0 does not work in this configuration ? I dont need a G28 for CNC but i tried it and the Z-Axis does not stop on contact. (G38.2, G29 and G30 woking fine)
On the other hand, the MPCNC is a multitool and can be used for 3D printing and Lasercutting. A G28Z is needed...

thinkyhead, 2017-07-10:
Does `G28 Z` work, but `G28Z` does not?

sergioausrio, 2017-07-10:
No. G28 works only with my regular Microswitch Endstops on X-Min & Y-Min but not on the Z-Axis with the Probe wich is on Z_MIN_PIN

Bob-the-Kuhn, 2017-07-10:
It's really strange that G28 Z doesn't work but the others do.

I think the easiest thing to try is to replace `#define Z_MIN_PROBE_PIN Z_MIN_PIN` with  `#define Z_MIN_PROBE_PIN xx` where xx is the number of the Z_MIN_PIN.  Sometimes the first method doesn't work.

I think the long term fix is to change the G38 code to use either the probe pin or the Z_MIN pin.  Try this:
a) change configuration.h
```
//#define Z_MIN_PROBE_ENDSTOP 
//#define Z_MIN_PROBE_PIN Z_MIN_PIN
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
```
b) in endstops.cpp
replace `#if ENABLED(G38_PROBE_TARGET) && PIN_EXISTS(Z_MIN_PROBE) && !(CORE_IS_XY || CORE_IS_XZ)`
with `#if ENABLED(G38_PROBE_TARGET) && (PIN_EXISTS(Z_MIN_PROBE) || PIN_EXISTS(Z_MIN)) && !(CORE_IS_XY || CORE_IS_XZ)`

thinkyhead, 2017-07-10:
> It's really strange that G28 Z doesn't work but the others do.

```cpp
  #define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
```

sergioausrio, 2017-07-11:
I changed 
```
#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 \
        && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
```

to

` #define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) || ENABLED(Z_MIN_PROBE_ENDSTOP))`

and G28Z (G38 too) works now with  Probe:
```
#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_PIN Z_MIN_PIN
```

commit 11c589c3 (2017-07-06): Fix compile issue with G38

Fixes #7250
closed 2017-08-28 by thinkyhead