MarlinFirmware/Marlin #7254: [FR] SERVO_DELAY should be an array of values

GMagician, 2017-07-06:
I don't know Marlin software very well so I may be on the wrong way, in this situation please forgive me.

If you enable BLTOUCH delay is overrided with a value of 50 that should good for it.
If you also have SWITCHING_EXTRUDER (my idea is something like 'thingiverse:673816') you need a delay of 1000.

I don't know if value in "configuration.h" is overidden by "conditionals_lcd.h" but if it's not overriden I think you will always wait 1000ms even when using BLtouch...thinking to a bed level with 9 points (MK2S Prusa) you have to wait 9000ms only for probe moving (or 18000 if up and down every probe)
and if it's overriden, almost sure of that, you never wait enough time to switch extruder.

Maybe using an array of constant values and in "servo.cpp"
void Servo::move(int value) {
  if (this->attach(0) >= 0) {
    this->write(value);
    delay(SERVO_DELAY_array[this->servoIndex]);
    #if ENABLED(DEACTIVATE_SERVOS_AFTER_MOVE)
      this->detach();
    #endif

could solve these issue without using too much memory

Bob-the-Kuhn, 2017-07-06:
The BLTOUCH_DELAY and the SERVO_DELAY do not interact.

BLTOUCH_DELAY is only used with the BLTouch.

SERVO_DELAY is used for all other servo based devices.

SERVO_DELAY defaults to 50mS unless defined otherwise within configuration.h.

GMagician, 2017-07-07:
Ok, used a wrong example, sorry.
What about 3dtouch? it really uses a servo

GMagician, 2017-07-07:
One more thing...inside "conditionals_lcd.h", when I have BLTOUCH, code redefines SERVO_DELAY to 50 (undef and define).
I really think this is done after compiling "configuration.h" (otherwise how can you know you have BLTOUCH?) then even when you have it you override SERVO_DELAY.

Bob-the-Kuhn, 2017-07-07:
Geeetech's 3DTouch is a clone of the BLTouch.  They're 100% interchangeable.  

Just specify BLTouch in the config file to use the 3DTouch.

----

You are correct about SERVO_DELAY and BLTouch.  When BLTouch is enabled then  SERVO_DELAY  is set to 50mS.

My first response was wrong.

I'll have to look at the other servo based devices before formulating a plan.

GMagician, 2017-07-17:
I checked the code...it seem more complicated than expected.
To use object I want, Dondolo,  I suppose I need to enable SWITCHING_NOZZLE.
Well activating this mode I have other 500ms delay since 'safe_delay' is called inside 'move_nozzle_servo', just after 'MOVE_SERVO' call, but here we have a potential issue I think.
This delay is after MOVE_SERVO (that is servo[].move) but what happens if I have 'DEACTIVATE_SERVOS_AFTER_MOVE'? this 500ms is done after detaching pin (detach is done after SERVO_DELAY inside 'move' method).
I really think that using an array is still the best solution. This way every servo can have its one delay before detaching it. This works with bltouch (it need 50ms) dondolo (it needs 1000, maybe 750) and for every other possible servo.

GMagician, 2017-08-10:
Hello, I would like to address this FR but I don't understand something in code.
In several places it's called "delay_save" after servo::move. What is the purpose of that call?
When 'DEACTIVATE_SERVOS_AFTER_MOVE' option is active these calls are done with servo detached (unpowered).
I think that using array for SERVO_DELAY all these calls can be removed.

I already tryed to implement array but it uses about 1030 bytes of program memory and 72 bytes for global variables. So strange for an array of 2 constant 16 bits variables and 1 access to it

Roxy-3D, 2017-08-10:
Can you post the changes you made?   Probably somebody can spot the reason for the extra RAM and Program Memory.   (Maybe even me!)

GMagician, 2017-08-10:
Modification is simple:
  Configuration.h
    '#define SERVO_DELAY { 50, 300 }' in place of original declaration
  servo.cpp
    const uint16_t servo_delay[MAX_SERVOS] = SERVO_DELAY;
    in Servo::move
       'delay(servo_delay[this->servoIndex]);' in place of original 'delay(SERVO_DELAY);'

my cnf is with BLtouch (servo 0) and switching extruder (servo 1)

Roxy-3D, 2017-08-10:
That alone doesn't seem like it should generate that much extra RAM and Program Memory usage.   Did both compiles start from a 'Clean' state?    If necessary a comparison of the the .ELF files will tell us what suddenly jumped in size.

GMagician, 2017-08-11:
I'm using Arduino IDE and I don't know how to start from a 'clean state' and I also don't find any .elf file..

fiveangle, 2017-08-11:
Max servos = 12 * [number of timers that exist on the board]

Roxy-3D, 2017-08-11:
Yeah!   I bet you are right.   

@GMagician    Are you using #define NUM_SERVOS 2 to control the size of the servo arrays?

GMagician, 2017-08-11:
Ok that right, I pasted wrong define... using MAX_SERVOS memory increase by 4 bytes and that right but memory increase 1070 bytes (I'm compiling now bugfix 1.1)

GMagician, 2017-08-19:
Merged

closed 2017-08-19 by GMagician