MarlinFirmware/Marlin #7262: Compile error in stepper.cpp with UBL on Printrboard RevF (AT90USB1286) w/o LCD

fiveangle, 2017-07-07:
[the template finally works - yay !]

### Compile Error

Previous compile error in #7015 appears resolved.

New error:

```
"/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-ar" rcs "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/core/core.a" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/core/yield.cpp.o"
Archiving built core (caching) in: /var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_cache_288060/core/core_teensy_avr_teensypp2_usb_serial,speed_16,keys_en-us_54b41c7cca20326e490399ec92de3489.a
Linking everything together...
"/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-gcc" -Os -Wl,--gc-sections,--relax  -mmcu=at90usb1286 -o "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/Marlin.ino.elf" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/G26_Mesh_Validation_Tool.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/I2CPositionEncoder.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/M100_Free_Mem_Chk.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/Marlin.ino.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/MarlinSerial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/Marlin_main.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/Sd2Card.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/SdBaseFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/SdFatUtil.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/SdFile.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/SdVolume.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/blinkm.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/cardreader.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/configuration_store.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/dac_mcp4728.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/digipot_mcp4018.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/digipot_mcp4451.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/endstops.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/gcode.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/hex_print_routines.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/least_squares_fit.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/mesh_bed_leveling.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/nozzle.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/pca9632.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/planner.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/planner_bezier.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/printcounter.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/qr_solve.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/serial.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/servo.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stepper.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stepper_dac.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stepper_indirection.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stopwatch.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/temperature.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/twibus.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/ubl.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/ubl_G29.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/ubl_motion.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/ultralcd.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/utility.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/vector_3.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/watchdog.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/libraries/Wire/Wire.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/libraries/Wire/WireKinetis.cpp.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/libraries/Wire/utility/twi.c.o" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/core/core.a" "-L/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324" -lm
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_65324/sketch/stepper.cpp.o: In function `Stepper::quick_stop()':
stepper.cpp:(.text._ZN7Stepper10quick_stopEv+0x2): undefined reference to `ubl_lcd_map_control'
collect2: error: ld returned 1 exit status
Using library Wire at version 1.0 in folder: /Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
Error compiling for board Teensy++ 2.0.
```

------------------------------------------------------------

https://github.com/fiveangle/Marlin/tree/ubl_compile_err

------------------------------------------------------------

fiveangle, 2017-07-07:
I think now that LCD is not a requirement for UBL, someone named Travis should be tasked with finding them instead of me ;)

Bob-the-Kuhn, 2017-07-08:
😉 Why should we waste Travis time when we have such a great resource in you?  

I have no clue as to why the Teensy IDE extension has these issues.  Definitely above my pay grade.

I got it to compile by making these changes:
stepper.cpp - changed line 66 to  `bool ubl_lcd_map_control = false;`
ultralcd.cpp - changed line 49 to  `extern bool ubl_lcd_map_control;`

No idea if this is functional or if it breaks something else.

Let me know if it is functional.  If it is I'll do a PR.

Tannoo, 2017-07-08:
In `stepper.cpp`, how about these changes instead?
```cpp
#if ENABLED(AUTO_BED_LEVELING_UBL) && ENABLED(ULTIPANEL) && DISABLED(DOGLCD)
  extern bool ubl_lcd_map_control;
#endif
```

```cpp
#if ENABLED(AUTO_BED_LEVELING_UBL) && ENABLED(ULTIPANEL) && DISABLED(DOGLCD)
    if (!ubl_lcd_map_control)
      cleaning_buffer_counter = 5000;
#else
     cleaning_buffer_counter = 5000;
#endif
   DISABLE_STEPPER_DRIVER_INTERRUPT();
   while (planner.blocks_queued()) planner.discard_current_block();
```

Tannoo, 2017-07-08:
`ubl_lcd_map_control` originates in `ultralcd.cpp`.

Roxy-3D, 2017-07-08:
Yeah but why do the 128KB boards keep having a problem with UBL without a LCD Panel?

Something weird is going on here....

Tannoo, 2017-07-08:
Very likely... didn't think to make it conditional for an lcd.

Tannoo, 2017-07-08:
Either ULTIPANEL or NEWPANEL.

Roxy-3D, 2017-07-08:
ULTIPANEL is probably the better choice.

fiveangle, 2017-07-08:
Why is every issue reported on an AT90USB board automatically blamed on the MCU ? :)

Tannoo, 2017-07-08:
>Switched to RAMPS and updated ticket

The cat's out of the bag already.

Tannoo, 2017-07-08:
Here's my PR to fix it: #7265

Roxy-3D, 2017-07-08:
> Why is every issue reported on an AT90USB board automatically blamed on the Teensy extensions ? :)
> Switched to RAMPS and updated ticket :-p

My apologies.  Developers do not like having stuff blamed on their code when their code had nothing to do with the problem!

fiveangle, 2017-07-08:
#7265 resolved

closed 2017-07-08 by fiveangle