MarlinFirmware/Marlin #7266: Can't Get BLTouch and Rambo 1.3L Board to Work

us682112, 2017-07-08:
All, I to am trying to get my BLTouch Smart working.  I have an UltiMachine Rambo 1.3L system board, using Marlin firmware Marlin-bugfix-1.1.x, downloaded 7-3-2017.  
My printer prints fine with this firmware using a ZMin mechanical end stop, and without BLTouch. 

I followed this diagram for wiring the BLTouch: https://plus.google.com/113792662927481823969/posts/RtWww5rCV3F. 

When I turn it on, the pin extends and retracts twice, and red led remains lit, but not seeing blue led.
Pictures of wiring and it mounted and on are here:  https://www.amazon.com/photos/share/uvYLL4m3rhL2sUWom9pZxvCK9N5OoGaHbLmVhWdxKGO

I’ve followed issues “BLTouch ignoring M-commands #7024” and “Point me to the right direction, i3 Prusa with BLTOUCH, how ? #7035”, but I’m not getting any reaction from the BLTouch except at power on.

My current config (Configuration.h, Configuration_adv.h, pins_RAMBO.h) and M43 I1 output is attached.

Other than custom settings for my printer, current settings are:

//Configuration.h
#define BLTOUCH
#if ENABLED(BLTOUCH)
  #define BLTOUCH_DELAY 375   // (ms) Enable and increase if needed
#endif

//  at the end 
#define Z_MIN_PROBE_PIN 10      // rbh

//Configuration_Adv.h
#define PINS_DEBUGGING

// pins_RAMBO.h
#define Z_MIN_PROBE_PIN Z_MIN_PIN     // rbh

For some reason I think my problem is with the servo connection / configuration, but can’t figure it.
Any help would be greatly appreciated.  Thanks.

[Marlin Config and M43 Output.zip](https://github.com/MarlinFirmware/Marlin/files/1133171/Marlin.Config.and.M43.Output.zip)

Bob-the-Kuhn, 2017-07-08:
The BLTouch recommended servo pin doesn't match the default servo pin definitions in pins_RAMBO.h.

There's two ways to fix it:
1) In the file "pins_RAMBO.h", change the line `#define SERVO0_PIN         22` to `#define SERVO0_PIN         44`
2) Move the orange wire two pins over so it's next to the brown pin.  

FYI - if you ever switch to VIKI2 or miniVIKI then you'll need to do something else.  These displays also use 22 and 44.

us682112, 2017-07-08:
Okay - I left SERVO0_PIN 22 and moved the orange over two.  It didn't work, but after re-tracing my patch cable and correcting my brown / orange swap - it works!  Blue led comes on, it deploys and stows.  On to calibration.  Thank you!

closed 2017-07-08 by Bob-the-Kuhn