MarlinFirmware/Marlin #7269: [BR] Marlin Bug: Invalid Move XY -> Z

Andrei-Pozolotin, 2017-07-10:
See: https://www.thingiverse.com/thing:2427868

1) When printing attached 170 mm x 170 mm frame,
printer makes first X and Y moves (see picture),
and after reaching upper right frame corner,
it starts lowering plate, moving positive Z,
until it crashes at the Z limit

2) When printing similar frame with 130 mm x 130 mm size
this problem does not happen

3) Firmware: https://github.com/random-builder/Marlin/tree/wanhao-d6-probe
changes are only in Configuration.h and Configuration_adv.h

4) Printer: https://www.monoprice.com/product?p_id=15710
is an Ultimaker 2 clone, and prints OK any objects
smaller then 130 mm x 130 mm in XY size

Bob-the-Kuhn, 2017-07-10:
Assuming you're using the config files from the firmware you pointed at ...

UBL thinks that the printable area is 165x165 because `UBL_MESH_INSET `is set to 35.  Try changing it to 25 and see if that fixes the problem.

I don't know how current the fork you're using is.  At one time UBL had problems when the nozzle was outside the printable area.

----

Here's the items that affect the printable area for UBL.

```
#define UBL_MESH_INSET 35

// Travel limits after homing (units are in mm)
  |#define X_MIN_POS 0
   #define Y_MIN_POS 0
   #define Z_MIN_POS 0
  #define X_MAX_POS 200
  #define Y_MAX_POS 200
  #define Z_MAX_POS 170

 #elif ENABLED(AUTO_BED_LEVELING_UBL)
   #define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
   #define UBL_MESH_MAX_X (X_MAX_POS - (UBL_MESH_INSET))
   #define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
   #define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET))

```

Andrei-Pozolotin, 2017-07-10:
Robert: thank you.

1) the fork above is current with `bugfix-1.1.x`
2) I went in 5 mm steps decrements for `UBL_MESH_INSET`
3) 170x170 frame starts to print OK at `UBL_MESH_INSET=10`
4) but with this inset Z inductive probe starts to react to edges and plate screws, UBL unusable
5) and finally even with `UBL_MESH_INSET=10`, the same problem arises now with 180x180 frame

so the solution could be:
a) force UBL use `UBL_MESH_?_?` sizes as measurement bounding box only
b) use bi-linear extrapolation from `UBL_MESH_?_?` box to whole bed area (200x200 in this case)

finally:
c) is this doable with current config options?
d) is there other leveling mode which is free of this bug?

Bob-the-Kuhn, 2017-07-10:
@Roxy-3D - as the god of UBL ...  we could use your guidance here.

I was mistaken on my offset calculation.  The offset is used on all 4 sides so an offset of 10 means a printable area of 180 x 180.

My suggestion is to set the printable area to whatever is correct for your printer, do the auto probing and then edit the mesh to correct the boundary issues.  After that do the G26 & fine tune process to dial in the mesh.

Have you looked at the [UBL startup guide](http://marlinfw.org/docs/features/unified_bed_leveling.html)?  It's well worth the time.

Roxy-3D, 2017-07-11:
Some how I missed this message today...   Or I would have responded...  Let me read this and see if there is anything I can add...

Andrei-Pozolotin, 2017-07-13:
@Roxy-3D

Roxanne: thank you for looking into this.

1) gcode is included in the frame thing: https://www.thingiverse.com/thing:2427868

2) I now set UBL_MESH_INSET=0 in https://github.com/random-builder/Marlin/tree/wanhao-d6-probe

3) octoprint terminal session for UBL leveling, which works with full size 200x200 frame:
```
Send: G28
Recv: echo:busy: processing
Recv: X:66.00 Y:113.00 Z:0.00 E:0.00 Count X:5283 Y:9044 Z:0
Recv: ok
[...]
Send: G29 P0
Recv: Mesh zeroed.
Recv: ok
[...]
Send: G29 P1
Recv: Default storage slot 0 selected.
Recv: Mesh invalidated. Probing mesh.
Recv: echo:busy: processing
Recv: ok
[...]
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,6)                                           (6,6)
Recv: (0,200)                                        (200,200)
Recv:      .          .          .          .          .          .          .
Recv: 
Recv:      .          .      -0.100   -0.095   -0.092   -0.127   -0.160
Recv: 
Recv:      .          .      -0.002   -0.002   -0.002   -0.070   -0.097
Recv: 
Recv:      .          .     [-0.002]  -0.002   -0.002   -0.062   -0.085
Recv: 
Recv:      .          .      -0.002   -0.002   -0.002   -0.072   -0.100
Recv: 
Recv:      .          .      -0.102   -0.075   -0.065   -0.092   -0.115
Recv: 
Recv:      .          .      -0.145   -0.002   -0.090   -0.117   -0.145
Recv: (0,0)                                            (200,0)
Recv: (0,0)                                             (6,0)
Recv: ok
[...]
Send: G28
Recv: echo:busy: processing
Recv: ok
Send: G42 I2 J0 F9000
Recv: ok
Send: G29 I1
Recv: Locations invalidated.
Recv: ok
Send: G42 I3 J0 F9000
Recv: ok
Send: G29 I1
Recv: Locations invalidated.
Recv: ok
Send: G42 I4 J0 F9000
Recv: ok
Send: G29 I1
Recv: Locations invalidated.
Recv: ok
Send: G42 I5 J0 F9000
Recv: ok
Send: G29 I1
Recv: Locations invalidated.
Recv: ok
Send: G42 I6 J0 F9000
Recv: ok
Send: G29 I1
Recv: Locations invalidated.
Recv: ok
[...]
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,6)                                           (6,6)
Recv: (0,200)                                        (200,200)
Recv:      .          .          .          .          .          .          .
Recv: 
Recv:      .          .      -0.100   -0.095   -0.092   -0.127   -0.160
Recv: 
Recv:      .          .      -0.002   -0.002   -0.002   -0.070   -0.097
Recv: 
Recv:      .          .      -0.002   -0.002   -0.002   -0.062   -0.085
Recv: 
Recv:      .          .      -0.002   -0.002   -0.002   -0.072   -0.100
Recv: 
Recv:      .          .      -0.102   -0.075   -0.065   -0.092   -0.115
Recv: 
Recv:      .          .          .          .          .          .     [    .   ]
Recv: (0,0)                                            (200,0)
Recv: (0,0)                                             (6,0)
Recv: ok
[...]
Send: G28
Recv: echo:busy: processing
Recv: X:66.00 Y:113.00 Z:0.00 E:0.00 Count X:5283 Y:9044 Z:0
Recv: ok
[...]
Send: G29 P3
Recv: ok
[...]
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,6)                                           (6,6)
Recv: (0,200)                                        (200,200)
Recv:  -0.100   -0.100   -0.100   -0.095   -0.092   -0.127   -0.160
Recv: 
Recv:  -0.100   -0.100   -0.100   -0.095   -0.092   -0.127   -0.160
Recv: 
Recv:  -0.002   -0.002   -0.002   -0.002   -0.002   -0.070   -0.097
Recv: 
Recv:  -0.002   -0.002  [-0.002]  -0.002   -0.002   -0.062   -0.085
Recv: 
Recv:  -0.002   -0.002   -0.002   -0.002   -0.002   -0.072   -0.100
Recv: 
Recv:  -0.102   -0.102   -0.102   -0.075   -0.065   -0.092   -0.115
Recv: 
Recv:  -0.102   -0.102   -0.102   -0.075   -0.065   -0.092   -0.115
Recv: (0,0)                                            (200,0)
Recv: (0,0)                                             (6,0)
Recv: ok
```
4) so essentially, I followed Robert's advice, to implement this sequence:
* measure leveling: G28; G29 P0; G29 P1
* invalidate border points:  G28; G42 I5 J0 F9000; G29 I1 ...
* execute "Smart Fill" to populate all missing points: G28; G29 P3

5) finally, conclusion: 

it would help if:

A) this leveling scenario would be more clearly documented, it seems a popular one;
for example, I was under impression that UBL_MESH_INSET is for measurement only,
and will not affect active print area

B) Marlin/UBL would fail more gracefully on "move outside UBL_MESH_INSET area" , 
like show error on LCD, plus stop, rather then triggering "Invalid Move XY -> Z" user experience

C) there is strange behavior of `G28; G42 I5 J0 F9000; G29 I1` : if you issue it on the same location
it will continue to invalidate more adjacent points, rather then treating additional invalidate as NOOP.

D) sometimes `G29 P3` has to be issued several times, in order to fill "all the dots"
I could not find any correlation as to why/when it works from the fist try

E) please advise people who want to maximize usable leveled printable area with UBL
to skip inductive sensors, in favor of those which provide 100% area coverage, such as FSR
https://github.com/JohnSL/FSR_Endstop

F) perhaps there is a way to print Topography Report in octoprint properly (no tabs?)

feel free to close this, thank you.

Roxy-3D, 2017-07-13:
> B) Marlin/UBL would fail more gracefully on "move outside UBL_MESH_INSET area" , like show error on LCD, plus stop, rather then triggering "Invalid Move XY -> Z" user experience

Yes...  

> C) there is strange behavior of G28; G42 I5 J0 F9000; G29 I1 : if you issue it on the same location  it will continue to invalidate more adjacent points, rather then treating additional invalidate as NOOP.

Actually...  That is by design.   From where ever the nozzle is, it will search out the closest point and invalidate it.   Or if you say I4, it will invalidate the closest 4.

D) sometimes G29 P3 has to be issued several times, in order to fill "all the dots"  I could not find any correlation as to why/when it works from the fist try

The G29 P3 fills a row at a time.   If you have a large offset, you may have multiple rows (or columns) that are unpopulated.   Going a row at a time lets the user have more control over how they fill points.   

F) perhaps there is a way to print Topography Report in octoprint properly (no tabs?)

The Topography map uses spaces and not tabs.   But the problem is the host programs are using proportional fonts so it is difficult to get every thing to line up nicely.

aviatorken89, 2017-07-30:
UBL mesh inset needs better documentation, I had it set to 1 and trying to print 300mm x 300mm square (my bed size) cause weird behavior and crashing of the Z nozzle, it would randomly either crash to the bed, or to the top of the leadscrew bracket range.

I doubt crashing is the intended behavior.

It's late so I don't have more info to post on this, I quickly set the print to 280x280 and it's running fine now, but I'm pretty sure UBL mesh inset is the issue.

aviatorken89, 2017-07-30:
Confirmed, my issue is exactly like the photo in the first post in this thread. I originally generated my mesh wih G26 P1 but at that time i had different value for my bed max range trying to figure out manual home position offsets.

What we need: Do interpolation if value requested for XY move is outside of original probe points from G26 instead of randomly crashing.

aviatorken89, 2017-07-30:
Set my mesh inset to 0 and redid G29 P1 and all is well and can print the entire bed surface now.

Roxy-3D, 2017-07-30:
You can't change the dimensions of your bed or the number of mesh points along an axis without re-generating the matrix.

aviatorken89, 2017-07-30:
Right, but the docs either need to be better and or the firmware needs improved to prevent crashing. I've must've had 8 crashes trying to figure out how offsets work, and having my nozzle off the bed to clean and prime the nozzle, then discovered the gem of manual home positions :)

Roxy-3D, 2017-07-30:
> Right, but the docs either need to be better and or the firmware needs improved to prevent crashing. 

The UBL firmware was written to be very conservative and protective of the user's machine.   But there are limits to what it can protect against.   In order to get best results, the nozzle needs to be allowed to go below Z=0.000    

There used to be a more robust sanity_check() for UBL.
```cpp
nt bed_leveling::sanity_check() {
  int j, k, error_flag = 0;

	if (this->state.n_x !=  MESH_NUM_X_POINTS)  {
	   SERIAL_PROTOCOLLNPGM("?MESH_NUM_X_POINTS set wrong\n");
	   error_flag++;
	}

	if (this->state.n_y !=  MESH_NUM_Y_POINTS)  {
	   SERIAL_PROTOCOLLNPGM("?MESH_NUM_Y_POINTS set wrong\n");
	   error_flag++;

	}

	k = E2END - sizeof( blm.state );
	j = (k - Unified_Bed_Leveling_EEPROM_start) / sizeof( z_values );

	if (j<1) {
	  SERIAL_PROTOCOLLNPGM("?No EEPROM storage available for a mesh of this size.\n");
	  error_flag++;
	}
	return error_flag;
}
```
Now it is
```cpp
  bool unified_bed_leveling::sanity_check() {
    uint8_t error_flag = 0;

    const int a = settings.calc_num_meshes();
    if (a < 1) {
      SERIAL_PROTOCOLLNPGM("?Insufficient EEPROM storage for a mesh of this size.");
      error_flag++;
    }

    return !!error_flag;
  }
```

Perhaps that should be put back.   If you make improvements to the code or the documentation, I'll merge it.

closed 2017-11-03 by thinkyhead