MarlinFirmware/Marlin #7270: Compilation error: Undefined reference to GCodeParser::parse when compiling with

marcio-ao, 2017-07-10:
### Compile Error
When I compile branch 1.1.x with the following changes to the Makefile:

```
HARDWARE_MOTHERBOARD ?= 301
ARDUINO_INSTALL_DIR  ?= ${HOME}/arduino-1.6.8
ARDUINO_VERSION      ?= 106
```

I get the following compile errors:

```
  CC    /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/wiring.c
  CC    /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/wiring_analog.c
  CC    /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/wiring_digital.c
  CC    /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/wiring_pulse.c
  CC    /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/wiring_shift.c
  CC    /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/WInterrupts.c
  CC    /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/hooks.c
  CXX   /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/WMath.cpp
  CXX   /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/WString.cpp
  CXX   /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/Print.cpp
  CXX   Marlin_main.cpp
  CXX   MarlinSerial.cpp
  CXX   Sd2Card.cpp
  CXX   SdBaseFile.cpp
  CXX   SdFatUtil.cpp
  CXX   SdFile.cpp
  CXX   SdVolume.cpp
  CXX   planner.cpp
  CXX   stepper.cpp
  CXX   temperature.cpp
  CXX   cardreader.cpp
  CXX   configuration_store.cpp
  CXX   watchdog.cpp
  CXX   /home/aleph/arduino-1.6.8/hardware/arduino/avr/libraries/SPI/src/SPI.cpp
  CXX   servo.cpp
  CXX   /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/Tone.cpp
  CXX   ultralcd.cpp
  CXX   digipot_mcp4451.cpp
  CXX   dac_mcp4728.cpp
  CXX   vector_3.cpp
  CXX   qr_solve.cpp
  CXX   endstops.cpp
  CXX   stopwatch.cpp
  CXX   utility.cpp
  CXX   printcounter.cpp
  CXX   nozzle.cpp
  CXX   serial.cpp
  CXX   /home/aleph/arduino-1.6.8/libraries/LiquidCrystal/src/LiquidCrystal.cpp
  CXX   /home/aleph/arduino-1.6.8/hardware/arduino/avr/cores/arduino/main.cpp
  CXX   applet/Marlin.elf
/tmp/ccwG6qQb.ltrans0.ltrans.o: In function `process_next_command()':
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x8c): undefined reference to `GCodeParser::parse(char*)'
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x92): undefined reference to `GCodeParser::command_letter'
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0xac): undefined reference to `GCodeParser::codenum'
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0xb0): undefined reference to `GCodeParser::codenum'
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x1074): undefined reference to `GCodeParser::codenum'
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x1078): undefined reference to `GCodeParser::codenum'
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x2ad4): undefined reference to `GCodeParser::codenum'
/tmp/ccwG6qQb.ltrans0.ltrans.o:ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x2ad8): more undefined references to `GCodeParser::codenum' follow
/tmp/ccwG6qQb.ltrans0.ltrans.o: In function `process_next_command()':
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x2b0c): undefined reference to `parser'
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x2b0e): undefined reference to `parser'
ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x2b10): undefined reference to `GCodeParser::unknown_command_error()'
/tmp/ccwG6qQb.ltrans1.ltrans.o: In function `GCodeParser::value_ulong() [clone .local.72]':
ccwG6qQb.ltrans1.o:(.text._ZN11GCodeParser11value_ulongEv.local.72+0x2): undefined reference to `GCodeParser::value_ptr'
ccwG6qQb.ltrans1.o:(.text._ZN11GCodeParser11value_ulongEv.local.72+0x6): undefined reference to `GCodeParser::value_ptr'
/tmp/ccwG6qQb.ltrans1.ltrans.o: In function `GCodeParser::value_long() [clone .local.73]':
ccwG6qQb.ltrans1.o:(.text._ZN11GCodeParser10value_longEv.local.73+0x2): undefined reference to `GCodeParser::value_ptr'
[... 10 lines cut ...]
ccwG6qQb.ltrans1.o:(.text._ZN11GCodeParser4seenEc.local.75+0x3c): undefined reference to `GCodeParser::command_ptr'
ccwG6qQb.ltrans1.o:(.text._ZN11GCodeParser4seenEc.local.75+0x44): undefined reference to `GCodeParser::value_ptr'
ccwG6qQb.ltrans1.o:(.text._ZN11GCodeParser4seenEc.local.75+0x48): undefined reference to `GCodeParser::value_ptr'
/tmp/ccwG6qQb.ltrans3.ltrans.o: In function `idle()':
ccwG6qQb.ltrans3.o:(.text._Z4idlev+0x18c): undefined reference to `GCodeParser::command_ptr'
ccwG6qQb.ltrans3.o:(.text._Z4idlev+0x190): undefined reference to `GCodeParser::command_ptr'
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(addsf3.o): In function `__addsf3':
(.text.avr-libc.fplib+0x8): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_round' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_round.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(addsf3x.o):(.text.avr-libc.fplib+0x0): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_pscA' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_pscA.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(addsf3x.o):(.text.avr-libc.fplib+0x4): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_pscB' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_pscB.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(addsf3x.o):(.text.avr-libc.fplib+0x10): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_nan' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_nan.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(addsf3x.o):(.text.avr-libc.fplib+0x18): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_inf' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_inf.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(addsf3x.o): In function `__addsf3x':
(.text.avr-libc.fplib+0x1c): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_split3' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_split3.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(addsf3x.o): In function `__addsf3x':
(.text.avr-libc.fplib+0x30): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_zero' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_zero.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(atan2.o):(.text.avr-libc.fplib+0x0): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_pscA' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_pscA.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(atan2.o):(.text.avr-libc.fplib+0xc): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_pscB' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_pscB.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(atan2.o):(.text.avr-libc.fplib+0x1a): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_nan' defined in .text.avr-libc.fplib section in /home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libc.a(fp_nan.o)
/home/aleph/arduino-1.6.8/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/lib/avr6/libm.a(atan2.o):(.text.avr-libc.fplib+0x1c): additional relocation overflows omitted from the output
collect2: error: ld returned 1 exit status
Makefile:489: recipe for target 'applet/Marlin.elf' failed
make: *** [applet/Marlin.elf] Error 1

```

jebba, 2017-07-10:
With Arduino 1.8.3 I can compile this fine on my workstation. On OP's workstation with 1.8.3 isn't compiling, so I believe this is a local workstation issue, not upstream problem.

marcio-ao, 2017-07-10:
It's compiling fine in the Arduino IDE. Not from the command line using "make".

marcio-ao, 2017-07-10:
You can close this is this is the expected behavior and compilation should only happen via the Arduino IDE.

fiveangle, 2017-07-10:
@marcio-ao - you should be able to close yourself.

marcio-ao, 2017-07-11:
@fiveangle: Okay. I meant to say, "you can close this *if* this is the expected behavior". I guess it is. I'll go ahead and close the issue.

closed 2017-07-11 by marcio-ao

fiveangle, 2017-07-12:
Sorry, I don't know about the make functionality and if it's working or not.  I would assume it should, so if you can help resolve, that would probably be a help.

But looking at the error messages more closely, I suspect you do not have FASTER_GCODE_PARSER enabled in Configuration_adv.h ?

marcio-ao, 2017-07-12:
@fiveangle: I do have FASTER_GCODE_PARSER in "Configuration_adv.h". My error message looks different than yours. From above:

ccwG6qQb.ltrans0.o:(.text._Z20process_next_commandv+0x8c): undefined reference to `GCodeParser::parse(char*)'

For one, I noticed gcode.cpp is not listed in Makefile anywhere. Should it be there?

Roxy-3D, 2017-07-12:
The GCode.cpp file is only 2 months old.   If the makefile has not been updated recently...  My guess is it needs to be updated to include that file.

marcio-ao, 2017-07-13:
For sure, gcode.cpp was missing. Once I put that in it got me closer. But I also had to add support for U8GLIB to the Makefile and now I am getting relocation errors:

  CXX   applet/Marlin.elf
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(addsf3.o): In function `__addsf3':
(.text.avr-libc.fplib+0x8): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_round' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_round.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(addsf3x.o):(.text.avr-libc.fplib+0x0): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_pscA' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_pscA.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(addsf3x.o):(.text.avr-libc.fplib+0x4): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_pscB' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_pscB.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(addsf3x.o):(.text.avr-libc.fplib+0x10): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_nan' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_nan.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(addsf3x.o):(.text.avr-libc.fplib+0x18): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_inf' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_inf.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(addsf3x.o): In function `__addsf3x':
(.text.avr-libc.fplib+0x1c): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_split3' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_split3.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(addsf3x.o): In function `__addsf3x':
(.text.avr-libc.fplib+0x30): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_zero' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_zero.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(atan2.o):(.text.avr-libc.fplib+0x0): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_pscA' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_pscA.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(atan2.o):(.text.avr-libc.fplib+0xc): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_pscB' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_pscB.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(atan2.o):(.text.avr-libc.fplib+0x1a): relocation truncated to fit: R_AVR_13_PCREL against symbol `__fp_nan' defined in .text.avr-libc.fplib section in /usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libc.a(fp_nan.o)
/usr/lib/gcc/avr/4.8.1/../../../avr/lib/avr6/libm.a(atan2.o):(.text.avr-libc.fplib+0x1c): additional relocation overflows omitted from the output
collect2: error: ld returned 1 exit status
Makefile:500: recipe for target 'applet/Marlin.elf' failed
make: *** [applet/Marlin.elf] Error 1

I've attached my modified Makefile. Any ideas?

[Makefile.txt](https://github.com/MarlinFirmware/Marlin/files/1145898/Makefile.txt)