MarlinFirmware/Marlin #7271: [FR] WS2812B leds (NeoPixels)?

NiklasOlofsson, 2017-07-10:
Hello, 

I know that Tanoo (spelling?) had support for the  ws2812b ledstrips and I used that fork with my leds mounted on one of my printers. I would like to use more ws2812b on my other printers so I am wondering. What happened to the efforts to support that kind of leds? Thanks!

Tannoo, 2017-07-12:
I know I said I would update the one I still use, I've just been busy. Marlin is changing at a very fast pace. It didn't make since to keep rebasing every few days to ensure that it works.

When things settle down a bit after 2.0(?) Is released, I will re-configure and rebase to that and submit another PR.

Tannoo, 2017-07-12:
What I do have current is in my `current_running` branch.

NiklasOlofsson, 2017-07-12:
I always used RCBugFix before but because of more frequent releases of stable version I only update when Marlin gets released. Good to hear that it is still being worked on. Thanks!

Tannoo, 2017-07-12:
So far, I haven't had to do any "work" to it. It works they way it is. And for most calls also. I did have to change how it was being handled during the stepper timeout, but I believe I have that resolved.

Tannoo, 2017-07-12:
So, currently it is just maintained on my printer for my printer.
Current_running does have other 'features' setup in it. This is a 'backup' of my printer's firmware.

Tannoo, 2017-07-12:
If you were to run that, you would have to reconfigure the config files for your printer... they are not defaulted like the marlin base.

RalphSch, 2017-07-25:
Take a look at PR7340 ;)

closed 2018-02-09 by thinkyhead