MarlinFirmware/Marlin #7273: Firmware Retract behaves erratically with UBL & changing retract settings on the fly can cause issues

FHeilmann, 2017-07-10:
Hi everyone, I'm back again with more FWRetract related bug reports:

Today I loaded up the latest bugfix-1.1.x and enabled, and set up UBL. With UBL activated, the following happens:

Everytime the machine retracts and unretracts, the nozzle ends up slightly higher above the bed. It seems that the Z-correction of UBL is somehow interfering with the FWRetract mechanism. Over the course of a print, the nozzle will move further and further away from the actual layer. 

The firmware does perform z-hop and z-unhop moves, however the unhop seems to only partially undo what the hop did. I verified that the interaction with UBL is the culprit by turning UBL off. The print then proceeds as expected.

Another Issue I have found:

If you change the retraction settings while the machine is printing, it can happen that the retract is performed with the old settings, while the unretract is performed with the new settings. Especially when modifying the z-hop, this can have catastrophic consequences, so we might have to look into caching the retraction parameters during retract moves and only accept updates if the machine is in the unretracted state.

Roxy-3D, 2017-07-11:
More likely...  when the retract stuff was put in place, some corner case was missed.    We need to make sure everything is happening for each bed leveling system in a symmetrical way.

FHeilmann, 2017-07-11:
@Roxy-3D sorry I didn't want to sound like I'm blaming UBL. I agree with what you're saying. Unfortunately my power supply went up in smoke yesterday so I can't test anything for now. But my best guess is the following causes problems:

- Retract
- Move somewhere, resulting in Z-corrections via the UBL mechanism
- Unretract

either the correction of UBL results in a case where the checks at [Marlin_main.cpp#L3092](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L3092) doesn't work properly, or the z-unhop somehow factors in the correction done by UBL 

e.g
- z-hop of 2mm
- move causes z-correction of .2mm
- z-unhop only undoes 1.8mm even though it should do 2mm

Roxy-3D, 2017-07-11:
> @Roxy-3D sorry I didn't want to sound like I'm blaming UBL. I agree with what you're saying. Unfortunately my power supply exploded yesterday so I can't test anything for now. But my best guess is the following causes problems:

No...  I get it...  Not a problem!    But even with a dead power supply, I have some ideas to suggest!  :)

I don't know who the big contributors to 'Firmware Retract' are.   I  should know, but I don't.   If you can look and call them into this thread it would be helpful.   Maybe go to the file in the Marlin source code directory and use 'Blame' to find who has been contributing big chunks of code to that topic.

FHeilmann, 2017-07-11:
Thats the thing, the one who suggested the last few edits to firmware retract was ... me 😝 

I think the problem might be this: When we retract, we [record the current z-position as `hop_height`(line 3082)](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L3082) and later [check the current z-position against that hop height (line 3093)](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L3093). Maybe, if any leveling has modified current_`position[Z_AXIS]` between the retract and unretract, the if statement in line 3093 evaluates to false, which prevents the z-unhop from happening.

I don't know enough about the variables that represent the actual position of the z-axis (which changes depending on where the nozzle is) and the logical position of zhe z-axis (i.e. 0 across the whole bed if only XY moves are made). But that would be my best guess.

I didn't see the problem with BILINEAR but that may simply be the case because with BILINEAR I used 7 points per axis whereas with UBL I use 10.

Roxy-3D, 2017-07-11:
OK...  When I get a few minutes of time...  I'll use that information to see if I can find something 'out of balance'.

FHeilmann, 2017-07-12:
Looking at #7198 and what @thinkyhead said, a rewrite of FWRETRACT may be a good idea. The way the z tracking is implemented now is elegant, but it is hard to catch all the corner cases, especially when leveling is involved.

My gunshot approach would be to remove the whole if/lower/then/blah and simply add some intelligence to G0/G1 so we can unhop based on whether the z-axis was actually moved via g-code, which would remove all the issues with a modified z-position due to leveling.

studiodyne, 2017-07-12:
> If you change the retraction settings while the machine is printing, it can happen that the retract is performed with the old settings, while the unretract is performed with the new settings. [...]

FHeilmann, 2017-07-12:
> 4-) Not a g29 mesh problem because i don't use it and i have this issues

for me, turning off leveling (M420 S0) fixes the problem, so for my "incarnation" of the problem activated bed leveling is the issue.

Roxy-3D, 2017-07-12:
> My gunshot approach would be to remove the whole if/lower/then/blah and simply add some intelligence to G0/G1 so we can unhop based on whether the z-axis was actually moved via g-code, which would remove all the issues with a modified z-position due to leveling.

Is the hop and unhop decision based on a Z-Height corrected or uncorrected number.  I think there might be a 'corner case' with Fade Height from what you are saying.   The fade gets to be less and less so is it possible the hop and unhop are not symmetrical?  

It would be interesting to hear if leaving UBL active, but with no fade height causes the problem...  Or does not cause the problem...    And a similar case...  Leave UBL active, but zero the mesh with a G29 P0.  Does that cause or not cause the problem???

FHeilmann, 2017-07-12:
> Is the hop and unhop decision based on a Z-Height corrected or uncorrected number. I think there might be a 'corner case' with Fade Height from what you are saying. The fade gets to be less and less so is it possible the hop and unhop are not symmetrical?

its based on `current_position[Z_AXIS]. Like I said I don't know enough about the marlin motion stuff to know if this is a corrected or an uncorrected value.

> It would be interesting to hear if leaving UBL active, but with no fade height causes the problem... Or does not cause the problem... And a similar case... Leave UBL active, but zero the mesh with a G29 P0. Does that cause or not cause the problem???

I should have my new power supply in the mail tomorrow. I hope I'll be able to run some tests and get back to you ASAP.

Roxy-3D, 2017-07-12:
Well...  The fade height is saved in the EEPROM.   It is what ever it was saved as...

The current_position[] is just that.   And if bed leveling is active, the current position is a result of what ever position was asked for...  Plus the correction for that location.   So it maybe fade height is entering into the calculations.

FHeilmann, 2017-07-13:
UBL on, non-zero mesh, no fade-height => problem persists
UBL on, **zeroed mesh**, no fade-height => problem does **not** persist

So it seems whenever UBL applys corrections FWRetract goes into derp mode

Roxy-3D, 2017-07-13:
OK.   That is helpful information!

studiodyne, 2017-07-13:
You can verify it to be sure of this .  When auto leveling , level you z axis to middle , and with your finger , touch the sensor , and create a big mesh curve , 5 cm on the edges and 0 on center , with this mesh , you will verify ,if it's a little error of something wrong , of it's REALLY the mesh value , and you will be sure !

It's true , the point of ZLIFITING is the currentvalue of a positionned Z height , if out of this position , the mesh must be applied with the new x y position 

With an ultra big mesh : , you will see immediatly if the origin of the z issues is equal to the mesh amplitude

Try this and you will confirm no other possibility

oysteinkrog, 2017-07-14:
Maybe related to https://github.com/MarlinFirmware/Marlin/issues/7198?

FHeilmann, 2017-07-27:
Tried out the latest and greatest by @thinkyhead over at #7308 For tests, I set a large Z-hop, and I can confirm that the firmware attempts to undo the Z-hop, it just doesn't unhop the entire distance.

FHeilmann, 2018-01-10:
As this issue is now almost half a year old and the code has changed significantly, I'm closing this for now.

closed 2018-01-10 by FHeilmann