MarlinFirmware/Marlin #7274: Save 40 bytes of RAM memory...

Roxy-3D, 2017-07-10:
In ultralcd.cpp we have the declaration:
```cpp
uint8_t lcd_status_message_level;
char lcd_status_message[3 * (LCD_WIDTH) + 1] = WELCOME_MSG; // worst case is kana with up to 3*LCD_WIDTH+1
```
It would seem...   if the user doesn't have  STATUS_MESSAGE_SCROLLING enabled...  And they are using a western type font we can save 40 bytes of RAM by shrinking the array size.    Kana fonts need the 3x size.   Does anybody know if the Greek and Cyrillic need double byte characters?

thinkyhead, 2018-02-09:
See if #9545 has it covered.

boelle, 2019-02-19:
@Roxy-3D did #9545 cover it?

boelle, 2019-03-12:
@thinkyhead i think we can close this one

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

Roxy-3D, 2019-06-22:
Yes...   This is now covered.   And with a good solution.   People with multi-byte fonts can still use them.   And it is only these people that get nicked for the extra memory.

boelle, 2019-06-22:
@Roxy-3D so maybe close the  issue? if nothing else then just for the sake of not having "fixed" issues on the list :-D

closed 2019-07-11 by thinkyhead