MarlinFirmware/Marlin #7277: Problem with Anet after 1.1.4 update

flodejr, 2017-07-11:
After configuring the 1.1.4 configuration.h and configuration_adv.h files exactly like those 1.1.3 (minus those in the standard support)  The Z-axis fail to zero properly after calibrations and M500 eeprom saves.  The Z-axis will revert to the original position after G28 or a board reset.

Further the BLtouch will not give the correct offsets.  After there any specific configurations required for the 1.1.4 release?

Zezoh, 2017-07-12:
I have the same issue, probe location not changed even after playing with offset values I have delta printer with BLtouch and marlin 1.1.4. 

Any idea ??

flodejr, 2017-07-20:
Ok, i tried updating to marlin 1.1.4 again on my Anet A8 V1.0 board.  I went through the configuration line by line and adjusted according to my needs.  After uploading using platformio, it destroyed my bootloader (probably overwrote it) and then after that when I do a G30 X110 Y110 which is what I normally use to adjust the Z-offset, it will always give me an incorrect value.  I checked the correct value in Skynet 2.4.5 (Marlin 1.1.3).

I went to check the code and the difference between the z_probe function in 1.1.3 and 1.1.4 is that in 1.1.3 it simply returns the z-axis raw probed value whereas in 1.1.4 it returns the z-axis value modified by HOME_OFFSETs and WORKSPACE_OFFSETS.   I have searched for the configuration for the HOME_OFFSETs and WORKSPACE_OFFSETS and it seems that it is related to some settings for the Delta Printers.  How can I disable that HOME_OFFSETs and WORKSPACE_OFFSETs to get back the same values as in 1.1.3?  Attached are my configuration.h and configuration_adv.h files.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1161226/config.zip)

RalphSch, 2017-07-22:
If flashing Marlin kills your bootloader then you are using the wrong hardware definitions/extensions (there is a bug with the maximum code size in some Skynet3D packages). The most recent one can be downloaded here: https://github.com/SkyNet3D/anet-board

You can disable these offsets in Configuration_adv.h:
/**
 * Enable this option for a leaner build of Marlin that removes all
 * workspace offsets, simplifying coordinate transformations, leveling, etc.
 *
 *  - M206 and M428 are disabled.
 *  - G92 will revert to its behavior from Marlin 1.0.
 */
#define NO_WORKSPACE_OFFSETS

fiveangle, 2017-07-25:
review PROGMEM usage guidelines here to see what you can disable to get Marlin to fit:

https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/progmem_usage_comments/Marlin/Configuration_adv.h

flodejr, 2017-07-25:
After compile, the sizes fit, but the bootloader is still being overwritten, I don't think there is a problem with the size of the compile.

Device: atmega1284p

Program:  121520 bytes (92.7% Full)
(.text + .data + .bootloader)

Data:       4363 bytes (26.6% Full)
(.data + .bss + .noinit)

RalphSch, 2017-07-25:
"Program: 121520 bytes (92.7% Full)" means that your maximum size is about 131090 bytes ... but you only have 126976 bytes of space when using the avr bootloader (upload speed 57000bps), 130048 bytes of space when using the optibootloader (uploadspeed 115200bps) and 131072 bytes of space without an bootloader. If this happens in ArduinoIDE then your are not using the recent version of the anet hardware extension from https://github.com/SkyNet3D/anet-board (maximum size was fixed on April, 6th. https://github.com/SkyNet3D/anet-board/commit/1607254ca16bcc018f7e968a26c62ad6e11059ee).

closed 2018-02-09 by thinkyhead