MarlinFirmware/Marlin #7278: AUTO_BED_LEVELING_BILINEAR - layer(s) left to right titled

Lord-Quake, 2017-07-11:
My layer(s) always seem to be tilted on the build plate. With tilt I mean that on the left side the nozzle seems to be too high and the right side too low compared to the middle. It doesn't seem to matter how big the printed object is. 
In the example shown I printed the first layer of a rectangle 200 x 200 mm and as you can see the left is too high / right too low. The nozzle seems to gradually lower itself while printing from left to right. Top / Bottom doesn't have the issue. I'm using Marlin V1.1.4.

[image]

What can I do to better analyze the problem?

BillyBob54, 2017-07-11:
I cannot explain why it is happening, but I can say that the same thing has happened to me using UBL, most recently with 1.1.2.  After the initial probing (G29 P1) and filling in the un-probed areas (G29 P3), when I ran G26 to check the leveling, the results were as you described above; poor/no adhesion on the left side, too close on the right side with a pretty linear trend in between (middle of plate was just about perfect).  Front to back was pretty consistent, but the front left had almost no adhesion were the back left would at least hold the pattern.  While I was trying to figure out the whole UBL process, I must have run G29 P1 10 or more times using anywhere from a 10X10 to 5X5 matrix and the results were the same every time; it didnt seem to matter if the heaters were on or off during probing.  I thought then that it was weird that the probe inaccuracies were so consistent from one side to the other time after time.  I was able to fairly easily adjust all the points (G29 P4) so that I got good adhesion across the bed, so never really went back to figure it out.  As I recall, the bed itself turned out to be pretty flat.

BillyBob54, 2017-07-11:
Oh, by the way - I have a BLTouch probe installed

benlye, 2017-07-11:
I had a similar problem. My probe Y offset was out by 2mm.

BillyBob54, 2017-07-11:
Hmmmmm - I'll double check that tonight but thought I looked at that pretty closely when I started.

fiveangle, 2017-07-11:
Would bed levelling debug be of any use here ?  I had a similar issue but worse that it gouged by bed, showing it clearly allowed travel below the Z nozzle offset (below min software endstops).  This means that either the probe during homing got an erroneous result (which makes no sense since the probe repeatability is dead on within +/-0.08mm worst case (typically +/- 0.03).  It's a stock Printrbot Simple Metal inductive probe.  First time this has ever happened since 2014 through stock PB fork 3-point linear, Marlin 1.1 RC7 MBL without any issues.  2 days of trying to get BILINEAR working on 1.1.3+bugfix1.1 and bad gouge happened.  It was in fact identical gcode to that which ran the night before without gouging the bed.  Normally it would point to HW but I can't find anything wrong with it.

benlye, 2017-07-11:
> Would bed levelling debug be of any use here ? I had a similar issue but worse that it gouged by bed, showing it clearly allowed travel below the Z nozzle offset (below min software endstops).

@fiveangle, were you using endstop interrupts, by chance?  There were a few reports of people having endstops trigger but the printer failing to register it when using endstop interrupts, myself included.  I don't think it was ever fully investigated.  I think you were in the discussion in #6807?

fiveangle, 2017-07-11:
Nope: Printrboard has one axis on non-interrupt capable pin so not using it.  I already considered that (if it were tied to the other issue Roxy brought up (forgot # now)

fiveangle, 2017-07-11:
@Roxy-3D - There couldn't possibly be any issue with the least-squares fit method vs the qr-solve that it replaced in these newer schemes (and now in LINEAR) ?  There's a ghost in the machine somewhere it seems there relatively limited places it can be.  So peculiar...

Roxy-3D, 2017-07-11:
I put in code to print the 'solutions' from both the QR_Solve code and the new least squares fit code and compared them.   Other than the numbers being negated, they were identical.  And you can see in the Pull Request the negation to get the numbers identical.     Bi-Linear does not use the Least Squares Fit...  So even if that was the cause, it would not explain the Bi-Linear impact.

> I cannot explain why it is happening, but I can say that the same thing has happened to me using UBL, most recently with 1.1.2. After the initial probing (G29 P1) and filling in the un-probed areas (G29 P3), 

And BillyBob is seeing this with UBL.  UBL is where the new Least Squares Fit code was developed.  That is the most tested and 'correct' implementation of it.

Roxy-3D, 2017-07-11:
@Lord-Quake   Is it possible to turn the current up on your stepper motor driver chips?   And can we turn the max feed rate and acceleration down a little bit (temporarily) in your configuration.h file?    I'm wondering if the stepper motors are losing position because they can't move fast enough.  It would be good to rule out this possibility.

fiveangle, 2017-07-11:
> Bi-Linear does not use the Least Squares Fit... So even if that was the cause, it would not explain the Bi-Linear impact

I was not aware of that.  Thanks for clarifying !  Yes, very peculiar.

Roxy-3D, 2017-07-11:
Bi-Linear is a mesh based leveling system.  The term Bi-Linear comes from the way the printer approximates point.  If it is trying to move the nozzle into the middle of a mesh cell, what it does is a Bi-Linear interpolation.   First, it interpolates how far it the nozzle is moving into the mesh cell along the Y axis.   It has the Z-height for all four corners of the mesh cell.  So, next it calculates the Z-height of each of the Y lines (where the X location is).    It then uses the amount of X the destination is into the mesh cell using those previously calculated Z-Heights along the Y border of the mesh cell.

Hence...   It is doing a Bi-linear interpolation.  First in one direction.  Then 90 degrees to that first calculation.

fiveangle, 2017-07-12:
So it does no overall plane offset vector calc (like in 3-point or LINEAR, to potentially reduce the overall offsets at each probe point before begining the interpolation), but instead just runs the bilinear on the absolute probe offsets observed.  Got it.  No wonder it's relatively efficient but still decently effective.

Roxy-3D, 2017-07-12:
Does Linear bed leveling fit into your 128KB board now that it is 7.7 KB leaner?

fiveangle, 2017-07-12:
Surely it does, since BILINEAR does, but LINEAR doesn't really work well my bed as it's only 150mm square (so I always struggle to fit parts onto the space) and the upper right corner of the MIC6 bed is definitely pushing their +/- 0.3xx mm flattest specs so I need levelling that can follow a contour.  MBL works, but I just feel wrong having the bed countour travel through the entire print (my OCD bleeding through ;)

Roxy-3D, 2017-07-12:
> MBL works, but I just feel wrong having the bed countour travel through the entire print (my OCD bleeding through ;)

Actually...  That is almost trivial to change.  I have never used MBL.  But I used it as the foundation for UBL.  I know exactly how it hooks in and what it thinks about things.   Seriously....   We can add F-ade to MBL in a couple hours of work.   Do you want to give it a try?   If so, we should probably start with a fresh fork of bugfix-1.1.x just so we can merge it when we think it is 'right'.

Lord-Quake, 2017-07-12:
@Roxy-3D My board does not have the ability to change stepper motor current with the only exception of the extruder stepper.

fiveangle, 2017-07-12:
> We can add F-ade to MBL in a couple hours of work. Do you want to give it a try? If so, we should probably start with a fresh fork of bugfix-1.1.x just so we can merge it when we think it is 'right'.

I'm game !

Roxy-3D, 2017-07-12:
> My board does not have the ability to change stepper motor current with the only exception of the extruder stepper.

OK.  But you can still back off the max feedrate numbers and the acceleration number by 25%.

Roxy-3D, 2017-07-12:
I'm game ! https://github.com/fiveangle/Marlin/tree/mbl_zfade :)

I was getting set up to help you do this...  But it looks like Fade is already implemented for Mesh Bed Leveling:
```cpp
#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT
#endif
```

Lord-Quake, 2017-07-12:
@Roxy-3D are you referring to these?:
```
#define DEFAULT_MAX_FEEDRATE          { 400, 400, 8, 50 }
#define DEFAULT_MAX_ACCELERATION      { 2000, 2000, 100, 10000 }
```
The above is what I'm using. What do you want me to use for the test?

Roxy-3D, 2017-07-12:
Yes.   Cut those to 75% of their current values and lets see if it still shifts.  If it does...  we probably have a firmware issue.

Lord-Quake, 2017-07-12:
I've done the test with the following settings:
```
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 6, 36 }
#define DEFAULT_MAX_ACCELERATION      { 1500, 1500, 75, 7500 }
```
The results are the same. Using a mechanical caliper I measure 0.35mm left and 0.25mm right for the first layer so I'm seeing a 0.1mm difference.

Side note:
`//  #define ENABLE_LEVELING_FADE_HEIGHT`
is deactivated.

jeno007, 2017-07-12:
Just for curiosity, what is the status of the

#define MIN_SOFTWARE_ENDSTOPS

?

I experienced the same and after redoing the bed leveling, (it was really bad) and commenting out MIN_SOFTWARE_ENSTOPS, it seems to be working OK.

Lord-Quake, 2017-07-13:
@jeno007 Status is
`//#define MIN_SOFTWARE_ENDSTOPS`

Lord-Quake, 2017-07-14:
Is there anything further I can do to debug the situation?

cisforcer, 2017-07-19:
I had a similar problem.   
https://user-images.githubusercontent.com/30031297/28353484-8f59f862-6c8e-11e7-9b24-e0bc699c6b2f.jpg

BillyBob54, 2017-07-19:
The above diagram got me thinking about an issue that I had as I was initially setting up UBL that may (or may not) have something to do with the original post in this thread:
I do not recall where the below quote came from; As I was researching setup/use of UBL, I compiled a set of notes combining the several sources of information for ready reference and I came across the following (I do not recall where the below quote came”:

“One concept that is important to remember and that will make using the Phase 4 command easy to use is this: You are editing the Mesh Points. If you have too little clearance and not much plastic was extruded in an area, you want to LOWER the Mesh Point at the location. If you did not get good adhesion, you want to RAISE the Mesh Point at that location.”

As I interpret the above, during the P4 correction routine if the nozzle was too close to the build plate at a given mesh point, and the mesh offset was say 0.1, to correct you would LOWER the mesh to .05 (just examples).  If there was not enough adhesion, you would RAISE the mesh to 0.15.  This in practice only made things worse. After several failed attempts, I realized that if I did the opposite, I could dial the mesh in in just a couple of passes.  So,  2 questions:
1.	Is the above quoted instruction accurate and is my interpretation of it correct?
2.	If the instruction and interpretation is correct, is it possible somehow the mesh correction is being applied “backwards”.  Everything else concerning the Z axis operates correctly, baby stepping, Z offset, etc).  If the bed had a general left-to-right tilt, an inverse application of mesh correction would result in to tight clearance on one side, to high on the other.  If the tilt was too great, bed collision would result. If fairly flat, you might not notice it at all.  To be honest, I am not sure what the general geometry of my bed was before I started; final mesh did indicate that it was fairly level but there might have been a general tilt.  

I have 1.1.4 all set up and ready to load, but just have not had time to play with it (maybe this weekend)

I would just like to say that the work everyone has put into UBL (and all the other methods and Marlin capabilities) is incredible.  Thanks for all the great work and support!

BillyBob54, 2017-07-19:
Self answered question one above:
@Roxy-3D; in #6399 , you write
On April 23rd you write:
“But... The way the system works is you can now move the nozzle over an area of the bed that is not 'perfect' and issue a G29 P4 R. You can dial up or down each mesh point. As a rule of thumb... If the line is too squished down and too thin... Bring the mesh point up. Don't go crazy... but depending on what you see, bring it up .1 or .2 mm. If the line is just barely adhering to the bed, bring the mesh point down .1mm. If the line is not adhearing at all, bring it down .25mm.”

On the 24th;
“After you do a G26 command, the mesh validation pattern will be on the bed. Different areas may be less than 'perfect'. Move the nozzle to the center of the area that is flawed. Do a G29 P4 R command. The nozzle will go to the closest mesh point that has not been edited yet. You can turn the encoder wheel to adjust what should happen at that mesh point. If you have bad adheasion at the mesh point, bring the nozzle down a little bit. If the filament is squished too thin at the mesh point, bring it up 'the right' amount”

The above 2 comments almost seem to contradict each other, but on the 25th, it appears that you clarify the above two comments:
“The mesh numbers say what to do to the nozzle to make it track the bed. So if the nozzle is too close to the bed... What ever the number is currently, make it bigger. Make it more positive. Make it less negative.
If the nozzle is too far from the bed... bring the mesh point number down. make the mesh point number smaller. make it more negative.
If you have any adhesion at all... don't move the number by more than .15 mm .15mm will cause a visible difference in how the lines at that point appear. You want the mesh lines to be flat on the top (because they are being squished into the bed) and a little wider than the nozzle's extrusion hole.”

Roxy-3D, 2017-07-19:
Yes...   It is easy to make a mistake with this stuff.   I said the wrong thing.    The mesh point value is added to the Z-Height so the nozzle will track the bed.   If you have a dip in your bed, the mesh point value will be negative, and adding that value to the Z-Height will pull the nozzle down at that location so the nozzle can track the bed.

If you can print a G26 pattern...    Pick a random mesh point and add 1 or 2 mm to it.  Make it positive number.   And when you print the validation pattern, you will the nozzle lift up in that area and not print those lines.

cisforcer, 2017-07-19:
@BillyBob54    sorry for my poor English first， did you mean above that  i should change the

cisforcer, 2017-07-19:
@BillyBob54    sorry for my poor English first， did you mean above that  i should change the values of mesh points that test by g29？  
if need change values manually  what g29 meaning？ auto bed level  ？

cisforcer, 2017-07-19:
@roxy-3d 
how about mine？ should i do g26 too？ and change the values of  mesh  points? How  to
 change  those  values?

Lord-Quake, 2017-07-19:
@Roxy-3D After reading your last post I'm not sure anymore if my specific problem is being understood.
In my case the auto bed leveling is working when you have different heights from one point to the other.
I tested that by placing a metal ruler in the middle of the bed. While printing the height will change accordingly.
The problem "we" are seeing is that within the mesh the movement in height is not constant.
As mentioned in my first post it seems that the layer on the left is being printed higher compared to the right.
What can I do to provide more debugging evidence?

BillyBob54, 2017-07-19:
@roxy-3D ; no problem - it actually all makes sense once you have the correct perspective and all the 3 of the comments above are really equivalent.  Thanks again for all the effort.
@cisforcer - not to worry on the language issue; I speak English natively and I have a hard time understanding this stuff!! If I understand correctly, you are asking why, if G29 P1 generates the correction to account for bed tilt/imperfections, why does any correction have to be applied.  G29 P1 generates the mesh that the print head must follow to remain an equal distance from the bed.  The results of P1 can be seen by using the G26 routine- it will print a pattern on the bed one layer thick. This mesh will generally not be perfect and must be corrected (mostly due to probe inaccuracies or humps/bumps between probing points).  This is done with the G29 P4 routine.  The nozzle will move to the closest mesh point, if adhesion of the G26 pattern is too tight or to loose, then correct as per the above post discussions.  There is a lot more to this process than I just laid out- Roxy may be able to point you to better instructions.

cisforcer, 2017-07-21:
@BillyBob54 🤘 yes  you got me ， then how to change the values？ what  commands used to changing the values of mash points 
 can i do  the values changing   once for a long times？

cisforcer, 2017-07-21:
i have a Plastic thinsection for cold printing，its very good first  printing 。may be the plastic make the 12mm pnp  ccuracy error？

cisforcer, 2017-07-21:
Aluminum plate on bottom

mrjoda66, 2017-07-21:
[image]

Same here with UBL, Bilinear and Linear leveling...

Roxy-3D, 2017-07-21:
> The problem "we" are seeing is that within the mesh the movement in height is not constant.  As mentioned in my first post it seems that the layer on the left is being printed higher compared to the right.  What can I do to provide more debugging evidence?

Is it possible to capture the problem on video?   And then you can tell us what to watch for and we can see the issue.

> if G29 P1 generates the correction to account for bed tilt/imperfections, why does any correction have to be applied. G29 P1 generates the mesh that the print head must follow to remain an equal [...]

If your probe have very good repeatability, it may be the G26 to check and do a final edit of the mesh is not necessary.   On my gMax printer there is a BL-Touch probe.  I switched to one of the most recent versions of the probe and magically...  I no longer need to do a G26 to perform a final edit.    The probing is that good.   But even so...  The G26 only takes 5 minutes to do.  You may as well confirm the mesh numbers you are going to lock in and use are perfect.

Roxy-3D, 2017-07-21:
> i have a Plastic thinsection for cold printing，its very good first printing 。may be the plastic make the 12mm pnp ccuracy error？

That is the purpose of the G29 J command.   If you have your print bed's mesh accurately defined, but it doesn't always hold position...    You can tilt that mesh to reflect the current situation.   The easiest way to envision a need for this is if you remove the glass to put it in the freezer so the parts release from it.   And when you put it back on the heated bed, you some times get a little piece of plastic under a corner of the glass.   The G29 J will tilt the mesh to compensate for that.

BillyBob54, 2017-07-21:
@cisforcer; see below link for more detail on the process 
http://marlinfw.org/docs/features/unified_bed_leveling.html

Before going there, you may want to go your Marlin folder and open up G26_Mesh_Validation_Tool.cpp.  The list of avaialbe commands for  G26 is listed there.  Also open ubl_G29.cpp ; options fopr the G29 command are provided there as well.

Basicall, after you perform a P1, then G26 to print the test pattern, G29 P4 R will visit all the targets that were printed by G26.  If there is not enough adhesion, make the cooresponding mesh point (as displayed on the LCD) smaller (or more negative), if to close, make the number bigger (or less negative) by rotating the LCD knob- Press the knob one time to move to the next point. See my conversation with Roxy above.  

All that said, we have hijacked this thread- the original question concerned a general left-right tilt aafter  the initial leveling routine.  My appolgies to the OP

Lord-Quake, 2017-07-21:
@BillyBob54 No problem what so ever. Learning a lot!

Lord-Quake, 2017-07-21:
Here another test that shows ABL is working as it should. I tilted the bed by several degrees and then did a 5x5 mesh calibration.
[image]
I then did the same print as in my first post with the same results. The left side is printing too high and the right side too low (x-axis). The layers in y-axis are evenly printed on the bed, no tilt changes.
[image]

Roxy-3D, 2017-07-21:
> Here another test that shows ABL is working as it should. I tilted the bed by several degrees and then did a 5x5 mesh calibration.

And you don't need to save the new mesh...  But if you do a G29 P4 R with the nozzle in the lower right of that picture....   And lower ever point in that corner by .1 mm      And....  redo that same print....   you will see the filament get smooshed together better.

Lord-Quake, 2017-07-21:
I'm not using glass on my aluminum heat bed but only a very thin layer of Kapton film.
I've never used G29 J4. Should I still use it with Kapton? What does the code do?

Roxy-3D, 2017-07-21:
It would have probed the bed and tilted the mesh the 'correct' amount to make your pictures up above be 'normal'.

thinkyhead, 2017-07-22:
If you want to debug `ABL_BILINEAR` at various points around the bed, you can enable `DEBUG_LEVELING`, then use `M111 S32` to enable extra logging, and then `G29 Q` will report the status of bed leveling, including the amount of Z adjustment at the current point.

Move around to different points across the bed and do `G29 Q` to see if the Z adjustment is sensible and matches the grid points that were originally measured and saved by `G29`.

Lord-Quake, 2017-07-22:
After doing my tilt heat bed test I mounted the heat bed back to it's normal position doing a manual 4 point adjustment with extruder nozzle. After that I heated the bed to 55°C and did a 5x5 auto level calibration which I saved to eeprom (M500) for later prints. Here is another test with the same results.
[image]
I can confirm:
- Carriage isn't getting pulled upwards by the filament.
- Cables are not interfering with extruder movement.
- X carriage idlers have free movement on the guide rails.

cisforcer, 2017-07-23:
@Lord-Quake     may be you can do a manual probe ，@agambini    said that  works

cisforcer, 2017-07-23:
version 010100 cant  go with g26 🤦🏻‍♂️

cisforcer, 2017-07-23:
[image]

Lord-Quake, 2017-07-23:
In my opinion doing a manual probe leveling will not help in solving the problem. The print will only be as good as how well the user manually sets up the build plate. Never the less I did it anyway.
After doing the manual setup I had to recalibrate the 5x5 matrix because the build plate had now changed again.
Both results are shown in the picture.
[image]

cisforcer, 2017-07-23:
yes，auto bed leveling  is a great work to saving times，but whats the problem we are
?  
1.1.4 fw result 
[image][image]

cisforcer, 2017-07-23:
nozzel  low left,high right

agambini, 2017-07-23:
The question about the manual probe or auto probe is : Why one works and another not, basically , a mesh is created in manual probe and the printer obey the distances but not in auto mode. I'm not a programmer , and my opinion is very superficial , but looks here that mesh created in auto mode is wrong stored or ignored in print process 
Is this possible ?

agambini, 2017-07-23:
Sorry for my english

cisforcer, 2017-07-23:
i agree with you ， now i cant use g29 before printing and my i3is out of working.  how this happen ？  i dont think  ok with  any manual changing values.  Auto  leveling  by
Machine  is what  i need . Sorry  for  English  too🤦🏻‍♂️

cisforcer, 2017-07-24:
[image][image] big pic upload failed 
。
i do nothing with my printer but  after g29 p1 and g26   its printing nicer then days before with using UBL  
 there  still has a little different  between left and right corner ，how to change values  by g29 j command？

Roxy-3D, 2017-07-24:
> how to change values by g29 j command？

Don't use the G29 J right now...  We want the mesh accurate before we add any complexity.   And once the mesh is accurate, it is very likely you want need the G29 J because you are printing straight on the bed.

> i do nothing with my printer but after g29 p1 and g26 its printing nicer then days before with using UBL there still has a little different between left and right corner 

So...   Move the nozzle to that corner that is not perfect.   Give it a G29 P4 R command.   I'm making some assumptions about your machine.  But lower the corner by .1 mm and lower 3 nearest points by .05mm.       Do another G26 and you should see that area have much better adhesion.     And using that information...   Go tweak other areas of the mesh that are not absolutely perfect.

It should only take 2 or 3 iterations to get the mesh perfect.   (You can edit multiple problem areas on each G26 pattern before repeating the process.)

[76 more replies]

closed 2017-11-28 by thinkyhead