MarlinFirmware/Marlin #7279: Who is using the Viki-2 LCD Panel on RAMPS ?

Roxy-3D, 2017-07-11:
There is a report over at 3D-Print-Board that Viki-2's need a change in the RAMPS pin file to work.   Specifically, the report is here: http://3dprintboard.com/showthread.php?30897-Marlin-1-1-4-w-Viki2-LCD-on-RAMPS&p=110045&viewfull=1#post110045

Is anybody here using a Viki-2 LCD Panel with RAMPS here?  If so, did you have to change the pins to get it working?  

If we have a problem with the RAMPS file, we want to get it fixed!

Bob-the-Kuhn, 2017-07-12:
Confirmed - the pins_RAMPS.h file must be modified per the Panucatt instructions to use the VIKI2 display.

I tried to see if the "bad" definitions fit an alternate interface scheme but couldn't find a match.  I think the definitions came out of an earlier version of RAMPS.  The definitions use logical pins 7 & 22 which are not available on RAMPS1.4  Pin 7 is available on RAMPS1.3.

Roxy-3D, 2017-07-12:
Should we just grab the Viki-2 pin mapping in the file attached to this post?  http://3dprintboard.com/showthread.php?30897-Marlin-1-1-4-w-Viki2-LCD-on-RAMPS&p=110072&viewfull=1#post110072

Is that safe to do?   I presume it is better than what we have right now?

TGMods, 2017-07-12:
@Roxy-3D While I don't specifically use a RAMPS 1.4 at the moment, I do use an Azteeg X3 Pro on my D-Bot. I have just checked pins_AZTEEG_X3_PRO.h, and pins_RAMPS.h against the VIKI2 wiring diagram and the default pins in pins_RAMPS.h are correct in this case.
It would seem that all the pin files would have to be looked at when making any changes to pins_RAMPS.h.

On the other hand I do have a Re-ARM, a RAMPS and a miniVIKI sitting waiting for Marlin 32bit. So it would be great to get the VIKI2 RAMPS pins sorted out as the miniVIKI uses the same pinout according to pins_RAMPS.h.

Roxy-3D, 2017-07-12:
> I have just checked pins_AZTEEG_X3_PRO.h, and pins_RAMPS.h against the VIKI2 wiring diagram and the default pins in pins_RAMPS.h are correct in this case.

Is it possible the Viki-2 wiring diagram is wrong?  That might explain why we have different numbers depending upon who you are talking to?    I don't know where the 'bad information' is.  But it sure seems like we have some 'bad information' in the mix.

TGMods, 2017-07-12:
> Is it possible the Viki-2 wiring diagram is wrong?

If you look at the wiring diagram the pins numbers are different between the Azteeg 3 Pro and the RAMPS. One thing I noticed is that the diagram shows the VIKI2 connecting directly to the AUX-3/4 pins instead of the usual display adapter.

Bob-the-Kuhn, 2017-07-13:
@TGMods - are you sure the unmodified pins_RAMPS.h file matches the Viki2 1.2 diagram?

The pins_RAMPS.h file uses logical pins 7 and 22.  Those pins are not available on RAMPS1.4.  

The pin definitions for RAMPS on the Vii2 1.2 diagram match the 1.2 graphic.

TGMods, 2017-07-13:
@Bob-the-Kuhn If you look back to my original post you will read that I'm not using a RAMPS 1.4 at the moment, in fact I'm using an Azteeg X3 Pro. The pins_RAMPS.h file is included by the pins_AZTEEG_X3_PRO.h
`
#define BOARD_NAME "Azteeg X3 Pro"

#include "pins_RAMPS.h"
`
In this specific configuration the pins defined in the RAMPS file are correct, as any pins that need redefining are done in the Azteeg file. As I wrote earlier any changes to the pins_RAMPS.h file are likely to have effects in other pins files and not just the RAMPS file.

That being said I am interested in the outcome as I am hoping to upgrade my HyperCube to use a Re-ARM for RAMPS as I have bought a miniVIKI specifically for this purpose.

Bob-the-Kuhn, 2017-07-13:
Now I'm understanding what you're saying.  😊

Things get complicated quickly when thinking of making changes to RAMPs.

thinkyhead, 2018-02-09:
I did some work for a client involving hooking up a VIKI2 display to a RAMPS clone. The pinouts I ended up using are now documented in `pins_MKS_13.h`.

closed 2018-02-09 by thinkyhead