MarlinFirmware/Marlin #7282: M32 appears to read wrong file and crashes

lutorm, 2017-07-13:
### Bug Report
- Description: I'm trying to use M32 in a file to repeatedly print the same file at different locations on the bed by changing home offsets, calling M32, changing home offsets, etc., 9 times.
- Expected behaviour: It should execute the subprogram 9 times.
- Actual behaviour: It executes the first two times correctly, then the serial console output indicates that it attempts to load some other file which doesn't exist, and after a few seconds Marlin reboots.

I'm testing this with a file "makesquare.gcode" (makesq~1.gco) containing a square movement:
```
G1 X0 Y0
G1 X10 Y0
G1 X10 Y10
G1 X0 Y10
G1 X0 Y0
```
This file is then supposed to be executed by a runner "9squares.gcode" containing:
```
M206 X-25 Y-25 Z-10
M32 P !makesq~1.gco#
M206** Y-140
M32 P !makesq~1.gco#
M206  Y-255
M32 P !makesq~1.gco#
<and 6 more in a 3x3 grid>
```

When I execute this file from the LCD, the serial console prints:
```
echo:enqueueing "M23 9squar~1.gco"
echo:enqueueing "M24"
echo:Now fresh file: 9squar~1.gco
File opened: 9squar~1.gco Size: 580
File selected
X:-44.00 Y:233.00 Z:0.00 E:0.00 Count X:-4422 Y:23417 Z:0
echo:SUBROUTINE CALL target:"makesq~1.gco" parent:"/9SQUAR~1.GCO" pos58
File opened: makesq~1.gco Size: 71
File selected
Done printing file
echo:Now fresh file: /9SQUAR~1.GCO
File opened: 9SQUAR~1.GCO Size: 580
File selected
echo:busy: processing
X:0.00 Y:-115.00 Z:0.00 E:0.00 Count X:0 Y:-11558 Z:0
echo:SUBROUTINE CALL target:"makesq~1.gco" parent:"/9SQUAR~1.GCO" pos123
File opened: makesq~1.gco Size: 71
File selected
Done printing file
echo:Now fresh file: /9SQUAR~1.GCO
File opened: 9SQUAR~1.GCO Size: 580
File selected
echo:enqueueing "M23 auto-1.g"
echo:enqueueing "M24"
echo:Now doing file: auto-1.g
open failed, File: auto-1.g.
Error:SD read error
Error:SD read error
Error:SD read error
<etc etc until Marlin reboots>
```
Note the "enqueing M23 auto-1.g", which comes out of the blue and is a nonexistent file. (There exist files "auto-l~1.gco" and "auto-l~2.gco" on the SD card, though, which seems suspicously close.)

This is with version 1.1.4 running on a Lulzbot TAZ6.
[Configuration.h.gz](https://github.com/MarlinFirmware/Marlin/files/1144242/Configuration.h.gz)
[Configuration_adv.h.gz](https://github.com/MarlinFirmware/Marlin/files/1144245/Configuration_adv.h.gz)

thinkyhead, 2018-02-09:
Should be patched-up. Please confirm that this issue is fixed in the `bugfix-1.1.x` branch.

boelle, 2019-02-19:
@lutorm is the problem gone with latest bugfix 2.0 ?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-03-16:
@lutorm i tried to see if i could replicate this error, works fine for me with the attached test files
[test files.zip](https://github.com/MarlinFirmware/Marlin/files/2974515/test.files.zip)

closed 2019-03-17 by p3p

lutorm, 2019-03-17:
Sorry, I'm not in a position to try an updated firmware atm.