MarlinFirmware/Marlin #7283: Feature request - configurable pause before probing

DjDemonD, 2017-07-13:
Hi all,

I now market and sell an open-source piezo Z-probe. More details can be found online at reprap forum, just see the general forum main page, I won't leave any spam here.

What we are interested in is having a configurable pause before a probing dive takes place. RRF already has this as DC42 and T3P3 are making a nozzle contact probe based on piezo like we do, and the M558 command now has an R (recovery) parameter which is pause (in seconds) after positioning over a probing point before diving to probe. A value of 0.2sec is usually enough. It allows mechanical noise to settle before probing an reduces/eliminates false triggers from this very sensitive probe.

These things are becoming quite popular now as they are by far the most convenient and accurate z-probe available.

Would you consider implementing it for the benefit of the now hundred of people using piezo z probes? I know you don't use M558 so it would have to be an option in configuration_adv, "recovery time before probing move" or something like that?

Kind regards
DjDemonD/Simon
Precisionpiezo.co.uk

Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

teemuatlut, 2017-07-13:
Would you be able to make it into a pull request?
The added feature itself sounds simple enough but as none of the Marlin developers have the module, it would be difficult to test it out before pushing it online. Or would you be willing to send out a test unit?

DjDemonD, 2017-07-13:
I can make it into a pull request.

I can ship a test unit out to someone to try.

teemuatlut, 2017-07-13:
I'd be happy to help. It seems like an interesting approach to implementing a z-probe. I can also make sure it works with the TRAMS platform as well.

DjDemonD, 2017-07-13:
Okay PM? me your details and I'll sort something out for you.

teemuatlut, 2017-07-13:
GitHub removed private messages a few years back.
Can I instead email you at precisionpiezo (at) yahoo.com ?

DjDemonD, 2017-07-13:
yes thats fine

FHeilmann, 2017-07-13:
Marlin already has facilities that can be used for this via the `PROBING_FANS_OFF` and `PROBING_HEATERS_OFF` options. I am going to forego the M558 G-code here since all you need is a delay that can be adjusted via the configuration:

In configuration_adv.h specify your delay in miliseconds, i.e.:

```C
#define PAUSE_BEFORE_PROBING
#ifdef PAUSE_BEFORE_PROBING
  #define PAUSE_BEFORE_PROBING_DELAY 200
#endif
```

In Marlin_main.cpp [@2043](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L2043)
change:
```C
  #if QUIET_PROBING
    void probing_pause(const bool p) {
      #if ENABLED(PROBING_HEATERS_OFF)
        thermalManager.pause(p);
      #endif
      #if ENABLED(PROBING_FANS_OFF)
        fans_pause(p);
      #endif
      if (p) safe_delay(25);
    }
  #endif // QUIET_PROBING
```
to:

```C
  #if QUIET_PROBING
    void probing_pause(const bool p) {
      #if ENABLED(PROBING_HEATERS_OFF)
        thermalManager.pause(p);
      #endif
      #if ENABLED(PROBING_FANS_OFF)
        fans_pause(p);
      #endif
      #if ENABLED(PAUSE_BEFORE_PROBING)
        if (p) safe_delay(PAUSE_BEFORE_PROBING_DELAY);
      #else
        if (p) safe_delay(25);
      #endif
    }
  #endif // QUIET_PROBING
```
and, to enable the whole block, in [Conditionals_post.h@655](https://github.com/MarlinFirmware/Marlin/blob/e68b7afc52792ef7e97dbefc19c82d5b4902744e/Marlin/Conditionals_post.h#L655)
change 

```C
#define QUIET_PROBING (HAS_BED_PROBE && (ENABLED(PROBING_HEATERS_OFF) || ENABLED(PROBING_FANS_OFF)))
```
to
```C
#define QUIET_PROBING (HAS_BED_PROBE && (ENABLED(PROBING_HEATERS_OFF) || ENABLED(PROBING_FANS_OFF) || ENABLED(PAUSE_BEFORE_PROBING)))
```

that should do it. However I have just typed this out hastily and there may be a bug or two to fix, but you get the idea.

DjDemonD, 2017-07-13:
Thank you that's very helpful, no further action required then.

FHeilmann, 2017-07-13:
You're welcome, also, nice effort, this would probably work a lot better than my current nozzle microswitch design so I'll be watching out for your results :)

FHeilmann, 2017-07-13:
If you need any more testers let me know, I'd be eager to put one these on my CoreXY too.

DjDemonD, 2017-07-13:
We are past the testing stage with Piezo module and piezo kits, but we have some new ideas in the pipeline and some coding help might be needed.

closed 2018-02-09 by thinkyhead

inventabuild, 2019-12-08:
Sorry, but the above does not work to delay the reading of the Piezo Board information on the RAMBO Board.  I tried numbers as high as "#define DELAY_BEFORE_PROBING 10000" and as soon as my hot end finishes homing x and y and as soon as it starts homing z the Piezo sensor triggers the Piezo Board's blue light and z min is triggered causing the hot end to rise immediately.

If the delay was working my hot end should travel down for a designated number of seconds (or wait for designated number of seconds before traveling down) before allowing z min to trigger.  Neither of those is happening.  Marlin zip file attached.
[Printer_2_Marlin-src_1.1.9.34.zip](https://github.com/MarlinFirmware/Marlin/files/3935886/Printer_2_Marlin-src_1.1.9.34.zip)

inventabuild, 2019-12-08:
...unless DELAY_BEFORE_PROBING was designed to only work after G29, but if the Piezo sensor system is also used for G28 then a DELAY should also be implemented for G28.  Is there a programmable delay for G28 as well?

inventabuild, 2019-12-08:
In parting because it's after midnight here, I'll mention I think it would be better to have a delay that allows the hot end to start traveling down for x millisec before accepting input from the probe.  I noticed that after running G28 X0Y0 from pronterface, pausing then running G28 Z0, the probe still triggers right away.

However if, after running G28 X0Y0, I run G91, G1 +Z10, G1 -Z10, wait for the moves to finish then run G29 I can get the probe to tap the bed.  So I am able to get a noise free z movement down, it's just starting the z movement that can have noise in it causing z min to trigger prematurely.

DjDemonD, 2019-12-08:
Hi. Not a marlin dev, but I am one of the originators of the piezo probe. If the level of noise in your machine is such that the probe triggers despite being set (on the piezo pcb) for a lower sensitivity, and despite using temporary settings to set jerk to 0, and reduce acceleration to a very low setting, then delay before probing even after the z movement begins, isn't going to help. It was mainly to reduce the effect of ringing from xy movement before commencing a probing dive. With temporarily low acceleration and jerk, the inertia of the probing dive shouldn't trigger the probe. If the mechanics of your printer are this noisy then, start there. I have an unberbed setup so sensitive, I can blow on the bed and trigger it, but with low jerk/accel it probes normally.