MarlinFirmware/Marlin #7288: Hotend Z offset not working while printing (SWITCHING_NOZZLE / SWITCHING EXTRUDER )

HD78, 2017-07-13:
Hotend Z offset not working while printing.

I have a setup with Dondolo switching extruder, defined as SWITCHING_NOZZLE. SWITCHING EXTRUDER has same behavior. 2nd extruder is slightly above 1st one, as seen below.

`#define SWITCHING_NOZZLE`
`#if ENABLED(SWITCHING_NOZZLE)`
`#define SWITCHING_NOZZLE_SERVO_NR 0`
`#define SWITCHING_NOZZLE_SERVO_ANGLES { 10, 120 } // Angles for E0, E1`
`#define HOTEND_OFFSET_Z { 0.0, 0.7 }`
`#endif`

And following:

`#define HOTEND_OFFSET_X {0.0, 10.70} // (in mm) for each extruder, offset of the hotend on the X axis`
`#define HOTEND_OFFSET_Y {0.0, -0.80}  // (in mm) for each extruder, offset of the hotend on the Y axis`

Full config files:
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1146046/Configs.zip)

When I test with commands T0 / T1, it works perfect.
But whe it starts to print, `#define HOTEND_OFFSET_Z { 0.0, 0.7 }` suddenly loses function. I tried, just to see what it does, `#define HOTEND_OFFSET_Z { 0.0, -9.0 }` so the 2nd extruder should be raised 9mm. It does do that on toolchange. But immediately it goes back to where it was. X/Y offsets work perfect.

One thing I noted when trying T0 and T1 was that I do have movement on all axis, as configured. But Z is the only one that changes coordinates on my LCD, by amount of 0.7 as configured. So that seems to be the problem, absolute coordinates are not compensated from the shift.

HD78, 2017-07-15:
Seems that it has been working after this:
https://github.com/MarlinFirmware/Marlin/pull/6021
But broken here again:
https://github.com/MarlinFirmware/Marlin/pull/6042

If I try to implement it just like in #6021 I end up going up incrementally as I do toolchanges.

I have found this setup to be almost there:

`          #if ENABLED(SWITCHING_NOZZLE)`
`            #define DONT_SWITCH (SWITCHING_EXTRUDER_SERVO_NR == SWITCHING_NOZZLE_SERVO_NR)`
`            // <0 if the new nozzle is higher, >0 if lower. A bigger raise when lower.`
`            const float z_diff = hotend_offset[Z_AXIS][active_extruder] - hotend_offset[Z_AXIS][tmp_extruder],`
`                        z_raise = 0.3 + (z_diff > 0.0 ? z_diff : 0.0);`

`            // Always raise by some amount (destination copied from current_position earlier)`
**`            float save_Z = current_position[Z_AXIS];  // save Z for later on`**
`            current_position[Z_AXIS] += z_raise;`
`            planner.buffer_line_kinematic(current_position, planner.max_feedrate_mm_s[Z_AXIS], active_extruder);`
`            move_nozzle_servo(tmp_extruder);`
`          #endif`

`        #if ENABLED(SWITCHING_NOZZLE)`
`          // Move back down, if needed. (Including when the new tool is higher.)`
`          if (z_raise != z_diff) {`
`            destination[Z_AXIS] += z_diff;`
`            feedrate_mm_s = planner.max_feedrate_mm_s[Z_AXIS];`
`            prepare_move_to_destination();`
`          }`
**`          current_position[Z_AXIS] = save_Z;  // restore original Z position so the 'Move to the "old position"' is correct`**
`        #endif`

That way, the Z coordinate stay the same on LCD. So far so good. When I start with E0 and switch to E1, everything goes perfect. Both at the exact same place. Then Back to E0. Nozzle moves to the exact right position. But after that, it will raise, perhaps that 0.3mm. So E0 is now too high.
After that, I can switch them as many times I want, E1 ends up on correct place and E0 moves to correct place and then lifts..

Does anybody have any ideas how that could be avoided? Anyone else having these problems?

fixoid, 2017-07-17:
Hello!
Tested SWITCHING_NOZZLE on 1.1.4 version just right now with this EEPROM settings: 
M218 T1 X32.8 Y-0.25 Z-0.1
works properly.

HD78, 2017-07-18:
I am at 1.1.4 also but for me, the X offset works perfect but Z offset doesn't. Well, it does work, if you just test switching extruder.

On extruder switch the hotend offset is applied but next command to move (the next line in gcode basically) it moves back due to the actual Z height isn't compensated.
When I look at LCD, for example I'm at 14mm. Do the extruder switch, LCD shows 13.4mm. So obviously next time there is gcode telling to move coordinates Xxx Yxx Z14 the hotend offset is made useless.
Machine should think it's at 14mm even though it's been lowered by 0.6mm but no. And because gcode uses absolute, not relative coordinates, it doesn't work as is.

fixoid, 2017-07-18:
I mean, I printed test model with two filaments, with switching_nozzle feature and 0.1 z offset - it's okay.

HD78, 2017-07-18:
Interesting. I can't make it work no matter what. I don't think there's much to configure either, it was quite simple to set up. Except correct height..

I've been trying to figure out what happens with the firmware but my coding skills were somewhat good 20 years ago.. I can't pinpoint where it goes wrong.

HD78, 2017-07-19:
Just out of interest, since I've been relying on firmware configuration, I tried to send `M218 T1 X10.7 Y-0.8 Z0.7` and then print. Nothing. Still perfect on extruder switch but at start of print, T1 is too high..

Does anybody know how can I manipulate the machine's Z coordinate without actually moving? So I could compensate the shift? I know now how I can take a record of the position before the extruder switch but I can't push it back without moving.

fixoid, 2017-07-19:
I reasseble my mount and tested one more time. So, its realy brocked..

fixoid, 2017-07-19:
My servo is dead right now and I can not check it properly, but maybe the problem exists.
But im pretty sure that in 1.1.2 all was ok.

HD78, 2017-07-20:
Tried that just now. But the same thing exists in 1.1.2 also. It did that on 1.1.0 too. But that didn't have switching_nozzle so the configuration was quite custom anyways, so don't know if it was that.

fixoid, 2017-07-20:
on 1.1.0 all was ok?
Behavior of the feature was moved from switching_extruder to switching_noozle in 1.1.2 without any changes in Z offset compensation.

HD78, 2017-07-20:
Yes, that 1.1.0 and switching_extruder had the same problem with z offset. That was actually the reason to move to 1.1.4 to hopefully get this fixed. And other thing good with switching_nozzle, distinct_e_factors works and it's easier to configure.

HD78, 2017-07-20:
Doesn't this do it for XY?

`          // The newly-selected extruder XY is actually at...`
`          current_position[X_AXIS] += xydiff[X_AXIS];`
`          current_position[Y_AXIS] += xydiff[Y_AXIS];`
`          #if HAS_WORKSPACE_OFFSET || ENABLED(DUAL_X_CARRIAGE)`
`            for (uint8_t i = X_AXIS; i <= Y_AXIS; i++) {`
`              #if HAS_POSITION_SHIFT`
`                position_shift[i] += xydiff[i];`
`              #endif`
`              update_software_endstops((AxisEnum)i);`
`            }`

And earlied xydiff is defined like this:
`            const float xydiff[2] = {`
`              hotend_offset[X_AXIS][tmp_extruder] - hotend_offset[X_AXIS][active_extruder],`
`              hotend_offset[Y_AXIS][tmp_extruder] - hotend_offset[Y_AXIS][active_extruder]`
`            };`

I can't see similar things done to z in this toolchange-part?

btpride, 2017-09-18:
I'm having the same problem.
l had taken coding myself.
I didn't solve this problem.
Did you solve this problem?
if you have some solution, please halp me.

HD78, 2017-09-18:
No, sorry, never could be able to solve it in the firmware.

I ended up adding a nut to nozzle throat and adjusting it mechanically. It was nerve wrecking task to do it but the only possibility at the moment..

Warbonic, 2017-11-10:
Has anyone solved this yet? Seems silly that the #define HOTEND_OFFSET_Z would just get cancelled out.

hal971, 2017-11-20:
Hallo,
I had the same problem with my printer, a modified Vellemann Vertex with a dondolo-like printhead of my design. I have replaced this code, in marlin_main.cpp:
```
#if ENABLED(SWITCHING_NOZZLE)
          // Move back down, if needed. (Including when the new tool is higher.)
          if (z_raise != z_diff) {
            destination[Z_AXIS] += z_diff;
            feedrate_mm_s = planner.max_feedrate_mm_s[Z_AXIS];
            prepare_move_to_destination();
          }
        #endif
```

with the following:
```
#if ENABLED(SWITCHING_NOZZLE)
          // Move back down, if needed. (Including when the new tool is higher.)
		current_position[Z_AXIS] -= z_diff ;
		planner.set_z_position_mm(current_position[Z_AXIS]);
		feedrate_mm_s = planner.max_feedrate_mm_s[Z_AXIS];
                 prepare_move_to_destination();		
        #endif
```
and the problem seems to be gone, at least on my printer.

Warbonic, 2017-11-20:
Thank you @hal971 
I'll implement that when I get home and see what happens. 
This is for a new printer design as well so the designs have slightly changed given that this function doesn't work properly in Marlin. But if this works then you may have saved me a lot of time re-designing!

Warbonic, 2017-11-24:
@hal971 I tested the code replacement that you supplied and it worked... but it ended up removing the actual Z increase for each layer. 
The Z axis lowered to compensate for the difference between the two nozzles as intended, which is great, and in tern something else has been cancelled out so that now each layer does not change the Z height - leaving the print to maintain the same Z height for the entire print. 

Any idea on this?

hal971, 2017-11-24:
@Warbonic Do you mean that the z axis doesn't move between layers and all the print is performed on the layer 0?

Warbonic, 2017-11-25:
@hal971 yes they do stay on layer 0 for the entire print. Any idea why this would happen, given that the code replacement was the only thing that I had changed?

hal971, 2017-11-25:
@Warbonic I'm printing with the modified firmware and the z-axis moves normally on my printer. Maybe differences between firmware version? I'm using version 1.1.3.

Warbonic, 2017-11-29:
@hal971 I'm using the current full release. Which looks like 1.1.6
Not sure if there are big differences in our issue here between the versions, but there may be. I'll see if I can try 1.1.3 as well.

thinkyhead, 2018-02-09:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch in case the issue is already fixed. If we already got it then you can close the issue. If you still see the bad behavior then we can investigate further.

HD78, 2018-02-12:
I have my machine occupied on a project right now. But I will have a test session at some point, maybe next weekend.

StrannikZet, 2018-02-14:
Hi! I corrected the code, that's how it works for me. Marlin 1.1.8. Try it.
***************************************************************************
#if ENABLED(SWITCHING_NOZZLE)
            #define DONT_SWITCH (SWITCHING_EXTRUDER_SERVO_NR == SWITCHING_NOZZLE_SERVO_NR)
            // <0 if the new nozzle is higher, >0 if lower. A bigger raise when lower.
            const float z_diff = hotend_offset[Z_AXIS][active_extruder] - hotend_offset[Z_AXIS][tmp_extruder];//,
                        //z_raise = 0.3 + (z_diff > 0.0 ? z_diff : 0.0);
            // Always raise by some amount (destination copied from current_position earlier)
            current_position[Z_AXIS] += z_diff;//z_raise;
            //planner.buffer_line_kinematic(current_position, planner.max_feedrate_mm_s[Z_AXIS], active_extruder);
            move_nozzle_servo(tmp_extruder);
          #endif

          /**
           * Set current_position to the position of the new nozzle.
           * Offsets are based on linear distance, so we need to get
           * the resulting position in coordinate space.
           *
           * - With grid or 3-point leveling, offset XYZ by a tilted vector
           * - With mesh leveling, update Z for the new position
           * - Otherwise, just use the raw linear distance
           *
           * Software endstops are altered here too. Consider a case where:
           *   E0 at X=0 ... E1 at X=10
           * When we switch to E1 now X=10, but E1 can't move left.
           * To express this we apply the change in XY to the software endstops.
           * E1 can move farther right than E0, so the right limit is extended.
           *
           * Note that we don't adjust the Z software endstops. Why not?
           * Consider a case where Z=0 (here) and switching to E1 makes Z=1
           * because the bed is 1mm lower at the new position. As long as
           * the first nozzle is out of the way, the carriage should be
           * allowed to move 1mm lower. This technically "breaks" the
           * Z software endstop. But this is technically correct (and
           * there is no viable alternative).
           */
          #if ABL_PLANAR
            // Offset extruder, make sure to apply the bed level rotation matrix
            vector_3 tmp_offset_vec = vector_3(hotend_offset[X_AXIS][tmp_extruder],
                                               hotend_offset[Y_AXIS][tmp_extruder],
                                               0),
                     act_offset_vec = vector_3(hotend_offset[X_AXIS][active_extruder],
                                               hotend_offset[Y_AXIS][active_extruder],
                                               0),
                     offset_vec = tmp_offset_vec - act_offset_vec;

            #if ENABLED(DEBUG_LEVELING_FEATURE)
              if (DEBUGGING(LEVELING)) {
                tmp_offset_vec.debug(PSTR("tmp_offset_vec"));
                act_offset_vec.debug(PSTR("act_offset_vec"));
                offset_vec.debug(PSTR("offset_vec (BEFORE)"));
              }
            #endif

            offset_vec.apply_rotation(planner.bed_level_matrix.transpose(planner.bed_level_matrix));

            #if ENABLED(DEBUG_LEVELING_FEATURE)
              if (DEBUGGING(LEVELING)) offset_vec.debug(PSTR("offset_vec (AFTER)"));
            #endif

            // Adjustments to the current position
            const float xydiff[2] = { offset_vec.x, offset_vec.y };
            current_position[Z_AXIS] += offset_vec.z;

          #else // !ABL_PLANAR

            const float xydiff[2] = {
              hotend_offset[X_AXIS][tmp_extruder] - hotend_offset[X_AXIS][active_extruder],
              hotend_offset[Y_AXIS][tmp_extruder] - hotend_offset[Y_AXIS][active_extruder]
            };
[... 40 lines cut ...]
        #endif

        // Tell the planner the new "current position"
        SYNC_PLAN_POSITION_KINEMATIC();

        // Move to the "old position" (move the extruder into place)
        #if ENABLED(SWITCHING_NOZZLE)
         // current_position[Z_AXIS] += z_diff;  // Include the Z restore with the "move back"
        #endif

        if (!no_move && IsRunning()) {
          #if ENABLED(DEBUG_LEVELING_FEATURE)
            if (DEBUGGING(LEVELING)) DEBUG_POS("Move back", destination);
          #endif
          // Move back to the original (or tweaked) position
          do_blocking_move_to(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS]);
        }
        #if ENABLED(SWITCHING_NOZZLE)
          else {
            // Move back down. (Including when the new tool is higher.)
            do_blocking_move_to_z(destination[Z_AXIS], planner.max_feedrate_mm_s[Z_AXIS]);
          }
        #endif
      } // (tmp_extruder != active_extruder)
**************************************************************

thinkyhead, 2018-02-23:
Did you ever test `bugfix-1.1.x`? Because corrections were already made to that code. So please try ours first (especially since you have the right machine and we don't).
```cpp
          #if ENABLED(SWITCHING_NOZZLE)
            #define DONT_SWITCH (SWITCHING_EXTRUDER_SERVO_NR == SWITCHING_NOZZLE_SERVO_NR)
            // <0 if the new nozzle is higher, >0 if lower. A bigger raise when lower.
            const float z_diff = hotend_offset[Z_AXIS][active_extruder] - hotend_offset[Z_AXIS][tmp_extruder],
                        z_raise = 0.3 + (z_diff > 0.0 ? z_diff : 0.0);

            // Always raise by some amount (destination copied from current_position earlier)
            current_position[Z_AXIS] += z_raise;
            planner.buffer_line_kinematic(current_position, planner.max_feedrate_mm_s[Z_AXIS], active_extruder);
            move_nozzle_servo(tmp_extruder);
          #endif

          /**
           * Set current_position to the position of the new nozzle.
           * Offsets are based on linear distance, so we need to get
           * the resulting position in coordinate space.
           *
           * - With grid or 3-point leveling, offset XYZ by a tilted vector
           * - With mesh leveling, update Z for the new position
           * - Otherwise, just use the raw linear distance
           *
           * Software endstops are altered here too. Consider a case where:
           *   E0 at X=0 ... E1 at X=10
           * When we switch to E1 now X=10, but E1 can't move left.
           * To express this we apply the change in XY to the software endstops.
           * E1 can move farther right than E0, so the right limit is extended.
           *
           * Note that we don't adjust the Z software endstops. Why not?
           * Consider a case where Z=0 (here) and switching to E1 makes Z=1
           * because the bed is 1mm lower at the new position. As long as
           * the first nozzle is out of the way, the carriage should be
           * allowed to move 1mm lower. This technically "breaks" the
           * Z software endstop. But this is technically correct (and
           * there is no viable alternative).
           */
          #if ABL_PLANAR
            // Offset extruder, make sure to apply the bed level rotation matrix
            vector_3 tmp_offset_vec = vector_3(hotend_offset[X_AXIS][tmp_extruder],
                                               hotend_offset[Y_AXIS][tmp_extruder],
                                               0),
                     act_offset_vec = vector_3(hotend_offset[X_AXIS][active_extruder],
                                               hotend_offset[Y_AXIS][active_extruder],
                                               0),
                     offset_vec = tmp_offset_vec - act_offset_vec;

            #if ENABLED(DEBUG_LEVELING_FEATURE)
              if (DEBUGGING(LEVELING)) {
                tmp_offset_vec.debug(PSTR("tmp_offset_vec"));
                act_offset_vec.debug(PSTR("act_offset_vec"));
                offset_vec.debug(PSTR("offset_vec (BEFORE)"));
              }
            #endif

            offset_vec.apply_rotation(planner.bed_level_matrix.transpose(planner.bed_level_matrix));

            #if ENABLED(DEBUG_LEVELING_FEATURE)
              if (DEBUGGING(LEVELING)) offset_vec.debug(PSTR("offset_vec (AFTER)"));
            #endif

            // Adjustments to the current position
            const float xydiff[2] = { offset_vec.x, offset_vec.y };
            current_position[Z_AXIS] += offset_vec.z;

          #else // !ABL_PLANAR

            const float xydiff[2] = {
              hotend_offset[X_AXIS][tmp_extruder] - hotend_offset[X_AXIS][active_extruder],
              hotend_offset[Y_AXIS][tmp_extruder] - hotend_offset[Y_AXIS][active_extruder]
[... 42 lines cut ...]
          if (DEBUGGING(LEVELING)) DEBUG_POS("Sync After Toolchange", current_position);
        #endif

        // Tell the planner the new "current position"
        SYNC_PLAN_POSITION_KINEMATIC();

        // Move to the "old position" (move the extruder into place)
        #if ENABLED(SWITCHING_NOZZLE)
          destination[Z_AXIS] += z_diff;  // Include the Z restore with the "move back"
        #endif
        if (!no_move && IsRunning()) {
          #if ENABLED(DEBUG_LEVELING_FEATURE)
            if (DEBUGGING(LEVELING)) DEBUG_POS("Move back", destination);
          #endif
          // Move back to the original (or tweaked) position
          do_blocking_move_to(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS]);
        }
        #if ENABLED(SWITCHING_NOZZLE)
          else {
            // Move back down. (Including when the new tool is higher.)
            do_blocking_move_to_z(destination[Z_AXIS], planner.max_feedrate_mm_s[Z_AXIS]);
          }
        #endif
      } // (tmp_extruder != active_extruder)
```

closed 2018-02-23 by thinkyhead

HD78, 2018-02-25:
Today (finally) I had the time to test bugfix-1.1.x and yes, I can confirm that it is fixed.

Thanks.