MarlinFirmware/Marlin #7291: 'Simulation Mode' Bug

BigBox3D, 2017-07-14:
Running IDEX Dual X-Carriages.

Send movement commands over serial I get the OK back from Marlin but nothing actually moves.

The LCD controls work as expected and I can move the head(s) around. 

If I send a G10 X150 Y100 Z50 F3000 command the LCD updates the position on the LCD but the machine remains motionless. However using the LCD to move the head will immediately move to the position set with the G1 command, plus any changes in position made via the LCD.

If I try to print via Octoprint all the lines are sent, OKs are received and the LCD updates the XYZ positions, but it sits perfectly still.

Same error with both 1.1.4 and 1.1 bugfix.

It is what I would term to be Simulation Mode.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1148162/configs.zip)

FHeilmann, 2017-07-14:
did it ever work with a version of Marlin you compiled yourself?

Seems like a PIN assignment is faulty. Make sure you select the right motherboard.

oysteinkrog, 2017-07-14:
There is a nasty bug with IDEX/DUAL_X now where it does not work at all.
See https://github.com/MarlinFirmware/Marlin/issues/7050 and
https://github.com/MarlinFirmware/Marlin/issues/6956

teemuatlut, 2017-07-28:
To reproduce:
* Enable DUAL_X_CARRIAGE
* Power up the machine
* On serial send G1 X100

Is there anything else needed to reproduce the problem? No T0/T1 commands needed?
If 1.1.1 works, does 1.1.2? 1.1.3? This would help me track down the specific commit.

You can easily switch through versions with ```git checkout 1.1.2```

oysteinkrog, 2017-07-31:
I've previously confirmed that 1.1.2 and later do not work.
I'm currently bisecting my way to the commit that introduced the problem, I'll let you know once I have it nailed down.

oysteinkrog, 2017-07-31:
Ok, 91841d75c9a473e15932bd4589d64ee59efe947e is the commit that introduced the problem.

oysteinkrog, 2017-08-04:
Hmm, for some reason my merged PR did not correctly close this, please close it now.

PR #7386 (2017-08-01): Fix DUAL_X_CARRIAGE movement
Broken by 91841d75c9a473e15932bd4589d64ee59efe947e (between 1.1.1 and 1.1.2).
Search for DUAL_X_CARRIAGE in that commit and the error is obvious (removed line in the #if chain).
The old code was:
```cpp
  #if IS_KINEMATIC
    if (prepare_kinematic_move_to(destination)) return;
  #else
    #if ENABLED(DUAL_X_CARRIAGE)
      if (prepare_move_to_destination_dualx()) return;
    #endif
    if (prepare_move_to_destination_cartesian()) return;
  #endif
```
…and was changed, in error, to…
```cpp
   if (
     #if UBL_DELTA // Also works for CARTESIAN (smaller segments foll [... cut]
commit e948f77c (2017-08-01): Fix DUAL_X_CARRIAGE not moving at all

Axis would only move when homing.
Bug introduced in 91841d75c9a473e15932bd4589d64ee59efe947e.
Fixes #6956, fixes #7050 and fixes #7291
closed 2017-08-28 by thinkyhead