MarlinFirmware/Marlin #7292: [FR] invert step pulse

rooiejoris, 2017-07-14:
is it possible to invert the step pulse for the stepper drivers...?!

[maybe I am explaining it wrong, my background in not in programming or electrical enigeering, but product design... sorry in advance]

Now the puls is a 5V signal and default 0
I want the situation where the pulse is 0 and the default is 5V

I was looking in configuraton advance and could find the step length in us, but not the invert
I think I have to manual change something in stepper.cpp, but not sure what [I am using an cartesian frame, so first test could be hacked by myself if I know what and where

thanks in advance...!

kind regards,
joris

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------

FHeilmann, 2017-07-14:
you can do that in Configuration_adv.h
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L354

GMagician, 2017-08-19:
Maybe INVERT_X_STEP_PIN, INVERT_Y_STEP_PIN may solve...

rooiejoris, 2017-08-23:
Thanks, I still have to try the options [many other things as well going on], but I should be fine...

closed 2017-08-23 by rooiejoris