MarlinFirmware/Marlin #7294: AUTO_BED_LEVELING_BILINEAR does not work correctly for long segments?

lutorm, 2017-07-15:
I'm using bilinear leveling probing the 4 corners to correct for my saddle-shaped bed (by about 0.4mm). This works correctly in the corners, but the height at the center of the bed is not consistent. If I stop at the center, the nozzle is at approximately the correct height, but if I move across the center from the corners (like it would if I was infilling something covering the entire bed) then the height of the nozzle when it passes the center is either higher or lower, depending on which direction I go. It's as if it doesn't correct the entire segment for the bed shape, but just connects the endpoints with straight lines.

This is with 1.1.4 on a TAZ6.

My G29 results in 
```
Bilinear Leveling Grid:
      0      1
 0 +0.541 +0.235
 1 +0.222 +0.636
```

Is this what the ABL_BILINEAR_SUBDIVISION is supposed to fix? If I enable that, I see stalls in long linear movements instead.

Roxy-3D, 2017-07-15:
Seriously....  There is not reason to use Mesh Bed Leveling scheme if you are only going to have a 2x2 grid.

Assuming you know that....  If you have an area of the mesh that is too high or too low...   Use M421 to adjust the mesh point.   Sub-Division only fixes things within a 'Mesh Cell'.   If the area is too high or too low...  You need to adjust the mesh points.   And especially if you got the mesh points using manual probing, there is a lot of error in that procedure.  Please be willing to adjust the mesh points to get good adhesion on the first layer.

Bob-the-Kuhn, 2017-07-15:
Basically you need more points in your mesh.  I'd recommend a 5x5 and see if that gets acceptable results.

lutorm, 2017-07-15:
"There is not reason to use Mesh Bed Leveling scheme if you are only going to have a 2x2 grid."

How do you figure? A single bilinear interpolation takes care of offset and planar tilt plus the lowest-order deviation from flatness. (Basically, the simplest thing you can do to a plane to make it not flat is to twist it into a saddle.) Since amplitude universally goes down with order (unless you have some really funky patterning on your bed) there will always be progressively diminishing results from going to higher and higher order.

Then there's the practical matter that the TAZ6 is set up for automatic probing of the four corners, and it does this very well (probe repeatability test shows < 0.03mm).

What this issue is about is the fact that even if I have a bed whose shape is *perfectly* described by a single bilinear interpolation, it appears that Marlin doesn't apply this correction except at the segment end points. If that's the case, you need more mesh points just for the segments, rather than because the bed shape requires it. Is that how it works?

fiveangle, 2017-07-15:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L2599

Roxy-3D, 2017-07-17:
> How do you figure? A single bilinear interpolation takes care of offset and planar tilt plus the lowest-order deviation from flatness. (Basically, the simplest thing you can do to a plane to make it not flat is to twist it into a saddle.)

Yes.  But  depending on the order you do the bilinear interpolation, you are going to get a different 'saddle' shape.   If you do the X interpolation first and then Y you will get different approximations of the correction height than if you do the Y first followed by the X.

So...  the easiest answer is add more mesh points so the smaller mesh cells can better approximate the shape of the bed.

lutorm, 2017-07-17:
Unless I'm completely mistaken the order in which you do the bilinear interpolation does *not* matter. Say we have a square:
```
z2 --- z3
 |       |
z0 --- z1
'''
and we call the fraction by which the interpolated point are between the two edges fx and fy, in the two axes. If we first interpolate in the x-direction, we get
```
z_temp1 = (z1 - z0)*fx + z0
Z_temp2 = (z3 - z2)*fx + z2
```
and the final interpolated point by interpolating these in y:
```
z = (z_temp2 - z_temp1)*fy + z_temp1
```

If you instead do y first, you get
```
z_temp1 = (z2 - z0)*fy + z0
z_temp2 = (z3 - z1)*fy + z1
```
and the final interpolated point is now:
```
z = (z_temp2 - z_temp1)*fx + z_temp1
```
If you write out these two expressions, they both come out to be
```
z = z0 + (z1 - z0)*fx + (z2 - z0)*fy + (z3 - z2 - z1 + z0)*fx*fy
```
regardless of the order in which you started. (And this is intuitively the way it should work because the interpolation is effectively laying down straight rulers first along the edges of the square and then across the interior in the different directions and because everything is linear, the result is unique.)

thinkyhead, 2018-02-09:
@lutorm You are correct. It is commutative.

closed 2018-02-09 by thinkyhead