MarlinFirmware/Marlin #7296: Question: A way to calculate the Z offset

Bergerac56, 2017-07-15:
This is a question, not an issue.

I would like to have an "assist function" in order to determine the best Z offset. (Perhaps it exists but I did not find it yet)

A possible way would be to  "doubble click" on the button to reach the "Z Offset" value and to be able to modify it, **having the Z motor activated** (which is not the case it seems). So a modification of the value would move the bed till the nozzle is as close of the bed as possible (or where needed). At Z value = 0, the bed would be where the Z probe reacts. The desired Z offset would be the result of this operation.

In UBL, there are functions doing a part of that (G20 P4 T for ex.) but it affects the mesh and starts far from the bed (if UBL is not activated or no mesh) or somewere defined by the mesh. This is not what I would like.

Is there a function or a set of commands (to put in user commands for exemple) to do that?

Thanks a lot

benlye, 2017-07-15:
Isn't that what `BABYSTEP_ZPROBE_OFFSET` does?

Bergerac56, 2017-07-15:
When I invoque M851, it changes the value of Z Offset. But what I want to achieve is to be able to modify this value with the button and at the same time to see the Z axe moving. So I can see if I touch the bed or not. On my printer (last bugfix available) and with babystepping activated, I can change the value of Z Offset with the button but the Z motor is not activated. 

Did I miss something?

NiklasOlofsson, 2017-07-15:
Noticed the same thing. Babystepping the z offset does not happen in real time. Should it do that?

Bergerac56, 2017-07-15:
It would be nice. So we can immediately check the quality of the Z value we want to input.

Bergerac56, 2017-07-16:
The desired function would be:

* Do a Z Home (So the Z=0 is known. This could be skipped if G28 already done)
* Move Z to 0 (Not compensated by the previous Z_OFFSET. The "0" defined by the Z_probe)
* Enable Z babystepping moves of Z (activating Z stepping motor) and show the result on LCD screen (So we can slowly move Z till the nozzle touches the bed)
* Long click to go back to main menu (There we can store settings/store Z offset if wanted)
 
Does this exist somewhere? Is it doable?

Thanks

flodejr, 2017-07-16:
What I do to calculate zero offset is
1) G28 - home the hotend and it will also clear all other settings
2) M851 Z0 (set Z offset to zero)
3) G1 X110 Y110 Z0 - move to center of bed and Z-axis to zero position (this is the zero position known by the software)
4) Use the knobs on the LCD to move the Z-axis to closest to the bed (must switch M211 S0, turn off software endstops to use -ve values for Z movement)
5) G92 Z0 (set current position as the new zero position)
6) G29 X110 Y110 (probe Z-axis offset, I am using BLTouch) 
7) M851 Z(value) set the offset given by the probe
8) M500 Safe to EEPROM

Would love to use baby stepping but not sure how it actually works.  But this is currently available to achieve what you want I guess.

Roxy-3D, 2017-07-16:
> Would love to use baby stepping but not sure how it actually works. But this is currently available to achieve what you want I guess.

Just turn it on...   And also turn on   #define DOUBLECLICK_FOR_Z_BABYSTEPPING

It is very simple...  When the printer is doing the skirt of your object, double click the encoder wheel...  You should drop straight into the Z-BabyStepping.   You can use the dial to adjust the nozzle height above the bed.

flodejr, 2017-07-16:
> Just turn it on... And also turn on #define DOUBLECLICK_FOR_Z_BABYSTEPPING
It is very simple... When the printer is doing the skirt of your object, double click the encoder wheel... You should drop straight into the Z-BabyStepping. You can use the dial to adjust the nozzle height above the bed.

Bergerac56, 2017-07-16:
Thanks flodejr this is not far from what I did before.
I also used the way Roxy is mentionning. But this seems to be only possible when printing.
My point was to be able to do it without printing. Just "doubleclick" and have the z motor activated.

flodejr, 2017-07-16:
> My point was to be able to do it without printing. Just "doubleclick" and have the z motor activated.

If I am not wrong, I think I was able to adjust baby steps while doing the Z-axis tuning movements through double-clicking.  But I am not sure if that is captured as I did not notice any movements in Z.

Roxy-3D, 2017-07-16:
No!  It is not 'captured'.   BabyStepping is outside of Marlin's control and knowledge.   It is a way for the user to feed step pulses to the stepper motors without Marlin knowing that it is happening.    If the printer is not moving and you do BabyStepping on any axis, you will see the nozzle move.  But the LCD Panel will keep displaying the same coordinate for the nozzle through out the entire operation.

tcm0116, 2017-07-19:
>My point was to be able to do it without printing. Just "doubleclick" and have the z motor activated.

That's an interesting idea. However, the Z axis would need to be homed if it is activated after having been deactivated since it's likely that the exact Z position was lost when it was deactivated. However, it would be possible to prevent disabling of the Z axis when the Z Offset screen is being displayed. What do you think of that idea @Roxy-3D?

Bergerac56, 2017-07-19:
It is exactly that ;)

As of today, I do it this way with the Graphic LCD, knowing that Babystepping and UBL are activated

I created 2  Custom Commands to be comfortable (The first-one could be avoided using existing commands): 
  #define USER_DESC_1 "Init.  Z_Offset Test"
  #define USER_GCODE_1 "G28\nG29 D"
  
  #define USER_DESC_2 "Exec. Z_Offset Test"
  #define USER_GCODE_2 "G28 Z\nG1 Z0"

Steps to refine Z_OFFSET :
1.	Do Custom Command “Init.  Z_Offset Test”  once (Custom Command/ Init. Z_Offset Test)
2.	Do Custom Command “Exec. Z_Offset Test” (Custom Commands/ Exec. Z Offset Test)
3.	Check with a piece of paper
4.	If not good: DoubleClick to reach Babystepping Z_Offset and add/retract a bit.
5.	Do Custom Command “Exec. Z_Offset Test” (Custom Commands/ Exec. Z Offset Test)
6.	Check with a piece of paper
7.	If not good, redo (4-5-6)
8.	If OK and you want to save it, do (Control/Store Settings)

I would love to have a function doing what tcm0116 proposes for Z_OFFSET. The global process would be: 
•	Initiate the process (G28 and start with the Z_OFFSET value stored in EEPROM. Deactivate UBL if needed as, I believe, it influences the Z_OFFSET calculation)
•	Place the nozzle in the center of the bed and at Z=0. Then give control to a function like the babystepping Z_OFFSET (doubleclick) but able to move the Z motor (so, by turning the encoder, we can directly see the result of the “sheet of paper” test.)
•	End by clicking again on the button. At this stage the user could store the new value into the EEPROM (Save settings) or leave it like this (exit without definitive changes)

Roxy-3D, 2017-07-19:
I pretty much leave my stepper motors turned on (activated) all the time.   Some times when I'm bringing up a new firmware on a printer or a new printer I end up with the stepper motors de-activating.   But I change it when I notice it happening.

I've never understood the idea of having the printer turned on but the stepper motors deactivated making sense.    But people obviously like that option.

tcm0116, 2017-07-19:
I let mine turn off because that's the default is Marlin, but also because it seems unnecessary to keep the drivers holding the steppers after the print ends (since I'm usually not home).

Bergerac56, 2017-07-19:
The UBL stuff is interesting. I have tried the Z_OFFSET process with and without it. For a certainly subjective reason, I had the impression that it was more accurate without UBL activated during the adjustment process. I will give it a new try :)

Roxy-3D, 2017-07-19:
> I have tried the Z_OFFSET process with and without it. For a certainly subjective reason, I had the impression that it was more accurate without UBL activated during the adjustment process. I will give it a new try :)

The 'right' way to set your Z_PROBE_OFFSET_FROM_EXTRUDER is to do it with no bed leveling active.   I prefer to do it at the center of the bed.  And I also prefer to home the Z-Axis at the center of the bed.

Once you can print a small object at the center of the bed...   Then...  And only then...   does it make sense to start building the UBL mesh and bringing up the system.

Bergerac56, 2017-07-20:
Thanks Roxy, this was what I thought too. Doing this way (without UBL, center of the bed) I get very good prints on all the bed. Obviously, the development of UBL improved a lot this as now, the quality is good all over the bed.

closed 2017-08-17 by Bergerac56