MarlinFirmware/Marlin #7297: Filament Diameter flow rate not correct

blackus3r, 2017-07-15:
Board: Ramps 1.4 + Atmega 2560
Version: 1.1.4

I have setup a filament diameter sensor. 
Marlin displays the correct diameter measured from the sensor on the LCD (1,8mm measured). 
Nominal diameter is set to 1.75
M404 delivers 1.75
M407 delivers 1.8
Slicer is set to 1.75 and the M405 command is set at the beginning. 

The strange thing is, that to the right of the current diameter the percentage is set to 239% (this means it would calculate the 1.8mm filament up to 4,x mm !?) and it does not change. I have a big over extrusion since I have connected the sensor. 

Is there a bug or did I something wrong with the configuration?

empakoso, 2017-07-17:
I've experienced this as well.  Check your hardware first. Verify that the signal output is exactly 1vdc to 1mm measured. That means if you measure the drill rod calibration tool, that millimetre reading will display on your DVM but in volts DC.  For example my filament measures 1.71mm and my DVM reads 1.71vdc.  This has likely already been done (inferred by your post) but it sets ground-zero.
Next manually purge the distance from sensor to nozzle (the number you loaded in your fw settings).  There is a chance that the firmware is applying some type of default (3mm) to the filament between sensor and nozzle before your actual readings hit the bed.
Like I mentioned, i get interesting 'V' values displayed on startup (fw upload, I believe, not sure, done so many times now) but it never affects my print.
Maybe @filipmu can step in and correct me, but I believe the LCD displays realtime measurements and volume percentage of what is being applied on the bed at that moment.
hth

blackus3r, 2017-07-18:
Thanks for your reply. 

My sensor is working correctly, because on the display appears 1,81mm but the flow rate on the right side shows always 239% !? 

Nominal filament diameter is set to 1.75 this would be a flow rate about 103,4% but there is always 239%

empakoso, 2017-07-18:
No problem,
If after purging the distance it still reads 239% then, might I suggest building a voltage divider using a 10 turn POT.  This is how I first setup and tested my printer and firmware settings before I bought and installed the FWS.  There could be an analog pin signal issue.
A very cool by-product is that you can simulate diameter measurements, where I had turned the POT voltage to 1.85vdc And it simulated under extrusion.  It created a super fine see-thru side wall. It was interesting and got me thinking about dynamic wall density adjustments like baby stepping the extruder.
In the end, it tested that my firmware settings where functioning normally.

empakoso, 2017-07-18:
Double check your firmware is set using whole numbers in centimetre units.  I understand the 239% is a value, but my comment above is addressing the functionality and is ignoring a possible LCD display calc issue for the moment.

blackus3r, 2017-07-18:
Thanks for your answer. I will test it with another method. 

But the strange thing is, if I disconnect the sensor, the diameter reches 4,xxmm and the extrusion rates is still at 239%.

filipmu, 2017-07-18:
Sounds like there might be a pull-up resistor on the input you are using (a temperature input?).   Filament sensor won't work with a pull-up. Might need to look for another input. 

Filip

blackus3r, 2017-07-19:
I have no pull up resistor connected. When I connect the filament width sensor, the voltage reaches 1,8 measured with a multimeter and the display shows the 1,8mm. 
Before I start any print, the flow rate is exact 100%, when the printer starts, the flow rate increases till 239%.

ExtremeSitting, 2017-09-10:
I'm having the same issue. It seems to be triggered by the M405 command. Moving that command around in the start GCODE changes when the extrusion rate is changed. In my case, the filament is measured at 2.85 and the sensor shouldn't be modifying the extrusion multiplier for any change <.05. However, it changes the extrusion to 260%.

This seems to be intermittent though. Most of the time the extrusion multiplier works exactly as it should. In fact, even printing the same model over and over has caused this issue. No variable changes, yet the extrusion rate will be amplified by 150-160%.

I'm using Marlin 1.1.4

empakoso, 2017-09-10:
Update to fixing my over extrusion...
First, I've upgraded to the latest version 1.1.5-bugfix.  There have been a few changes made to the FWS code. Especially the M405 by @GMagician
The second major improvement I've made is shielded cable.  My sensor cable was wrapped together in a cable chain alongside all the other Hotend wires.  I was getting unreliable readings while the Hotend PID cycles and the extruder stepper we functioning.
My solution was to find a good quality USB cable with a ferrite choke (larger round cylinder at one end) and cut the ends off for connecting the FWS to my control board.
The ferrite choke should be at the end connected to the controller.
Make sure your shielding is very well protected (trimmed back carefully not to short the wires out, and covered with tape/shrink tube) at the FWS end.
Then for best results at the controller end, twist back the shielding wires (cut the foil away) so that you can crimp a lug to the end for grounding it to the 0v input terminal or controller 0v shield, or 15ft ground rod :) This should be as short as possible.
Connect the wires to the terminal. You may have to solder a bit of length to these wires because of the overmolded choke design of your cable which would limited your length.  It is important not to have the wires lay beside any noise sources between exposed shield and connection point.  The wires can cross perpendicular but not run alongside (or you will be in the same situation).
My final result is that now I've done away with the intermittent huge extrudes and the blobby over extruded layers in my print layers.
The last major improvement is the quick responsiveness my FWS now provides to my LCD.
In case of other interference from power up, I've created a 9cm extrude to clear out the buffer of any large readings. 9cm is what my firmware is set to between FWS and my nozzle.  I now this is not possible for everyone and it may be overkill so use that last one as needed.
I was in the same situation as you mentioned. I was adjusting too many variables in the firmware.  I too was adjusting LIN_ADV and FWS.  So after setting and calibrating, I simply leave a piece of filament in my sensor and let it ride along without feeding through to the hot end.  Allows me to keep an eye on the diameter readings for stability.  I also put M407 at every layer change and/or action I could to report diameter values through out my prints.  This would report the known diameter of my stationary filament width. I'd collect the readings throughout the print from my host and take a look for stability.  First I'd perfect the LIN_ADV all while monitoring my sensor without affecting my LIN_ADV study.
BTW, the easiest way to create a ride-along FWS is to use a small earth magnet that is glued to my sensor. This keeps it anywhere I need it; either in service or out of service.  The best part is the sensor bends to the direction of the filament as the head moves to prevent strain on the filament angle.  All the while the sensor is kept in place and is prevented from getting in the way.
Just my $0.02

ExtremeSitting, 2017-09-10:
@empakoso Thanks for the information. I don't have the wires shielded for the FWS. I'll get that set up and try it out. It seems like a likely issue in my case. I have my bed leveling sensor (BLTouch) in the same unshielded wire bundle as the FWS all in close proximity to the extruder motor.

empakoso, 2017-09-10:
@ExtremeSitting Good Luck
Keep us posted.

empakoso, 2017-09-11:
Oh BTW I found that on my cloned Arduino 2560 (ZRIB v2.2) that if I tied my unused analog input pins to 0v, my readings also improved.

blackus3r, 2017-09-11:
@ExtremeSitting 
Does your sensor shows the correct Diameter and Marlin calculates 260% flowrate? Or is your diameter also wrong? 
In my case, everything is well. My sensor says "1.8mm", marlin displays this value, but the flow multiplier doesn't fit.

empakoso, 2017-09-11:
Yes this happened to me. The measurement is just a static update from time to time. The flow rate is the dynamic variable in use. Some times my display wouldn't even show the glow rate change and my extruder would chatter, meaning the flow rate changed and the measurement had already travelled the 9cm to the nozzle. 
I am almost certain it was interference and the phenomena of "open" or ungrounded analog inputs.

blackus3r, 2017-09-11:
My flow rate hangs always at 239% :-/

empakoso, 2017-09-11:
If it stays static or doesn't fluctuate you may have either of these two conditions:
A) the sensor is not wired properly to its source or it is broken and outpouring the wrong voltage.
B) the analog input number is wrong in the Marlin config( might be a digital and not analog input) or it may not even. E on the right pin

It seems your analog pin is receiving 7.2v
You have to be able to measure that voltage at the sensor connection point of the analog pin.
The sensor provides a 1mm to 1volt dc measurement ratio.  So 1.8mm should output exactly 1.80mV dc when not in ratio metric mode (or flashing LED).
Oh I am assuming you are using the sensor that is sold by Filip

empakoso, 2017-09-11:
My mistake Marlin is reporting the analog pin input as being 4.14 volts DC and 1.8mm will output at the sensor 1.8 volts DC

My comment above incorrectly says 7.2v and 1.8mVdc these were my mistake.

blackus3r, 2017-09-11:
My multimeter says 1.8V, Marlin says Filamentdiameter: 1.8mm, flow rate stucks at 239%, Filament Diameter is configured with 1.75mm 

Maybe I will test it again with the current available version.

empakoso, 2017-09-11:
Do you have access to a scope? Would be interesting to see a flat line.  Rule out noise.
I was just going to suggest v1.1.5-bugfix.  There has been some work done to the filament gcodes. Else unless you have Arduino software that you can monitor readings in realtime, I am not sure were that value is coming from.  Can you configure to another analog pin?

blackus3r, 2017-09-12:
I will test it with the current version. Maybe it was fixed. I will let you know :-)

closed 2017-11-25 by thinkyhead

blackus3r, 2017-12-09:
Hi, 

I have tested it today again with Marlin 1.1.6. 
Display shows 1.79mm 
During print it display V260%
I am using this sensor: https://www.tindie.com/products/JasonKits/3d-printer-filament-width-sensor-pcb-kit/

M404 delivers: 1.75mm
M407 delivers: 1.78 - 1.8mm 

Nominal diameter is set via Firmware and again via slicer.

blackus3r, 2017-12-09:
Hi,
I have tested it today again with Marlin 1.1.6.
Display shows 1.79mm
During print it display V260% sometimes 253%
I am using this sensor: https://www.tindie.com/products/JasonKits/3d-printer-filament-width-sensor-pcb-kit/
M404 delivers: 1.75mm
M407 delivers: 1.78 - 1.8mm
Nominal diameter is set via Firmware and again via slicer.

blackus3r, 2017-12-09:
Sometimes it shows up 100% 102% and then it jumps up to 260% for one seconds and then back to 100%.

thinkyhead, 2017-12-10:
Marlin 1.1.6 hasn't been patched. Recommend you test with the `bugfix-1.1.x` branch.

If the display is showing over 200% but the filament width isn't showing over 3.4mm that's an interesting discrepancy. We'd need to add logging to figure out where it stems from.

blackus3r, 2018-01-31:
I do not want to open a new issue. 
I haven‘t test 1.1.6 yet, but I made many tests with 1.1.4 and here are my reaults:
- my bowden is 60cm long (from sensor until heating chamber)
- at the start of the print, the extrusion miltiplier is set to the configured min diamter. Marlin thinks, that the filament from the sensor until the chamber is at the minimum of the diameter. 
- the print goes on with heavy over extrusion. Now we have so much filament extruded (60cm) that the filament which was meassured at the begining reachea the heating chamber. Marlin changes correctly the Multiplier and everything is fine. 
- There are some cases, where the sensors freaks out or marlin does not work correctly and the multiplier reaches for some seconds the maximum. 

I will test the behavior with 1.1.6, or does this makes no sense, because its included no fixes?

thinkyhead, 2018-02-05:
Only `bugfix-1.1.x` matters to us for bug reports. Please always test the bugfix branch before reporting an issue.