MarlinFirmware/Marlin #7298: Printrbaord RevD Firmware Upgrade To Marlin 1.1.x

Karansheth, 2017-07-15:
I am not able to flash My **Printrboard RevD to Marlin 1.1.X**
I am using the **Arduino IDE 1.8.3** And **teensduino 1.37**  
When trying to upload marlin the compilation is successfull and the teensy app opens but the teensy app is not able to detect the Printrboard , I even tried using the Printrboard firmware updater for mac but even that is not able to detect the board , so i tries to flash the older firmware using arduino 0022 and it successfully uploaded But I want the Mesh Bed Levelling Support So I want to flash the latest Marlin , Please Guide

Bob-the-Kuhn, 2017-07-15:
You're not using the Teensy bootloader so the Teensy loader won't work.  I don't know of a way to reload the Teensy bootloader.

I'm not aware of an Arduino IDE extension that works with that bootloader and will compile Printrboard.

If you look in the file **pins_PRINTRBOARD_REVF.h** you'll find instructions on how to install an extension called "Printrboard".  It contains a HID bootloader that can be installed via instructions in the file.

You may want to wait a couple hours to see if another Printrboard user can point you to an extension that is compatible with the LUFA CDC bootloader.

fiveangle, 2017-07-16:
The printer board never came with the teensyduino boot loader. You need to use Atmel Flip (on Windows) or dfu-programmer (on Linux). If you're not using the stock dfu boot loader then you'll need to use whatever app is required for that  then you'll need to use whatever app / driver that is required for that boot loader.

fiveangle, 2017-07-16:
How do you know you're using LUFA CDC ?

Karansheth, 2017-07-16:
I Bought from this Place [http://makemendel.com/mechanical-parts/electronics-parts/printrboard](url) , There they have mentioned that they sell with LUFA CDC Bootloader , I have the Extension for Arduino IDE but the extension only works on the old IDE and not the latest one so I am stuck using Old MArlin Firmware or the new Repetier Firmware as the Repetier Firmware is uploading Using the old IDE

fiveangle, 2017-07-16:
Yes,  what you bought is essentially a clone of the a product from 2013  with some vendor specific customizations.   I wish I could help but I don't know anything about that boot loader. 

My suggestion would be to Google the correct AVR dude settings required for hand flashing to a board with that boot loader,  or use an AVR ISP  to burn a stock Atmel DFU boot loader to the board  then use the suggested DFU methods.

Learning the method for programming using that boot loader (again, google) shouldn't be too difficult, but I have zero experience with it as I am familiar only with rev F and newer.

One benefit of the LUFA CDC boot loader is that it presumably occupies just 4k  of PROGMEM flash area, while the stock Printrboard DFU boot loader occupies 8k, and every bit helps on these small 128k devices.

fiveangle, 2017-07-16:
Looking at the [RepRap Printrboard wiki entry](http://reprap.org/wiki/Printrboard), I find this being either outdated info, or evidence that your boot loader is absolutely useless:

> LUFA's CDC Bootloader: Allows direct uploading of firmware through Arduino/avrude via avr109 protocol. Requires no driver on Linux/Mac and free INF installer to use the built-in Windows driver, and Arduino 022 with modified Teensylu boards configuration. Avrdude within Arduino directory must also be upgraded to newest version. Limited to 64k flash space by the protocol. Note that this is a severe limitation rendering most modern firmwares (with e.g. SD card and/or graphical display support) unusable.

But I think I recal in practice avrdude will throw a programming error because it can't verified beyond 64k using that boot loader but it actually can program the whole 128k. 

For my $$ I'd probably spend the effort to swap to a known solid bootloader now so you can move forward using standard methods. DFU bootloader is solid as a rock, but does consume 8k and requires using Flip or dfu-programmer to burn your hex (with the requisite jumper dance).

Bob-the-Kuhn, 2017-07-16:
@fiveangle - the options I see are:
1) New bootloader - Buy a hardware programmer (USBtinyISP or similar), switch to an Arduino IDE that supports the Printrboard and has a bootloader option.
2) Switch to a programming environment that supports the LUFA CDC bootloader.  My understanding is  Platformio will be able to do this.
3) Hack a current Arduino IDE extension so that it uses the AVR109 protocol.  I could do that but it means he's stuck with that hack forever which could get complicated if Adrduino IDE ever has a major change.

Any other options?

fiveangle, 2017-07-17:
there's actually even more options than that (which is redic, if you think about it).

@Karansheth - here's the Rabbit hole, should you choose to go down it:  http://blog.lincomatic.com/?p=548 (plus some interesting info here: https://deskthority.net/workshop-f7/bootloaders-t7608.html)

[This post shows how to make CDC work with the ancient Arduino](https://my3dprinterbuild.wordpress.com/2012/08/29/arduino-printrboard/), but perhaps you can figure out how to hand-jam the updates into the latest 1.8.3/1.3.4 Ard/Teensy config ?

For my $$, I'd grab an AVR ISP for next to nothing ([something like this](http://www.ebay.com/itm/USB-ISP-Programmer-Downloader-for-51-AVR-ATMega-ATTiny-Adapter-G5Z1-/332261995109?epid=683873217&hash=item4d5c5bb265:g:JTYAAOSw3YJZP1wj)), wire it up appropriately to the [JP4 ISP pad on the Rev D](http://reprap.org/mediawiki/images/d/d5/Printrboard_RevB_Schematic150.png) and drop a known working bootloader on it (worst case you can load Atmel Studio and should be able to load the DFU bootloader hex with the required fuse settings).

If you go that route (not ideal either, I know), the DFU boot loader works with [Atmel Flip](http://www.atmel.com/tools/flip.aspx) and [dfu-programmer ](http://www.technoburst.net/2013/04/atmel-dfu-programmer-tutorial-for-linux.html).  Follow [this for windows](https://printrbot.zendesk.com/hc/en-us/article_attachments/203320786/Windows_How_to_Flash_Firmware.pdf), or [this for Linux](http://www.printrbottalk.com/wiki/index.php?title=Firmware_Update_Linux)  (using apt-get dfu-programmer to get the binaries), or Mac (using [dfu-programmer binaries for Mac](http://rudix.org/packages/dfu-programmer.html) within the Mac terminal).

I have my Printrboard connected exclusively to my RaspberryPi and use dfu-programmer on it to upgrade constantly w/o any issues and I like the security that i have to connect a physical jumper to change FW, but whatever bootloader you can get to work, go with it.

If you do end up going through the trouble to switch to DFU, just compile the hex within Arduino with the Teensyduino extension (which you've already done), ensure that the resulting hex is no bigger than 122880 (Teensyduino extensions will not warn you since the Teensy boot loader uses only 1k on AT90USB (so claims 130048 availabe, which is incorrect with larger bootloaders), while most other boot loaders use 4k (HID, CDC, etc) or 8k (DFU).   Then:

1. Short the "ALE" (D) / "BOOT" (F) JP14  jumper
2. Hit HW reset button on the board
3. dfu-programmer at90usb1287 erase
4. dfu-programmer at90usb1287 flash <hex filename>
5. Remove jumper
6. Hit HW reset on the board
7. Drink beer

:)

fiveangle, 2017-07-17:
Also, if you really want to get bleeding edge, you can try [this one I just dug up](http://www.avrfreaks.net/projects/usbl-256-word-usb-bootloader?module=Freaks%20Academy&func=viewItem&item_id=2739&item_type=project) that should fit in the 1k minimum boundary (same as the PJRC "Halfkay") - let us know how you make out on that one ;)

fiveangle, 2017-07-17:
btw- after a lot of the latest Marlin optimizations, I'm able to load a relatively full feature set on 128k RevF with 8k DFU boot loader: BILINEAR levelling, LIN_ADVANCE, Full-Graphics LCD, Babystepping, NO SD.

Karansheth, 2017-07-18:
@fiveangle I will try to flash the dfu bootloader in a day or two and report if there's any problem or if i am able to get it working with the latest firmware , thanks for the support

fiveangle, 2017-07-18:
it's critical to get the fuse settings correct when changing the boot loader address space, or you can brick the board.  just make sure you've found a recipe that works correctly before you do it (or spend the effort to install the behemoth atmel studio, which i've heard should automatically set them correctly for the dfu boot loader, but i've not tried it myself).  Good luck !

Karansheth, 2017-07-18:
Ok I will keep that in mind

fiveangle, 2017-11-14:
Close ?

Roxy-3D, 2017-11-14:
yeah....

closed 2017-11-14 by Roxy-3D