MarlinFirmware/Marlin #7299: [FR] Debug Mode similar to repetier firmware.

mmiscool, 2017-07-15:
Hello,
In the repetier firmware you can enable a debugging mode that constantly sends a wait msg when the machine is doing nothing.

I searched around and was unable to find a similar option in marlin. 

Would it be possible to add such an option or if it is there point me to it?

fiveangle, 2017-07-16:
Search configuration.h for HOST_KEEPALIVE_FEATURE.  I believe it does what you want, but doublecheck.

mmiscool, 2017-07-16:
I did try to enable that and looked with a serial terminal. Did not see any thing happening when the printer was idle.

p3p, 2017-07-18:
HOST_KEEPALIVE_FEATURE is only for keeping the Host notified when running a long command, so its a busy signal rather than idle signal, as far as I know there is no idle keep alive feature available in Marlin.

thinkyhead, 2018-02-09:
What you're looking for is `NO_TIMEOUTS`. Set to non-zero to get a "wait" message.

closed 2018-02-09 by thinkyhead