MarlinFirmware/Marlin #7300: Layer shift at Y axis at random heights on 1.1.3 and 1.1.4

andergrin, 2017-07-16:
I have problem with 1.1.3 and 1.1.4 versions - shifted layer on the Y axis on the random hight. GCODE generated by Simplify 3D. Previosly i use'd RC7 and it's work fine without any problems, and settings in 1.1.3 and 1.1.4 was identical with RC7. So now i have to return to RC7. Printer I use - Prusa i3 clone, RAMPS 1.4 board.
[1.1.4.zip](https://github.com/MarlinFirmware/Marlin/files/1150869/1.1.4.zip)
[1.1.0-RC7.zip](https://github.com/MarlinFirmware/Marlin/files/1150870/1.1.0-RC7.zip)

Kaibob2, 2017-07-16:
This sounds exactly like #7009

andergrin, 2017-07-16:
It's not my case. I have active cooling on: 2 fans on RAMPS board exactly.

Kaibob2, 2017-07-16:
The Jerk and acceleration handling was changed somewhen. Have you tried lowering these values or printing the same object at a lower speed?

andergrin, 2017-07-17:
I lower acceleration (2000 -> 1500) and jerk (20 -> 10) on 1.1.4 version, and after several print its seems to work fine. Will hope, that's solve the problem.

iNherjer, 2017-08-10:
I have the same problem since updating to 1.1.4 ... it's on the Z tower (the one on the back) it loses steps at different highs ... 
changed the motor 
Changed the driver 
Checked the endstop 
Checkt the belt 
Checked the pulley 

Still the same, it drives me crazy 

I lowered the acceleration and jerk via the display but that didn't seems to affect the movement- also the speed, I can turn it down to 10% but the printer still moves at full throttle... it's weird 

Oh and i tried to reduce the segments per second to 60 and tried to run the motors at 1/16 instead of 1/32 ... same problem

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead