MarlinFirmware/Marlin #7304: Calibrating Z-Height: M564 S0 not allowing to move below stated Z-Height

mbabauer, 2017-07-16:
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1151372/Configuration.h.zip)
### Bug Report
- Description: While trying to manually calibrate my Delta (Anycubic Kossel), issuing the `M564 S0` does not allow me to move the printhead below Z0.
- Expected behaviour: I am under the impression [this command](http://reprap.org/wiki/G-code#M564:_Limit_axes) should allow the printhead to move out of bounds.
- Actual behaviour: Printhead stops at Z0
- Steps to reproduce: Try to move the printhead below Z0 using Printerface

p3p, 2017-07-17:
```
// If enabled, axes won't move below MIN_POS in response to movement commands.
#define MIN_SOFTWARE_ENDSTOPS
```
I'm pretty sure this stops traversal outside of bounds at all times, M564 is not in the list of supported gcode (http://marlinfw.org/meta/gcode/).

mbabauer, 2017-07-17:
You are quite right.  I'll turn that back off so I can calibrate.  Thanks!

flodejr, 2017-07-17:
You can use M211 S0 to turn off the software endstops

thinkyhead, 2018-02-09:
`M564`?

closed 2018-02-09 by thinkyhead