MarlinFirmware/Marlin #7305: Display filename when printing from SD/LCD

iBuMx, 2017-07-16:
Hello, is it possible to display the filename of the file that is printing in the LCD's row status without modifying the gcode?
(I'm aware that with the M117 you can show anything on the display, but I want to show it when I select the file from the LCD and begin the print, and my printer is not connected to a PC). 

To be more specific:
When I select the file to print on the SD, the message on the LCD is "Heating...", then "Heating done." and the printer begins to print... but "Heating done" remains forever. I want it to display the name of the file I'm printing.

I'm an amateur and would appreciate any kind of help.

tcm0116, 2017-07-18:
Looking at the code, it seems as if the file name is printed to the LCD as soon as it is opened. However, the call to M109 in the file to heat the nozzle will then result in the LCD status message being overwritten with "Heating..." followed by "Heating done."

What slicer are you using? It might be possible to add a M117 to the end of your start gcode that will display the file name without you having to modify every gcode file you generate.

rmoravcik, 2018-02-08:
Hi, I really liked the idea of showing filename of printed file in status line, so I tried to implement it.
I'm currently testing the implementation and it seems to work.

So, if anyone is interested in testing it, the code is here:
https://github.com/rmoravcik/Marlin/commit/827fabe118ca88c4ceaf19559d41929a6baf8cfb

oli71, 2018-02-14:
Dear experts, I'm interested in the feature as well, but I have a REPRAP_DISCOUNT_SMART_CONTROLLER and the progress bar active, flipping every few seconds. I would be interested to still get the filename instead of "Heating done".

boelle, 2019-02-19:
@thinkyhead has this not been added? i seem to remember that when i last printed from sd it showed the filename during print

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

ellensp, 2020-08-24:
Been implemented for ages, closing.

closed 2020-08-24 by ellensp