MarlinFirmware/Marlin #7309: SD card not working

fr33dimension, 2017-07-18:
I am using Marlin 1.1.4 with a viki LCD. The sd card is not working correctly. The light on the viki is off, The viki displays no card. I can select "no card" and see the card content, and when selecting a file to print, the sd led blinks a few times and goes dark and nothing else happens. The sd content is also displayed in Repetier. When inserting or removing the card, nothing happens.

This is the case since upgrading to 1.1

simon-jouet, 2017-07-18:
Did you see https://github.com/MarlinFirmware/Marlin/issues/7279? it might be that the pins are incorrect

fiveangle, 2017-07-18:
and if that isn't relevant, you may want to provide your config*.h zipped up and attached here for review

fr33dimension, 2017-07-25:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1174577/Configuration.zip)
Thanks guys for getting back and sorry for the delay, please see attached and let me know if you see anything.

fr33dimension, 2017-07-30:
It's pin 82 SD card detect, for the Viki (not viki2), it was on pin 81.

thinkyhead, 2018-02-09:
We have `miniVIKI` and `LCD_I2C_VIKI` as alternative options, but no plain "Viki" to select from.

The `SD_DETECT_PIN` assigned for most displays in `pins_RAMBO.h` is still 81.

boelle, 2019-02-19:
@fr33dimension you still have an issue with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-11 by fr33dimension