MarlinFirmware/Marlin #7312: G29/33 problem

cdedwards, 2017-07-19:
yesterday I tried a G29 and it worked today I was going to attempt to refine the mesh. When I started the G29, everything looked good until about 20mm above the bed, at which point the probe started probing at that height. almost like software limits are on. so I did a M211 S0 to turn that off after restarting the printer. no matter what I do, it seems to want to probe about 20mm above the height it should. This includes the G33.

iteration 1 of several. notice the height. considerably less that the real height of 341mm
READ: Iteration : 01                                    std dev:0.406
READ: .Height:328.24    Ex:-1.78  Ey:-1.96  Ez:+0.00    Radius:166.81
READ: .Tower angle :    Tx:-0.25  Ty:-0.78  Tz:+0.00

Unified Bed Leveling:
M420 S0
Endstop adjustment:
M666 X-0.95 Y-1.65 Z0.00
Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
M665 L360.32 R169.50 H341.25 S160.00 B100.00 X0.50 Y-0.50 Z0.00
Z-Probe Offset (mm):
M851 Z-1.65

M211 S0
Soft endstops: Off  Min: X-100.00 Y-100.00 Z-12.66  Max: X100.00 Y100.00 Z329.14

[Marlinconfig.zip](https://github.com/MarlinFirmware/Marlin/files/1157722/Marlinconfig.zip)

So where is Marlin retrieving the erroneous height. Mind you, occasionally this works. not sure exactly how to trigger this behavior though.

Roxy-3D, 2017-07-19:
When you say:
> yesterday I tried a G29 and it worked today I was going to attempt to refine the mesh. When I started the G29, everything looked good until about 20mm above the bed, at which point the probe started probing at that height.

do you mean you were doing a G29 P1 ?    Or do you mean on the interactive mesh edit - G29 P4 R   ?

G29 P4 R is going to position the nozzle above the bed.   Probably you should do a G26 Mesh Validation Pattern prior to doing the G29 P4 R.   By having the validation pattern still on the bed, you can make a determination at each mesh point whether it needs to move up or down, and by how much.    

You have   #define UBL_MESH_EDIT_MOVES_Z turned on so the nozzle will move as you adjust it up and down.   But its height above the bed is not important at this part of the mesh setup.

cdedwards, 2017-07-19:
I'm not at the point of doing a manual interactive edit. I'm just using G29 S0 P1. Right now G29 is attempting to probe way outside where I want it probing. I set the M665 B to 100 and then down to 90. On the left side it's not probing or at least I can't let it continue as it probes way outside of the 90mm radius it should be attempting and actually crashes the head into a upright.

Roxy-3D, 2017-07-19:
> I set the M665 B to 100 and then down to 90. 

I just noticed this is a Delta.   It might be worth while to start over.   And with the nozzle at the center of the bed do:
- G29 P1 
- G29 T
- G29 P3  (of some flavor)
- G29 P3 (of some flavor)
- G29 T
- G29 S0

And if that goes OK...  We will pick up from there and continue on...

cdedwards, 2017-07-19:
I have started over. M502, M500. And M501 to verify. M211 S0 to turn off soft. although I have now just completely disabled min/max soft limits. At this point, I start a G28, followed by a G29 P1. 12mm off the bed it starts to end probing at each point. and eventually crashes into the Z column. If it does work sorta correctly, it will probe the bed at each point about 12mm off the bed in each location. M211 returns a Z offset of aprox 12mm which is subtracted from the 341mm I should be probing at. M851 is set to -1.65 as this is the height of the IR probe from the bed for triggering.

cdedwards, 2017-07-19:
[image]

This is a pic of the one time I did get it working properly.

oldmcg, 2017-07-20:
It should never move outside DELTA_PRINTABLE_RADIUS, at least on bugfix_1.1.x or newer.  Your config shows DELTA_PRINTABLE_RADIUS at 155mm which is a 310mm diameter bed?  Is there really that much room for your delta to move laterally without binding/crashing?  Your comment above 90mm radius but that doesn't match the configuration.h file you posted.

LVD-AC, 2017-07-31:
> I set the M665 B to 100 and then down to 90...
This is the delta_calibration_radius used in G33 and the LCD manual calibration menu. G29 does not do any calibration to the printer.

closed 2017-11-17 by cdedwards