MarlinFirmware/Marlin #7313: Auto Level Correct is inverted

agambini, 2017-07-19:
Hi Guys
I trying to configure the Auto Level in a Ramps 1.4 
But the auto level looks correct inverted, where nozzle have to up, down , where have to down , up
I tried all the found in internet and nothing resolve this issue
Can you help me ?
I put a picture from the printed test 

Marlin 1.1.X (1.1.3)
[image]

[log-autolevel.txt](https://github.com/MarlinFirmware/Marlin/files/1158027/log-autolevel.txt)

cisforcer, 2017-07-19:
i   have the same  problem with  1.1.0 RC8 custom by sean liang （https://github.com/sean-liang/Marlin）  which  running in anycubic I3。
this is my pic showing the moving of extruder  ，its looks like correct over

cisforcer, 2017-07-19:
[image]

FHeilmann, 2017-07-19:
> 1.1.0 RC8 custom by sean liang

you probably won't get any support for this issue unless you update to a current version of the firmware (as of now, 1.1.4) and stay away from custom versions (unless their changes are limited to the configuration changes)

cisforcer, 2017-07-19:
@FHeilmann    i modified  the   firmware  that  given by  the  anycubic i3 （no ABL ）  to  add  AUTO BED LEVEL ， i  had the same problem。

agambini, 2017-07-19:
Hi guys , after very very long time, the things look work now
I had to desactive EEPROM in firmware and adjust the Z offset in slicer, and all looks good
I'm doing more tests to have sure

cisforcer, 2017-07-19:
you mean // #define eeprom？
not to use eeproom？

agambini, 2017-07-19:
yes

agambini, 2017-07-21:
Unfortunately , after some prints , I realize that not was working proper, the auto bed still not working
I dont know what do more
Have someone some clue  ?
Is possible that probed values no saved ?
Why the inverted values are apllied ?

agambini, 2017-07-21:
I already using the current version 1.1.4

cisforcer, 2017-07-21:
see more in #7278

Roxy-3D, 2017-07-21:
@agambini 
Different people wrote and support different AUTO_BED_LEVELING systems.   Which one are you seeing the problem on?

agambini, 2017-07-21:
Auto bed level bilinear grid

agambini, 2017-07-22:
Hi guys
I tried to use manual probe option, and works great
So I don't understand because the automatic option not work, looks the auto option not store the mesh

cisforcer, 2017-07-23:
manual  probe option works？ so the auto  bed levelling  has bugs we dont konw？

agambini, 2017-07-23:
Hi
Works, I too not understood
and works very, very well

closed 2017-07-26 by thinkyhead