MarlinFirmware/Marlin #7314: [FR] Add tildes to spanish translation

MrAmericanMike, 2017-07-19:
So this is a Feature request I guess. I don't know if it's something that is missing on the translation, or if it's the firmware not been able to use letters with tildes.

But the question/request is, would it be possible to have the tildes on the spanish translation?

Like: á é í ó ú and letters like ñ (in case of needed)

If this is doable, I postulate myself to help fix the translation to include the tildes ;)

thinkyhead, 2018-02-09:
> If this is doable, I postulate myself to help fix the translation to include the tildes ;)

That would be great. You just need to make a UTF-8 version of the `language_es.h` file. See other "_utf8" versions of languages to see what is required.

MrAmericanMike, 2018-02-19:
Ok I did this file. I'm not sure what else I need to change on the configs to test it properly, as when I use mine, or even the PT one with UTF8 support, I get weird characters on my display. I guess I need to change some other variable I don't know somewhere.
But here is the file.
https://pastebin.com/uMWzv0qk
Soon I will have my 3D Printer ready (One I'm building) and use this a lot more, so at that point, I can probably improve the translation. As for now I basically used the already provided spanish one. (Mainly changed "Plataforma" to "Cama" (Platform to Bed) and added all the tildes.
Let me know if you know what else I should change for the tildes to display correctly. 
Thanks a lot.

AnHardt, 2018-02-19:
Try to add
```
#define MAPPER_C2C3
```
near the charset definition.

MrAmericanMike, 2018-02-23:
Perfect, is working correctly now.

This version:
https://pastebin.com/hCnG1Fqt

Fixed also some #define with the word MENU on it that were wrongly translated to Menú (Defines shouldn't be translated) my mistake by using find/replace

But now it works, I used: #define LCD_LANGUAGE es_utf8

With the file language_es_utf8.h present and tildes work nicely.

PS: Someone should add the #define MAPPER_C2C3 to the PT-BR file also, so that gets fixed also.

thinkyhead, 2018-02-24:
Thanks much! PRs are ready to go and will be merged shortly!

closed 2018-02-24 by thinkyhead