MarlinFirmware/Marlin #7317: [FR] lack of material sensor

Firesped, 2017-07-19:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
I have built a special version of the marlin firmware based on the 1.1.0 build. It requires special coding to talk to the raise3d touchscreen. since I didn't want my repository to be telling me it was constantly out of date, I created it's own, located here. https://github.com/Firesped/Marlin-Raise3D-N-Series-Unofficial

raise3d put out some code to add a simple mechanical filament runout sensor and I am trying to get it working on this firmware. I have the code ported over. the code for m119 reports the state of the endstop correctly.

the other portion of code is located in checkHitEndstops in stepper.cpp on marlin 1.0.2 build.

I don't know where that went in the 1.1.0 build and I need help finding it please.

Thank you,

Roxy-3D, 2017-07-19:
endstops.cpp ???    AstroGrep helps you find things pretty quickly if you install it.

Firesped, 2017-07-19:
This is the code in 1.0.2 based firmware. I have tried putting it into both endstops::report_states and endstops::update. I'm 
code from 1.0.2
`void checkHitEndstops() { 
  static bool lack_checked_e0=0;       //raise3d lack of materia
  static bool lack_checked_e1=0;       //raise3d lack of materia
  static unsigned int lack_check=0;    //raise3d times
  if (endstop_x_hit || endstop_y_hit || endstop_z_hit) {
    SERIAL_ECHO_START;
    SERIAL_ECHOPGM(MSG_ENDSTOPS_HIT);
    if (endstop_x_hit) {
      SERIAL_ECHOPAIR(" X:", (float)endstops_trigsteps[X_AXIS] / axis_steps_per_unit[X_AXIS]);
      LCD_MESSAGEPGM(MSG_ENDSTOPS_HIT "X");
    }
    if (endstop_y_hit) {
      SERIAL_ECHOPAIR(" Y:", (float)endstops_trigsteps[Y_AXIS] / axis_steps_per_unit[Y_AXIS]);
      LCD_MESSAGEPGM(MSG_ENDSTOPS_HIT "Y");
    }
    if (endstop_z_hit) {
      SERIAL_ECHOPAIR(" Z:", (float)endstops_trigsteps[Z_AXIS] / axis_steps_per_unit[Z_AXIS]);
      LCD_MESSAGEPGM(MSG_ENDSTOPS_HIT "Z");
    }
    SERIAL_EOL;

    endstops_hit_on_purpose();

    #if defined(ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED) && defined(SDSUPPORT)
      if (abort_on_endstop_hit) {
        card.sdprinting = false;
        card.closefile();
        quickStop();
        setTargetHotend0(0);
        setTargetHotend1(0);
        setTargetHotend2(0);
        setTargetHotend3(0);
        setTargetBed(0);
      }
    #endif
  }
  //Start Lack of Materia
  lack_check++;
  if(lack_check>=10){
    lack_check=0;
    //Lack of material testing
    #if defined(E0_MATERIAL_LACK_PIN) && E0_MATERIAL_LACK_PIN > -1
      if(READ(E0_MATERIAL_LACK_PIN)^E0_LACK_ENDSTOP_INVERTING){
        if(lack_checked_e0==0){
          SERIAL_PROTOCOLLN("Custom: Filament Error T0");
          lack_checked_e0=1;
        }
      }else{
        lack_checked_e0=0;
      }
    #endif
    #if defined(DUAL)
    #if defined(E1_MATERIAL_LACK_PIN) && E1_MATERIAL_LACK_PIN > -1
      if(READ(E1_MATERIAL_LACK_PIN)^E1_LACK_ENDSTOP_INVERTING){
       if(lack_checked_e1==0){
          SERIAL_PROTOCOLLN("Custom: Filament Error T1");
          lack_checked_e1=1;
        }
      }else{
        lack_checked_e1=0;
      }
    #endif
    #endif
  }
  //end Lack of Materia
  
}`

code for 1.1.0
`    #if ENABLED(RAISE3D_FILAMENT_RUNOUT_SENSOR)
      static bool lack_checked_e0=0;//lack of materia
      static bool lack_checked_e1=0;//lack of materia
      static unsigned int lack_check=0;//times
      #if ENABLED(RAISE3D_DEBUG)
        SERIAL_ECHOLN("sensor ping");
      #endif
      lack_check++;
        if(lack_check>=10){
          lack_check=0;
          //Lack of material testing
          if (planner.lack_materia_sensor_state[0] == true) {
            #if ENABLED(RAISE3D_DEBUG)
              SERIAL_ECHOLN("testig Filament E0");
            #endif
            #if defined(E0_MATERIAL_LACK_PIN) && E0_MATERIAL_LACK_PIN > -1
              if(READ(E0_MATERIAL_LACK_PIN)^planner.lack_materia_sensor_norm[0]){
                if(lack_checked_e0==0){
                  SERIAL_PROTOCOLLN("Custom: Filament Error T0");
                  lack_checked_e0=1;
                  }
                }
              else{
                lack_checked_e0=0;
                }
          }
          #endif
          if (planner.lack_materia_sensor_state[1] == true) {
            #if ENABLED(DUAL)
              #if ENABLED(RAISE3D_DEBUG)
                SERIAL_ECHOLN("testig Filament E1");
              #endif
              #if defined(E1_MATERIAL_LACK_PIN) && E1_MATERIAL_LACK_PIN > -1
                if(READ(E1_MATERIAL_LACK_PIN)^planner.lack_materia_sensor_norm[1]){
                  if(lack_checked_e1==0){
                    SERIAL_PROTOCOLLN("Custom: Filament Error T1");
                    lack_checked_e1=1;
                    }
                  }
                  else{
                    lack_checked_e1=0;
                    }
              #endif
            #endif
          }
        }
    #endif`

not sure if I broke something with the added code.

Firesped, 2017-07-19:
I figured out where it needed to go. It goes in the endstop::report_states. I had it inside the endstop_hit_bits if statement and it needed to be outside of it.

closed 2017-07-19 by Firesped

fiveangle, 2017-07-20:
fyi, is the code simply to check endstops in the stepper isr ?  if so, we have interrruptable endstops feature in case it may be related (so you can use a feature of marlin that exists already instead of having to constantly port your code.  same with filament runout sensor feature.

Firesped, 2017-07-21:
It's raise3d's code. I'm just trying to make sure it reports back what they setup. I'm sure some of this could be done cleaner.  personally I would like the code to automatically pause but I don't have direct access to the touchscreen software.