MarlinFirmware/Marlin #7318: [FR] More descriptive error message for thermal runaway

mperdue, 2017-07-20:
I recently had a thermal runaway on one of my printers. After a bit of troubleshooting I determined it was a defective relay on one of my bed heaters. Then, a couple days later, I got  the same error on my other, identical, printer. I assumed that the error was, once again with the bed heated since the error message was the same. After messing with it for far too long I determined that the problem was actually with the extruder (and only when the printer gantry was moving.) 

I could have saved a fair amount of time if the error message had identified if it was the extruder or the bed heater that was failing, i.e. "Extruder: Thermal Runaway" or "Bed: Thermal Runaway."

fiveangle, 2017-07-23:
Error routine could certainly be duplicated so the bed and hot end are separate (with whatever small increase in firmware size for the code and new strings) buy it seems extremely easy to troubleshoot if an issue is bed or not since one can control them independently ?

mperdue, 2017-07-23:
True, but that can be said for any error. These machines are reasonably simple devices that allow you to figure out what the problems are without too much effort. That's no reason to not provide a little additional information to speed the troubleshooting process.

thinkyhead, 2018-02-09:
@MarlinFirmware/host-software-team Would it adversely affect hosts if we add "Bed" or "Extruder" to the current thermal runaway message?
```cpp
#define MSG_T_THERMAL_RUNAWAY               "Thermal Runaway"
```

foosel, 2018-02-09:
@thinkyhead no parsing and hence no problem with OctoPrint

repetier, 2018-02-09:
We don't parse it either. Just write it to log, so should be fine.

thinkyhead, 2018-02-17:
@mperdue  — Apparently the thermal runaway message that we print to serial output is already detailed enough…
```
error: Thermal Runaway, system stopped! Heater_ID: 0
error: Thermal Runaway, system stopped! Heater_ID: bed
```
But at this time the LCD always shows "THERMAL RUNAWAY" for both.

There's not enough space on a 16 character display to say "`EXTRUDER: THERMAL RUNAWAY`." But we can fit "BED THERMAL RUNAWAY" within 19 characters, or shorten it to "BED RUNAWAY TEMP" to fit within 16 characters. So, perhaps we can have a separate message pertaining specifically to the bed for your convenience, but leave the extruder message as it is.

mperdue, 2018-02-17:
It's not that big of a deal. I'd leave it as is unless someone else is wanting it changed.

thinkyhead, 2018-02-28:
@mperdue — It has been changed in the firmware. A bed error includes "BED." Otherwise it's an extruder. Later we might extend it to show `E0`, etc.

closed 2018-02-28 by thinkyhead