MarlinFirmware/Marlin #7320: BUG: manual Move Axis Delta extremities

JanVelleman, 2017-07-20:
### Bug Report
- Description: When move an axis (X or Y or Z) manually via the menu to the max (like "-100mm" on a 200mm bed) on a Delta after homing the Delta goes crazy(sends 1 slider fully up and does not stop at endstop).
- Expected behaviour: The printer should go to -100mm and stop.
- Actual behaviour: The printer sends one slider up and it does not stop at the endstop causing a collision.
- Steps to reproduce:
  - First home on a Delta
  - Then choose move axis on LCD
  - Choose an axis and then 10mm
  - Scroll to the maximum possible.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1162009/Config.zip)

JanVelleman, 2017-07-20:
Forgot to mention this is on version 1.1.4.

Ordenkeim93499, 2017-12-30:
I discovered the exact same issue with my Delta today. 
I only wanted to update Marlin from 1.1.5 to 1.1.8 and add TMC2130 Stepsticks but well even with the old Stepsticks i'am having the same issue. 
I hope i will find a solution or maybe you've already found one.

thinkyhead, 2018-02-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-19:
@JanVelleman Please test with the latest `bugfix-2.0.x` branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead