MarlinFirmware/Marlin #7321: FIX_MOUNTED_PROBE Marlin 1.1.4

HansGrub3r, 2017-07-20:
Hi!

I've got a Mini Kossel which currently got Marlin 1.1.0 Rc7 on it which I wanted to update to the newest Marlin.

Going through Configuration.h down at the probing section I can't really find the proper probe for my specific printer, this Mini Kossel is using the nozzle as a probe.
On previous version There was this line:

```
// A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
// For example an inductive probe, or a setup that uses the nozzle to probe.
// An inductive probe must be deactivated to go below
// its trigger-point if hardware endstops are active.
#define FIX_MOUNTED_PROBE
```

And on the newest version that particular probe isn't mentioned other than in comments, no real probe is there to define really.

```
/**
 * Probe Type
 *
 * Allen Key Probes, Servo Probes, Z-Sled Probes, FIX_MOUNTED_PROBE, etc.
 * Activate one of these to use Auto Bed Leveling below.
 */
```

So at the moment I am kinda stuck here, I guess I could do as way back and tell the printer it uses a allen key probe just that I have to make sure that it's not going to try and deploy/ retract it I guess.

Anyway thanks for keeping this firmware going!

FHeilmann, 2017-07-20:
Look at line [601 of configuration.h](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L610)

HansGrub3r, 2017-07-20:
That's not in my configuration.h ..the heck ?!

FHeilmann, 2017-07-20:
something mustve gone wrong then, 

I just now grabbed this zip file: https://github.com/MarlinFirmware/Marlin/archive/1.1.4.zip from the releases page and the section is there as expected!

HansGrub3r, 2017-07-20:
Oh I know what went wrong, I did take the example file for mini kossel. Whoops, anyway thanks for helping out mate!

teemuatlut, 2017-07-20:
All the example configuration files need to updated whenever there's a change in either configuration.h or configuration_adv.h. It's a bit of a chore really when you just want to add a simple new toggle for a feature and there are so so many configuration files you have to go through.

fiveangle, 2017-07-20:
If the example is for your specific printer, it should be updated.  You likely took the example from a previous release ?

If you find it isn't, you'll have to manually go through the configuration*.h files by hand on the latest (and honestly, if you go through the trouble, you may want to do it on rcbugfix1.1.x because there were some delta fixes in there for G33 and others that are pretty important i think.

Once you've figured it out and fixed the "config" problem, submitting an update PR to fix the example configuration would help others (if it is indeed broken).

closed 2018-02-09 by thinkyhead