MarlinFirmware/Marlin #7322: LCD not work

CosminDB, 2017-07-20:
------------------------------------------------------------
### Compile Error
When I upload the program in arduino the LCD only show all pixel on.
When active hot bed( with thermistor 11) and LCD option: #define ULTRA_LCD. Compilation fails

```
In file included from sketch\ultralcd.cpp:78:0:

sketch\ultralcd_impl_HD44780.h: In function 'void lcd_implementation_status_screen()':

ultralcd_impl_HD44780.h:702: error: 'CHAR' was not declared in this scope

         lcd.print((CHAR)LCD_BEDTEMP_CHAR);

                    ^

exit status 1
'CHAR' was not declared in this scope
```

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1162270/Configuration.zip)

fiveangle, 2017-07-20:
ULTRA_LCD is a feature of a board, but not a an actual board, as I understand it.  You need to set the LCD panel you actually have (REPRAP_FULL_DISCOUNT_SMART_CONTROLLER for an example).

See similar discussion about enabling display here: https://github.com/MarlinFirmware/Marlin/issues/7301

thinkyhead, 2018-02-09:
That was a typo/bug. Now fixed.

closed 2018-02-09 by thinkyhead