MarlinFirmware/Marlin #7323: [FR] Make new behavior of G92 an option

marcio-ao, 2017-07-20:
In Marlin 1.0, G92 would set the head position. In Marlin 1.1 this behavior has changed in a way that is backwards incompatible. Now G92 not only sets the position, but also adjusts the software endstops.

For a prototype product, we have a particular use case where our Z-Homing button is raised 5.5mm above the print bed. In Marlin 1.0, I worked around the problem by issuing "G28 Z" followed by "G92 Z5.5" which would have the effect of correcting the offset. Under the new Marlin, the G92 adjusts the software endstops such that I can no longer reach the bed.

I realize the old behavior can be restored by enabling NO_WORKSPACE_OFFSETS, but this also disables workspace offsets, which I assume could be useful to have in some cases. I would like to propose that G92 retain its old behavior, but include an additional flag for adjusting the software endstops, so that this becomes a user choice and does not require workspace offsets to be disabled.

thinkyhead, 2018-02-09:
The old behavior of `G92` was absolutely wrong w/r/t software endstops. Under the most recent Marlin we got rid of internal offsets to the workspace, so it always operates in native workspace internally.

This means that your `G92` offset is now subtracted from XYZ on input from G-move commands, and added to the output values when the position is reported, as with `M114`. It also means that internally the endstops never change.

You'll need to describe your use-case better, and then we may be able to pinpoint a more proper solution. In the meantime, you can simply set your `MANUAL_Z_HOME_POS` to 5.5 so Marlin knows that when the endstop triggers, it is 5.5mm above the bed.

marcio-ao, 2018-08-24:
@thinkyhead: In 1.1.9, I see MANUAL_Z_HOME_POS is referenced only in "Conditional_post.h":

```
#ifdef MANUAL_Z_HOME_POS
  #define Z_HOME_POS MANUAL_Z_HOME_POS
#else
  #define Z_HOME_POS (Z_HOME_DIR < 0 ? Z_MIN_POS : Z_MAX_POS)
#endif
```

However, Z_HOME_POS is referenced nowhere else in the Marlin code. I could be have missed something, but I suspect MANUAL_Z_HOME_POS does nothing.

-- Marcio

Roxy-3D, 2018-08-24:
You are correct!    We should probably remove it.

marcio-ao, 2018-08-24:
@Roxy-3D : The configuration option is useful, aside the fact it is broken right now. We have a custom modification in our FW that does just that. The reason is in one of our toolheads the Z_MIN switch position is actually higher than the Z_MIN travel limit.

The other modification I do to your FW is I modify the homing procedure so that rather than leaving the toolheads in a position in which the endstop triggered, it moves away from the endstops by a configurable amount in X,Y, Z (this is different than bump, because bump actually re-homes a second time). Parking away from the endstops seems to be necessary when using sensorless homing to avoid rattling if two "G28s" are done in a row, so I'm curious how people have made do without it.

closed 2018-10-15 by marcio-ao