MarlinFirmware/Marlin #7324: [FR] Raise to Z_SAFE_HOMING after G28

marcio-ao, 2017-07-20:
Currently G28 raises the head to Z_HOMING_HEIGHT prior to homing, but it does not raise the head after homing. In the case of LulzBot printers, the homing button is recessed, so the head will crash if it is not raised after G28.

If the purpose of Z_HOMING_HEIGHT is to allow the head to clear obstacles prior to homing, it would seem like it would be logical to have it return to that safe position after homing.

An alternative, perhaps more general solution, is to offer a AFTER_Z_HOME_CMD which could be set to GCODE to be executed after homing. This could be used for raising the head, or performing other adjustments (such as using G92 to adjust the offset, see #7323). This is the solution I am choosing to implement in the LulzBot branch of Marlin 1.1.4.

thinkyhead, 2018-02-09:
I propose a new feature, if vendors are commonly going to want this.
```cpp
#define PARK_AFTER_HOMING { X_HOME_POS, Y_HOME_POS, Z_HOME_POS + 10 }
```

marcio-ao, 2018-08-20:
Catching up on old comments...

Yes, something like this would be useful. However, I would recommend that instead of using a specific hard coded action like PARK_AFTER_HOMING, that there is instead something like:

```
//#define AFTER_HOMING_GCODE "G0 Z10\nG0 X100 Y100\nM117 Homing complete"
```

This allows the user to craft their own action of arbitrary complexity. This action can be executed in the code using "execute_commands_immediate_P"  from PR https://github.com/MarlinFirmware/Marlin/pull/10450, something like:

```
#if defined(AFTER_HOMING_GCODE)
execute_commands_immediate_P(PSTR(AFTER_HOMING_GCODE));
#endif
```

I feel like Marlin should move away from hard-coded actions and make better use of GCODE scriptlets as it give the user and vendors so much more flexibility.

closed 2018-10-15 by marcio-ao